micropython/micropython #12651: ESP32 asyncio run in _thread

straga, 2023-10-10:
I'm using MicroPython v1.22.0-preview.4.g9f835df35.dirty on an ESP32 device without PSRAM, with ESP-IDF v5.0.2.

The problem arises when running asyncio in a thread. Here are the steps that lead to the issue:

- Run:

    loop = asyncio.get_event_loop()
    _ = _thread.stack_size(12 * 1024)
    _thread.start_new_thread(loop.run_forever, ())

- Connect to WIFI successfully.

Inside the loop, connect to MQTT using asyncio:
reader, writer = await asyncio.wait_for(asyncio.open_connection(self.addr, self.port), 5)

- Run the WebREPL service.

- Try connecting to WebREPL and typing commands or a lot time some "sdfsdfdsfds".

However, during this process, the following issues occur:

Loss of ping and disconnection from Wi-Fi:
Disconnected, received disassociation: no activity (4), signal strength -35
WebREPL raises an exception in the read() method and deactivates with the error:
OSError: [Errno 113] ECONNABORTED

[image]

- Interestingly, when running the same code without a _thread, it works normally. Additionally, when using an ESP32 with PSRAM and thread, the code also works normally too.

andrewleech, 2023-10-10:
Is there asyncio code running on the main thread as well as the new thread?

Micropython doesn't support multiple event loops so accessing asyncio on multiple threads definitely will cause conflicts and failures.

You should restrict asyncio tasks to the main thread and use (asyncio.ThreadSafeFlag)[https://docs.micropython.org/en/latest/library/asyncio.html#class-threadsafeflag] to communicate between threads.

straga, 2023-10-10:
asyncio only run once inside thread. Runing only one thread.
Code work very well on esp32 with psram. But with out psram work not normal.

straga, 2023-10-16:
Debug log:
[image]