micropython/micropython #12660: Crashes in mpz_hash

gwangmu, 2023-10-11:
# Description

In the UNIX port version, we found MicroPython crashed in `mpz_hash` while executing `uctypes.struct()`. The crash exhibited in two ways; heap-buffer-overflow when the argument `addr` was another module and null-dereference when it was a value (e.g., string). We have attached three PoCs for each crash case.

[poc.zip](https://github.com/micropython/micropython/files/12868355/poc.zip)

# Proof of Concept

```
$ # build unix port with ASAN, at the root source code directory.
$ export CC=clang
$ export CXX=clang++
$ export CFLAGS="-fsanitize=address -fno-omit-frame-pointer"
$ export CXXFLAGS=$CFLAGS
$ export LDFLAGS=$CFLAGS
$ export DEBUG=1
$ make -C mpy-cross -j
$ make -C ports/unix -j all lib
$
$ # run a poc.
$ export ASAN_OPTIONS="detect_leaks=0"
$ ./ports/unix/build-standard/micropython <poc_file>
```

# Environment

Ubuntu 20.04
Intel(R) Xeon(R) Gold 5218 CPU @ 2.30GHz
Memory: 64 GB

# Affected Version

v1.20.0 (commit a3862e726, latest as of 2023-09-26)
v1.20.0 (commit 813d559bc, 2023-06-19)
Discovered in the UNIX port version.

dpgeorge, 2023-10-11:
Thanks for the report.  It looks like this is all due passing in a non-integer object as the first argument to `uctypes.struct()`, which assumes the first argument is an integer.

This can be fixed by changing a call to `mp_obj_int_get_truncated()` to `mp_obj_get_int_truncated()`, which will first validate that the argument is an integer.

dpgeorge, 2023-10-12:
See #12669 for a fix.

closed 2023-10-12 by dpgeorge

gwangmu, 2023-10-12:
Thank you for a quick fix @dpgeorge. :slightly_smiling_face: