micropython/micropython #12671: Unix port doesn't build if the variant directory is a symbolic link

agatti, 2023-10-12:
Unix port doesn't build if the variant is a symbolic link

Due to project organisation reasons I have my own MicroPython custom Unix port variant outside the MicroPython source tree.  With MicroPython 1.21.0, when putting a symbolic link from my variant directory to the port's `variants/` directory (so a new directory shows up there), building fails.  I guess that one or more python scripts in the build either resolve symbolic links (or they don't, no idea)?

The errors are as follows (Arch Linux, clean build):

```bash
mkdir -p build-TEST_RUNNER/genhdr
mkdir -p build-TEST_RUNNER/build-TEST_RUNNER/
mkdir -p build-TEST_RUNNER/extmod/
mkdir -p build-TEST_RUNNER/lib/berkeley-db-1.xx/btree/
mkdir -p build-TEST_RUNNER/lib/berkeley-db-1.xx/mpool/
mkdir -p build-TEST_RUNNER/lib/littlefs/
mkdir -p build-TEST_RUNNER/lib/mbedtls/library/
mkdir -p build-TEST_RUNNER/lib/mbedtls_errors/
mkdir -p build-TEST_RUNNER/lib/oofatfs/
mkdir -p build-TEST_RUNNER/py/
mkdir -p build-TEST_RUNNER/shared/libc/
mkdir -p build-TEST_RUNNER/shared/readline/
mkdir -p build-TEST_RUNNER/shared/runtime/
mkdir -p build-TEST_RUNNER/shared/timeutils/
GEN build-TEST_RUNNER/genhdr/mpversion.h
GEN build-TEST_RUNNER/genhdr/qstr.i.last
In file included from ./mpconfigport.h:36,
                 from ../../py/mpconfig.h:75,
                 from ../../py/mpstate.h:31,
                 from ../../py/mpstate.c:27:
variants/TEST_RUNNER/mpconfigvariant.h:3:10: fatal error: ../mpconfigvariant_common.h: No such file or directory
    3 | #include "../mpconfigvariant_common.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
[A few more of these]
Traceback (most recent call last):
  File "<...>/micropython/ports/unix/../../py/makeqstrdefs.py", line 211, in <module>
    preprocess()
  File "<...>/micropython/ports/unix/../../py/makeqstrdefs.py", line 76, in preprocess
    for output in p.imap(pp(flags), chunks):
  File "/usr/lib/python3.11/multiprocessing/pool.py", line 873, in next
    raise value
  File "/usr/lib/python3.11/multiprocessing/pool.py", line 125, in worker
    result = (True, func(*args, **kwds))
                    ^^^^^^^^^^^^^^^^^^^
  File "<...>/micropython/ports/unix/../../py/makeqstrdefs.py", line 60, in run
    return subprocess.check_output(args.pp + flags + files)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/subprocess.py", line 466, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/subprocess.py", line 571, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '['gcc', '-E', '-DMICROPY_VFS_FAT=1', '-DMICROPY_VFS_LFS1=1', '-DMICROPY_VFS_LFS2=1', '-DMICROPY_PY_SSL=1', '-DMBEDTLS_CONFIG_FILE="mbedtls/mbedtls_config.h"', '-DMICROPY_SSL_MBEDTLS=1', '-I../../lib/mbedtls/include', '-DMICROPY_PY_BTREE=1', '-DFFCONF_H="lib/oofatfs/ffconf.h"', '-DLFS1_NO_MALLOC', '-DLFS1_NO_DEBUG', '-DLFS1_NO_WARN', '-DLFS1_NO_ERROR', '-DLFS1_NO_ASSERT', '-DLFS2_NO_MALLOC', '-DLFS2_NO_DEBUG', '-DLFS2_NO_WARN', '-DLFS2_NO_ERROR', '-DLFS2_NO_ASSERT', '-I../../lib/berkeley-db-1.xx/PORT/include', '-I.', '-I../..', '-Ibuild-TEST_RUNNER', '-I../../shared/readline', '-Wall', '-Werror', '-Wextra', '-Wno-unused-parameter', '-Wpointer-arith', '-Wdouble-promotion', '-Wfloat-conversion', '-std=gnu99', '-DUNIX', '-Os', '-DNDEBUG', '-fdata-sections', '-ffunction-sections', '-Ivariants/TEST_RUNNER', '-g', '-U', '_FORTIFY_SOURCE', '-DMICROPY_USE_READLINE=1', '-DMICROPY_PY_TERMIOS=1', '-DMICROPY_PY_SOCKET=1', '-DMICROPY_PY_THREAD=1', '-DMICROPY_PY_THREAD_GIL=0', '-DMICROPY_PY_FFI=1', '-DMPZ_DIG_SIZE=16', '-DMICROPY_ROM_TEXT_COMPRESSION=1', '-DNO_QSTR', '../../py/mpstate.c', '../../py/malloc.c', '../../py/gc.c', '../../py/pystack.c', '../../py/qstr.c', '../../py/vstr.c', '../../py/mpprint.c', '../../py/unicode.c', '../../py/mpz.c', '../../py/reader.c', '../../py/lexer.c', '../../py/parse.c', '../../py/scope.c', '../../py/compile.c', '../../py/emitcommon.c', '../../py/emitbc.c', '../../py/asmbase.c', '../../py/asmx64.c', '../../py/emitnx64.c', '../../py/asmx86.c', '../../py/emitnx86.c', '../../py/asmthumb.c', '../../py/emitnthumb.c']' returned non-zero exit status 1.
In file included from ./mpconfigport.h:36,
                 from ../../py/mpconfig.h:75,
                 from ../../py/obj.h:31,
                 from ../../py/builtin.h:29,
                 from ../../py/builtinhelp.c:30:
variants/TEST_RUNNER/mpconfigvariant.h:3:10: fatal error: ../mpconfigvariant_common.h: No such file or directory
    3 | #include "../mpconfigvariant_common.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make: *** [../../py/mkrules.mk:111: build-TEST_RUNNER/genhdr/qstr.i.last] Error 1
make: *** Deleting file 'build-TEST_RUNNER/genhdr/qstr.i.last'
```

A quick way to reproduce the issue is as such (it also happens on Git HEAD):

```bash
cd micropython/ports/unix
mv variants/standard ../..
cd variants
ln -s ../../../standard standard
cd ..
make
```

stinos, 2023-11-08:
This is standard C (or at least gcc) behavior as far as I can tell so not something micropython can easily fix

agatti, 2023-11-08:
Well, the ESP32 port doesn't have this behaviour (although that uses CMake instead and maybe does something to work around that?).

jimmo, 2023-11-09:
@agatti Why do you need your variant to be in `ports/unix/variants`? Can you build with `VARIANT_DIR=path/to/name` instead of `VARIANT=name` ?

agatti, 2023-11-09:
@jimmo, right now I'm not involved with that particular customer project for the time being, however it was due to some compliance rules regarding the physical location of source code on the machines and whatnot.  Regarding `VARIANT_DIR`, I didn't know you can do that, I looked in the documentation at the time and there was no mention of that.