micropython/micropython #12675: Prebake vfs partition

MauroDiamantino, 2023-10-12:
Hello, I'm compiling micropython v1.18.0 in the port ESP32 with my scripts as frozen codes, but I need to store some data files (for example html files) in the vfs partition for scripts to work normally. I've tried to use rshell to copy the files after the firmware has been written in the flash, but it is slow and sometimes fails becuase the main program start to execute and then reset the board because it doesn't find the required files. So, there is a way to prebake (or prefilled) the vfs partition when the micropython firmware is built, so the firmware already has my scripts and the required data files?

AmirHmZz, 2023-10-13:
If your files are big enough, you are already doing it in the best way. 
But If you have enough space in your vfs partition (compared to files size) you can [_boot.py](https://github.com/micropython/micropython/blob/master/ports/esp32/modules/_boot.py) and inject your files in base64 encoding to be written as files on the first boot.

andrewleech, 2023-10-13:
I've used pre-built VFS partitions for loading arbitrary files on stm port before, it just needs a Unix copy of micropython to run the script: https://gist.github.com/andrewleech/3264bc910afdaf5809be99aa673766a8

That builds the filesystem into a dfu file.

That won't work for esp however, you'd need to write the filesystem to a binary file and somehow merge it into the image written to the esp. I'm not sure how they're structured myself.

glenn20, 2023-10-19:
@DvdGiessen thank you - that's very interesting. I didn't  know you could build littlefs blobs with python. I happen to have a tool for reading and writing blobs to partitions on esp32 devices or firmwware files (among other things): [`mp-image-tool-esp32`](https://github.com/glenn20/mp-image-tool-esp32). It uses `esptool.py` to read and write data to the flash storage on devices. That might be useful to you as well. The next thing I was looking for was a way to build filesystems off the device - thanks for showing me how.

robert-hh, 2023-10-20:
There is also https://github.com/littlefs-project/littlefs-fuse which allows to create & mount a file with a lfs file system, or to mount a external block device.

agatti, 2023-10-20:
I've solved this by modifying the ESP32 port and merging functionality from the `examples/storage/fatfsgen` project present in ESP-IDF.  In my case the issue wasn't generating the filesystem image but to let `make deploy` build and flash images in one step (and have some basic support for cross-platform serial port detection) in my CI/CD pipeline.

jonnor, 2024-07-13:
Is there some workflow here that could be documented to solve this issue?

vshymanskyy, 2024-07-31:
I have explored several ways to achieve this and I seem to have a nice solution:
- pack the file system content into a `tar.gz` file
- put that package into the MPY firmware. I have 2 ways to do it:
  - convert it into a '.py' file and freeze it during MicroPython build - works for ANY board
  - create a separate flash partition and put it there - works on ESP32
- Add (and freeze) a simple bootloader that runs before `boot.py` and unpacks the file when needed

As an additional benefit, this approach is easily reused for OTA updates of MicroPython apps, on any boards.
We've been using this for ~6 month already, and I'm preparing to release this solution.

glenn20, 2024-09-25:
> @DvdGiessen thank you - that's very interesting. I didn't know you could build littlefs blobs with python. I happen to have a tool for reading and writing blobs to partitions on esp32 devices or [...]

In case anyone is interested, I have incorporated [littlefs-python](https://github.com/jrast/littlefs-python) functionality into my [mp-image-tool-esp32](https://github.com/glenn20/mp-image-tool-esp32) tool (see the `--fs` options). This includes building new littlefs filesystems on serial-attached devices and reading/writing files to/from the device:

```bash
mp-image-tool-esp32 --fs mkfs vfs # Create a new littlefs filesystem on the vfs partition
mp-image-tool-esp32 --fs put rootfs  # Recursively copy the files from the rootfs directory onto the board.
mp-image-tool-esp32 --fs ls  # List all the files in the vfs partition
mp-image-tool-esp32 --fs grow vfs 4M  #  Grow the fs on vfs to 4 megabytes (4 * 1024 * 1024)
mp-image-tool-esp32 --fs grow vfs  #  Grow the fs on vfs to match the partition size (can be used after resizing a partition)
```
Available commands include: `ls, get, put, mkdir, rm, rename, cat, info, mkfs, df and grow`.

This is also generally faster than mpremote to transfer files over the serial port onto the device, eg:

```bash
mp-image-tool-esp32 --baud 1.5M --fs put bigfile.data  # Default baud rate is 921600
```

glenn20, 2024-09-25:
If you prefer to prebuild and then flash your littlefs filesystem, you can use:

```bash
littlefs-python create --compact --no-pad --fs-size 2mb --block-size 4096 rootfs vfs.bin
mp-image-tool-esp32 u0 --write vfs=vfs.bin
```

OR, you may prefer to build a small initial filesystem and then grow it to match the partition onto which you are installing.

```bash
littlefs-python create --compact --no-pad --fs-size 1mb --block-size 4096 rootfs vfs.bin
mp-image-tool-esp32 u0 --write vfs=vfs.bin --fs grow vfs
```