micropython/micropython #12689: ghost files y.mpy and .mpy on ENOENT Errors

webdeb, 2023-10-15:
Some ghost files are being created on the rp2 pico, when there is a problem (just happens sometimes) with opening a file for writing. Those files cannot be removed with `os.remove()`

I also don't know how to reproduce it properly, my file storing routine is inside an async task, and once in 5 or 10 writes, an OSError ENOENT happens.

```
f = open("file.json", "w")
f.write(json.dumps(data))
f.close()
```

When I then os.listdir(), I can see those files: **y.mpy** or **.mpy** but when I try to delete them, its says ENOENT. Strange.
The FS is `os.VfsLfs2`. On a waveshare rp2040 zero.

MP: 1.21 / But got the same problem on the latest master as well.

webdeb, 2023-10-15:
This is how the output of os.listdir("data") looks like. **.mpy** is the strange file.

```
['.mpy', '4p.json', 'settings_store.json']
```

webdeb, 2023-10-22:
This shows that those files cannot be removed "sometimes"
```
>>> import os
>>> os.listdir("data")
['pppp.json', 'settings_store.json', '.mpy', 'sixpwm.json', 'twofam.json']
>>> os.remove("data/.mpy")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: [Errno 2] ENOENT
```

webdeb, 2023-10-23:
Please help, at least point me where to look for..

[image]

webdeb, 2023-10-24:
Issue was, that I've used a relative path name for the file, and I guess, that it sometime/somehow correlated with the asyncio loop. @peterhinch if this could be the reason, maybe this is a hint worth.

closed 2023-10-24 by webdeb

peterhinch, 2023-10-24:
There is evident fiesystem corruption which may hint at a bug. I can't see a way that `asyncio` might be a cause, as the code
```py
f = open("file.json", "w")
f.write(json.dumps(data))
f.close()
```
is synchronous. `asyncio` won't get a look-in until after the file is safely closed. I assume you're not using `_thread` as this might complicate things.

Please could you produce a minimal script to reproduce this and we can investigate the cause.

peterhinch, 2023-10-25:
You may be confusing cooperative multi-tasking with pre-emptive. `asyncio` is cooperative. When a piece of synchronous code is running it monopolises the Python runtime. Code can be pre-empted by ISR's and threads, but this applies to all synchronous code. The fact that a piece of code is running as an `asyncio` task does not affect its execution context.

It's a shame you can't reproduce this as filesystem corruption really shouldn't happen,

webdeb, 2023-10-25:
@peterhinch thank you for clarification. You said that ISR could cause the context switching, interesting, because I am using ISR for input handling, maybe thats why I couldn't reproduce the bug in an automated fashion only with asyncio involved 🤔 
I agree, that's a shame that I couldn't reproduce it, but my code is pretty complex, where the bug appeared and a reproduction should be as minimal as possible.

peterhinch, 2023-10-25:
You might be interested in [this doc](https://github.com/peterhinch/micropython-async/blob/master/v3/docs/THREADING.md) which is my attempt to explain context switching - and its implications - in MicroPython. I would definitely suspect ISR woes in your bug. See [this official doc](https://docs.micropython.org/en/latest/reference/isr_rules.html#).

webdeb, 2023-10-25:
Thank you very much @peterhinch! The ISR topic is more complicated than I though (and my implementation pretty naive), I am using it to handle a Rotary Encoder. If you like to look into it, it's here https://github.com/webdeb/nanoshim/blob/hackpwm/src/lib/rotary.py#L24

peterhinch, 2023-10-26:
I suggest you look at [this doc](https://github.com/peterhinch/micropython-async/blob/master/v3/docs/DRIVERS.md#6-quadrature-encoders). Your encoder driver has potential issues, notably the fact that the handlers run in an ISR context. The driver I reference avoids this.

webdeb, 2023-10-26:
@peterhinch got it Peter, I think it will be sufficient to run the actual user handler in a tread safe manner like you did in your encoder implementation and what you described in https://github.com/peterhinch/micropython-samples/blob/master/encoders/ENCODERS.md#5-problem-3-concurrency

PS: Very well written docs btw. 😅

webdeb, 2023-10-27:
@peterhinch I can now confirm, that with the `ThreadSafeFlag` the issue with relative paths also do not appear, so it was exactly this. Thanks again!