micropython/micropython #12695: nlr_jump in MPY 1.21 causing ESP32 exception when catch exception executed in other micropython threads, not main thread REPL

bhcuong2008, 2023-10-15:
nlr_call_jump_callbacks causing this issue. Please refer to my detail issue description here.

### Discussed in https://github.com/orgs/micropython/discussions/12690

<div type='discussions-op-text'>

<sup>Originally posted by **bhcuong2008** October 15, 2023</sup>
I have 3 test cases as below with MicroPython v1.21.0 on ESP32, also in my program.

1. Test case 1, I run this code, and the result as expected.
```
nlr_buf_t nlr;
if(nlr_push(&nlr) == 0) { 
  MP_CMOS_DICT = mp_load_global(MP_QSTR_CMOS);
  nlr_pop();
} else {
  mp_obj_print_exception(&mp_plat_print, (mp_obj_t)nlr.ret_val);
}
```
It prompts `NameError: name 'CMOS' isn't defined` as expected behavior in MicroPython.

2. Test case 2, the code is
```
MP_CMOS_DICT  = mp_const_none;
nlr_buf_t nlr;
if(nlr_push(&nlr) == 0) {
  mp_obj_t mp_fn = mp_load_attr(MP_CMOS_DICT, MP_QSTR_clear);
  mp_call_function_0(mp_fn);
  nlr_pop();
} else {
  mp_obj_print_exception(&mp_plat_print, (mp_obj_t)nlr.ret_val);
}
```
I can't catch python exception in C code. Exception decoder is:
```
0x401b4270: nlr_pop_jump_callback at micropython/py/nlr.c line 63
0x401b4297: nlr_call_jump_callbacks at micropython/py/nlr.c line 79
0x401b42bf: nlr_jump at micropython/py/nlrsetjmp.c line 32
0x401c0e93: mp_raise_msg_varg at micropython/py/runtime.c line 1676
0x401c11c9: mp_load_method at micropython/py/runtime.c line 1233
0x401c11d9: mp_load_attr at micropython/py/runtime.c line 1054
```
It can catch exception, but failed at `nlr_pop_jump_callback , *top = (*top)->prev;`, causing ESP32 exception and restart.

3. If I test in Micropython environment REPL,
```
CMOS = None
CMOS.clear 
```
```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: 'NoneType' object has no attribute 'clear'
```
So, in micropython env REPL, it can catch exception as expected. I think test case 2 must be the same. It must prompt `AttributeError: 'NoneType' object has no attribute 'clear'`, and not causing ESP exception LoadProhibited.

So what's wrong with my C code in test case 2?

Thanks,
Cuong.</div>

stinos, 2023-10-15:
You mention threads in the discussion; how is your C code synchronized so that no other micropython code is running at the same time?

bhcuong2008, 2023-10-16:
In ESP32 port of MicroPython, it synchronizes with thread mutex. So all threads under micropython is run one at a time, not same time. You can try with standard ESP32 port in MicroPython, with my test code in discussion. It will cause ESP32 exception as described.
```
#import uasyncio as asyncio # for 1.16
import asyncio
import _thread

async def _async_cb(params):
   print(params)   
   while True:
      print('Hello ' + str(params))
      await asyncio.sleep(params)

def testThread():
   asyncio.create_task(_async_cb(5))
   asyncio.create_task(_async_cb(10))
   loop = asyncio.get_event_loop()
   loop.run_forever()
 
_thread.start_new_thread(testThread, ())
```

bhcuong2008, 2023-10-16:
@DvdGiessen, I just try your code with simplified one as below, and run in main thread REPL on my build. But not causing the issue.
```
def raises_exception():
    raise RuntimeError('The things! They are wrong!')

def testException():
    try:
        raises_exception()
    except Exception:
        print('We caught it!')
    print('The end.')

>>> testException()
We caught it!
The end.
>>>
```

Could you try your program with code  from MPY 1.16, no call to `nlr_call_jump_callbacks(top)`.
```
void nlr_jump(void *val) {
    nlr_buf_t **top_ptr = &MP_STATE_THREAD(nlr_top);
    nlr_buf_t *top = *top_ptr;
    if (top == NULL) {
        nlr_jump_fail(val);
    }
    top->ret_val = val;
    MP_NLR_RESTORE_PYSTACK(top);
    *top_ptr = top->prev;
    longjmp(top->jmpbuf, 1);
}
```

jimmo, 2023-10-17:
@bhcuong2008 Please see #12711

PR #12711 (2023-10-17): py/modthread.c: Initialise nlr_jump_callback_top on threads.
This should fix #12695 -- @bhcuong2008 could you please test and see if it solves your problem?

---

The main thread gets this because the thread state is in bss, but subsequent threads need this field to be initialised.

Also added a note to mpstate.h to help avoid missing this in the future.

_This work was funded through GitHub Sponsors._
closed 2023-10-17 by bhcuong2008