micropython/micropython #12703: OTA partitions in ESP32-C3

pbdaran, 2023-10-16:
I am trying to enable OTA partitions in ESP32-C3 by changing the custom partition file in sdkconfig file.
But it's not getting enabled. Can you please help?

glenn20, 2023-10-19:
I'm sorry this doesn't answer your question directly, but the [`mp-image-tool-esp32`](https://github.com/glenn20/mp-image-tool-esp32) tool **may** be an alternative way to achieve your goal. It allows you to write an OTA partition table into an ordinary micropython firmware file or esp32 device:

```bash
mp-image-tool-esp32 firmware.bin --table ota
```
- will make a copy of the `firmware.bin` with the partition table changed to an OTA-enabled table.

```bash
mp-image-tool-esp32 a0 --table ota
```
- will overwrite an OTA partition table on the device attached to `/dev/ttyACM0`.

(EDIT: I originally read your message as an S3 device, but I see now you have a C3. I have only tested with ESP32, ESP32S2 and ESP32S3).

glenn20, 2023-10-19:
But, to reply to your actual question, I just tried the following:

First, activate IDF environment. (`source $IDF_DIR/export.sh`).

```bash
cd ports/esp32
# Copy contents of sdkconfig.ota to the end of sdkconfig.c3usb
cat boards/ESP32_GENERIC/sdkconfig.ota >> boards/ESP32_GENERIC_C3/sdkconfig.c3usb
make submodules BOARD=ESP32_GENERIC_C3
make BOARD=ESP32_GENERIC_C3
```

I don't have a C3 device to test the firmware with, but I can confirm that the firmware has an OTA partition table:

```console
$ mp-image-tool-esp32 build-ESP32_GENERIC_C3/firmware.bin 
Opening image file: build-ESP32_GENERIC_C3/firmware.bin... 
Chip type: esp32c3
Flash size: 4MB
Micropython App size: 0x16df40 bytes (1,463 KB)
Partition table (flash size: 4MB):
# Name             Type     SubType      Offset       Size      (End)  Flags
  nvs              data     nvs          0x9000     0x4000     0xd000  0x0  (16.0 kB)
  otadata          data     ota          0xd000     0x2000     0xf000  0x0   (8.0 kB)
  phy_init         data     phy          0xf000     0x1000    0x10000  0x0   (4.0 kB)
  ota_0            app      ota_0       0x10000   0x180000   0x190000  0x0   (1.5 MB)
  ota_1            app      ota_1      0x190000   0x180000   0x310000  0x0   (1.5 MB)
  vfs              data     fat        0x310000    0xf0000   0x400000  0x0   (0.9 MB)
Micropython app fills 95.3% of ota_0 partition (72 kB free)
```

pbdaran, 2023-10-20:
Thank you @glenn20. Will test per procedure you mentioned and update. Since I'm on vacation and not having access to my hardware, will get back on 25 Oct.

pbdaran, 2023-10-25:
Hey @glenn20 ,
`$ mp-image-tool-esp32 build-ESP32_GENERIC_C3/firmware.bin ` didn't work for me. The command didn't create OTA partion.
[image]

Fortunately, `mp-image-tool-esp32 firmware.bin --table ota` worked and it created the `OTA` partition.
Thank you for your help.

glenn20, 2023-10-25:
> Hey @glenn20 ,
> `$ mp-image-tool-esp32 build-ESP32_GENERIC_C3/firmware.bin ` didn't work for me. The command didn't create OTA partion.

> Fortunately, `mp-image-tool-esp32 firmware.bin --table ota` worked and it created the `OTA` partition.
> Thank you for your help.

Glad it helped. 👍

If you check my first message above, I was suggesting to use the `--table ota` option, though I understand how my second message might have confused that.

The second message used the command without `--table` to print out the partition table of the firmware I had compiled - showing that it did have an OTA partition table.