micropython/micropython #12709: Unix: File descriptor initialization error

smurfix, 2023-10-16:
Consider this program.
```
import gc, sys
try:
    with open("/nonexist","r") as f:
        f.read()
except OSError:
    pass
gc.collect()
gc.collect()
sys.stdin.read()
```
The result:
```
Traceback (most recent call last):
  File "/tmp/fuf.py", line 9, in <module>
OSError: [Errno 9] EBADF
```
Apparently file descriptors are initialized with zero instead of -1, causing stdin to be closed when a non-opened FD is cleaned up.

webdeb, 2023-10-19:
I am not a specialist in fs but could this issue be somehow related to this bug? #12689

stinos, 2023-10-19:
> try to access standard input, i.e. file descriptor zero. It shouldn't affect anything that's not stdin.

Unless I'm mistaken in POSIX if you close stdin, then open something else for reading, that gets file descriptor 0. Doen't matter for this fix though, but good to know.

smurfix, 2023-10-19:
> POSIX if you close stdin, then open something else for reading, that gets file descriptor 0

<increasingly-off-topic>

That's correct, which is why you shouldn't ever close file descriptors 0/1/2 in any program that might possibly be running asynchronous or multithreaded code. If you need to reassign any of them, use the `dup2` system call.

</increasingly-off-topic>

webdeb, 2023-10-19:
@smurfix you are right, I also tried to build mp with your fix, but it had no effect on my issue. Thank you anyway!

closed 2025-10-16 by dpgeorge