micropython/micropython #12720: ESP32 - Hardware Timer - higher priority, shorter periods

JohannesMaierhofer, 2023-10-17:
Hello, I know that the Timers for ESP32 are a very deep and complicated issue. Is there any chance to improve the timer behavior in terms of higher reliability in terms of timing accuracy and also allow shorter periods than 1ms? 
Or does one of the experts know a way to circumvent the issue via custom C-Routines that one can write and compile into Micropython for one's specific tasks?

mattytrentini, 2023-10-24:
Are you aware of the proposal to add machine.Counter (#12346)? Will this help your use-case?

JohannesMaierhofer, 2023-11-02:
Thanks for the hint. Unfortunately this does not help me out. I need the output-direction. I need to switch on a pwm-signal (high-frequency) for exact xx.x ms. 
I thought about using the RMT but this does not allow a pwm-pulse. Any further ideas, hints?

JohannesMaierhofer, 2023-11-02:
Can I use the carrier modulation via uPython? 

From espressif docs: 
```
rmt_carrier_config_t tx_carrier_cfg = {
    .duty_cycle = 0.33,                 // duty cycle 33%
    .frequency_hz = 38000,              // 38 KHz
    .flags.polarity_active_low = false, // carrier should be modulated to high level
};
```

mattytrentini, 2024-07-15:
Please let us know how you go @JohannesMaierhofer. I'm going to close this issue since you have work-arounds to look into, but please feel free to re-open it or create new tickets based on your scenarios.

closed 2024-07-15 by mattytrentini