micropython/micropython #12743: Native code emitter: "while True"-loop can not be interrupted.

nspsck, 2023-10-19:
Here is a simple code which is interruptable with the `Ctrl+C` command without the `@micropython.native` decorator. However, if the `@micropython.native` decorator was specified, you can no longer interrupt the code using the 'Ctrl+C' command.

```python
# @micropython.native
def foo():
    while True:
        pass
        
foo()
```

After reading about [the native code emitter and it's limitations](https://docs.micropython.org/en/latest/reference/speed_python.html#the-native-code-emitter), I assume this is a unwanted behavior.

robert-hh, 2023-10-19:
I see it more as an accepted behavior for the sake of speed.

nspsck, 2023-10-19:
If this is indeed a required behavior for the sake of speed, I couldn't agree more with you. To be honest, I can barely think of any important usecases that require this issue to be solved. Hmmm, in that case, should I close this issue and call it a day?

robert-hh, 2023-10-19:
Native and viper code is meant to be short and fast. If you have native code which should be interruptible, you can e,g, add machine.idle() at proper places. idle() usually checks for scheduled code to be executed, and handles keyboard interrupts as well. So the code below can be stopped with Ctrl-C:
```
import machine
@micropython.native
def foo():
    while True:
        idle()
        
foo()
```
Even with pure Python, a tight loop like:
```
while True:
    pass
```
is not recommended. Maybe we can change this topic into a discussion item.

nspsck, 2023-10-19:
Ye, thx for the advice! I am going to close it!

closed 2023-10-19 by nspsck