micropython/micropython #12750: esp32s2 v1.21 network.WLAN(network.STA_IF) RuntimeError: Wifi Unknown Error 0x0101

nubcore, 2023-10-20:
Using a direct cut and paste of the standard WLAN connection `sta_if = network.WLAN(network.STA_IF)` generates an error with the current [v.1.21.0 firmware](https://micropython.org/resources/firmware/ESP32_GENERIC_S2-20231005-v1.21.0.bin). This is consistent on multiple versions of the [ESP32-S2-Saola-1](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/hw-reference/esp32s2/user-guide-saola-1-v1.2.html). The same error occurs on the latest _ESP32_GENERIC_S2-20231019-v1.22.0-preview.32.g86c7b957a.bin_  build. Reverting to v1.20.0 resolved the error and Wifi is functional.

Further device details per _esptool.py v4.6.2_

Detecting chip type... ESP32-S2
Chip is ESP32-S2 (revision v0.0)
Features: WiFi, No Embedded Flash, No Embedded PSRAM, ADC and temperature sensor calibration in BLK2 of efuse V1
Crystal is 40MHz
Manufacturer: 20
Device: 4016
Detected flash size: 4MB
Flash type set in eFuse: quad (4 data lines)

AnSaNet, 2023-10-27:
The same error is generates with https://micropython.org/resources/firmware/LOLIN_S2_MINI-20231005-v1.21.0.bin.
This is consistent on the https://www.wemos.cc/en/latest/s2/s2_mini.html. Reverting to v1.20.0 resolved the error and Wifi is functional.

fabse-hack, 2023-11-01:
yes, here the same bug with ESP32 S2 ....

```python
>>> import network
>>> network.WLAN(network.STA_IF)
E (666693) esp_netif_lwip: esp_netif_new: Failed to configure netif with config=0x3ffd8a50 (config or if_key is NULL or duplicate key)
E (666693) esp_netif_lwip: esp_netif_new: Failed to configure netif with config=0x3ffd8a50 (config or if_key is NULL or duplicate key)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
RuntimeError: Wifi Unknown Error 0x0101```

jimmo, 2023-11-08:
I was able to repro this on my S2 board by lifting the CS pin on the spiflash chip.

Previously this worked fine, now gives the same output as reported above.

```
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe6108,len:0xf34
load:0x4004c000,len:0x8ec
load:0x40050000,len:0x2f18
entry 0x4004c13c
E (316) quad_psram: PSRAM ID read error: 0x00ffffff, PSRAM chip not found or not supported
E (316) esp_psram: PSRAM enabled but initialization failed. Bailing out.
MicroPython v1.22.0-preview.132.g958c6d917d on 2023-11-08; Generic ESP32S2 module with ESP32S2
Type "help()" for more information.
>>> import network
>>> sta = network.WLAN(network.STA_IF)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
RuntimeError: Wifi Unknown Error 0x0101
```

jimmo, 2023-11-09:
Still investigating, but also found a similar issue in the IDF here: https://github.com/espressif/esp-idf/issues/11971

projectgus, 2023-11-14:
Thanks everyone for your patience while we pondered this together. The PR linked above should fix it.

closed 2023-11-23 by dpgeorge

ds-digid, 2024-01-09:
I experiance this issue still with MicroPython v1.22.1 on 2024-01-05; LILYGO TTGO LoRa32 with ESP32.
Will this be fixed later or is there a way to workaround this issue?

dpgeorge, 2024-01-09:
@ds-digid this issue should be fixed now.  Note that the original bug report and fix is for ESP32-S2.  A standard ESP32 shouldn't show this problem.

Can you please post the exact code you used to trigger this problem?  Also which exact firmware do you run?

ds-digid, 2024-01-09:
The code to trigger this problem:
self.wl = network.WLAN(network.STA_IF)

Firmware:
https://micropython.org/resources/firmware/LILYGO_TTGO_LORA32-20240105-v1.22.1.bin

dpgeorge, 2024-01-10:
I don't have a LILYGO TTGO, but running that same firmware on a standard ESP32 it works just fine.  I can start and use the WLAN interface via:
```py
import network
wl = network.WLAN(network.STA_IF)
wl.active(1)
wl.scan()
```
You must have some additional code that's running, eg in `boot.py` or `main.py`.

ds-digid, 2024-01-30:
Here is some more information gathered with the esp32.idf_heap_info() function:

stack: 2080 out of 15360
GC: total: 165952, used: 80592, free: 85360, max new split: 14336
 No. of 1-blocks: 958, 2-blocks: 279, max blk sz: 282, max free sz: 2905
Heap_data: [(240, 4, 0, 4), (7288, 4, 0, 4), (16648, 1888, 1792, 1744), (87584, 9388, 9216, 9388), (15072, 14656, 14336, 14656), (113840, 304, 288, 304)]
Heap_exec: [(15072, 14656, 14336, 14656), (113840, 304, 288, 304), (1580, 1248, 1152, 1232)]
Mem info: None
Mem free: 99696
E (4230) wifi:Expected to init 10 rx buffer, actual is 7
Traceback (most recent call last):
  File "main.py", line 139, in <module>
  File "asyncio/core.py", line 1, in run
  File "asyncio/core.py", line 1, in run_until_complete
  File "asyncio/core.py", line 1, in run_until_complete
  File "main.py", line 83, in main
  File "communication/microwifimanager/wifimanager.py", line 144, in __init__
RuntimeError: Wifi Unknown Error 0x0101
MicroPython v1.22.1 on 2024-01-05; LILYGO TTGO LoRa32 with ESP32

projectgus, 2024-01-30:
> The program I am developing has a significant amout of submodules which are loaded on startup. (Many imports and after each importblock in a .py file I call the garbage collector to save memory).

@ds-digid It looks like MicroPython is expanding the Python heap to cover Python's requirements, and then there isn't enough RAM left outside of Python to initialise Wi-Fi. If you move Wi-Fi initialisation as early as possible in the startup sequence, does this work?

projectgus, 2024-02-01:
I think there is enough memory, but memory availability will depend on the sequence of events that happen in your app. It will depend heavily on factors such as whether you are loading .py files or pre-compiled .mpy files or frozen bytecode. Without more details of your app, it's hard to tell.

If you are able to share your app, or enough of your app to reproduce the problem, then suggest opening a new Discussion to get feedback on how to optimise the memory usage. This is a different matter to the bug reported here.

PR #12968 (2023-11-23): esp32/network: Use non-PSRAM Wi-Fi settings if PSRAM fails to initialise
In ESP-IDF, enabling SPIRAM in menuconfig sets a Kconfig option called [Wi-Fi Cache TX Buffers](https://docs.espressif.com/projects/esp-idf/en/v5.0.2/esp32s2/api-reference/kconfig.html#config-esp32-wifi-cache-tx-buffer-num) and switches the [default TX Buffer type to Dynamic](https://docs.espressif.com/projects/esp-idf/en/v5.0.2/esp32s2/api-reference/kconfig.html#config-esp32-wifi-tx-buffer). These settings use 54,528 bytes of heap at runtime.

Not a problem if PSRAM is available, but it's quite a lot of RAM to use up if  [... cut]