micropython/micropython #12767: micropython doesn't reply to ARP requests

yshui, 2023-10-22:
firmware info: `MicroPython 856e08b1931b88271816a2f60648f6ff332235b2, inky_frame v1.20.3 on 2023-06-19; Raspberry Pi Pico W with RP2040`

I am having problem with making http requests from pico, tcpdump on my router shows the board is not replying to ARP requests:

```
06:49:18.250980 IP 192.168.1.241.38952 > 1.1.1.1.53: 50883+ A? pool.ntp.org. (30)
06:49:18.279499 IP 1.1.1.1.53 > 192.168.1.241.38952: 50883 4/0/0 A 213.5.132.231, A 139.143.5.30, A 85.199.214.101, A 85.199.214.102 (94)
06:49:18.287255 IP 192.168.1.241.51056 > 213.5.132.231.123: NTPv3, Client, length 48
06:49:18.335666 IP 213.5.132.231.123 > 192.168.1.241.51056: NTPv3, Server, length 48
06:49:23.192697 ARP, Request who-has 192.168.1.241 tell 192.168.1.1, length 28
06:49:24.232697 ARP, Request who-has 192.168.1.241 tell 192.168.1.1, length 28
06:49:25.272813 ARP, Request who-has 192.168.1.241 tell 192.168.1.1, length 28
```

yshui, 2023-10-22:
I don't actually know if micropython is responsible for ARPs, or if the pico SDK is? let me know if I opened this issue at the wrong project.

yshui, 2023-10-22:
This seems to be an interactive with network and `_thread`.

I tried manually `deinit` the network and reconnect, `urlopen` worked fine right after that. then I tried calling urlopen from a background thread, which fails.

dpgeorge, 2023-10-22:
You can enable a trace of ETH packets on the Pico W using `.config(trace=15)`.  Eg:
```
>>> wl=network.WLAN()
>>> wl.config(trace=15)
>>> wl.active(1)
>>> wl.connect(<ssid>, <password>)
```
That should show all incoming and outgoing packets from the point of view of the Pico W, and help to debug the issue.

bc-109, 2024-02-21:
Hi,

Same problem here 
`MicroPython v1.20.0 on 2023-04-26; Raspberry Pi Pico W with RP2040`

The Pico does not answer to ARP requests. 

My Pico has address 192.168.1.144 obtained by DHCP. I'm trying to ping 1.1.1.1. 
- On the Pico, when using `trace=15`, I can see the outgoing ICMP ECHO REQUESTS, but nothing else. 
- On my router, I see the ECHO REQUEST from the Pico, the ECHO REPLY from 1.1.1.1,then my router sends ARP requests ('Who has address 192.168.1.144), but the Pico does not answer.
- Of course, my router never sents back the ECHO REPLY, then my ping remains in "Tilmeout - No answer" on the Pico.

In some previous tests, a few days ago, in some rare cases, I remember I obtained exactly one answer to the first ICMP packet, and no more answer after that. Today, I didn't manage to reproduce that.

Also, I don't know who is responsible about answering to ARP and/or ICMP. So I have no idea about where to look at.

bc-109, 2024-02-21:
Addendum :
- ARP, and ping, works perfectly after hardware reset
- ARP answer does not answer correctly after software reboot (which is done just after uploading new software).

Then, it may have something to do with network initialization, and what survives a software reboot. I'll dig further.

yshui, 2024-02-23:
@bc-109 do you use `_thread` at all?

bc-109, 2024-02-27:
Hi @yshui;  no, I never used `_thead`. My test code was very basic : initializing network and pinging a host.

Here's what I found :
- after cold boot or hardware reset, the network interface is (obviously) down
- after software reboot, if the network interface was up, it remains up; then, if you set it up a second time, this causes the odd ARP behavior

I solved the problem in my network initialization code, by always de-activating the interface before activating it :
`iface.active(False)`
`iface.active(True)`
`iface.connect(SSID, PASSWORD)`

Doing so, it solves my problem.  ARP always work as expected. Hope this helps...

jonnor, 2024-08-24:
Thank you for debugging the issue and documenting the fix/workaround that you found.

guyandrewdawson, 2024-12-22:
Hi

I am experiencing the same symptoms (no ARP response) with the code for the Waveshare Nixie clock, 

The original code came from https://github.com/michaelmargolis/lcd_ntp_nixie_clock.git

Adding the code suggested by bc-109 does not fix the problem.

Any ideas

> Hi @yshui; no, I never used `_thead`. My test code was very basic : initializing network and pinging a host.
>
> Here's what I found :
>
>     * after cold boot or hardware reset, the network interface is [...]

jordanarpad, 2025-05-01:
I recently started to experience that my Pico W is loosing connection more often. I did experience similar behaviour previously also but much less frequently. I upgraded to latest MicroPython v1.25.0 on 2025-04-15; Raspberry Pi Pico W with RP2040 but it kept loosing connection.
Then I increased the ARP timeout on my router to 300 and it started to work more reliably again.
What I observe now is that Pico W is sending out a gratitious ARP when obtaining an IP address and my router is using unicast ARP request to proactively refresh its ARP cache but the timeout.
By deleting the ARP cache for IP of Pico W on the router I can break the connection because for some reason I do not see the broadcast ARP request coming up on the Pico W with trace=15 config.
So maybe this happened to me and some others in this issue that Pico W never got the broadcast ARP request. The wifi access point probably has an important role how to send broadcast ARP requests to the clients connected to it.

guyandrewdawson, 2025-05-02:
Interesting...

Although my problem is with connecting to the Pico (IE when another device needs the IP address of the PIco), reconfiguring the Pico to use a different SSID, and thereby a different DHCP, does seem to improve matters. It looks like the Pico is overly sensitive to the network infrastructure. 

Time to revisit the RFC's

jordanarpad, 2025-05-03:
I concur that Pico W wifi behaviour with micropython is dependant on the wifi access point indeed.

I spent some time trying to reproduce my issue with Raspberry Pi 4 and Zero 2 W both as an access point and my Pico W does not ping reliably behind Pi 4 as an access point but ping works well with Pi Zero 2 W as an access point. I used exactly the same Network Manager configuration for wifi access point on both, starting from a freshly flashed Pi OS release on both.

guyandrewdawson, 2025-05-06:
I'm now certain the issue was with the router rather than the Pico.  While I was attempting to set up an isolated subnet in order to make things clearer (pointless, as windows and it's associated cruft will "spam" any network interface it sees), I must have tweaked something on the router which solved the problem. No idea exactly what, I'm afraid to say.