micropython/micropython #12773: SoftSPI v1.21 unreliable

emard, 2023-10-22:
I use SoftSPI as JTAG accelerator for flash read and write
on my project https://github.com/emard/esp32ecp5

micropython v1.14 SoftSPI works well
micropython v1.21 SoftSPI is unrealiable

softspi is initialized at spi_freq = 20000000 (20 MHz)
swspi=SoftSPI(baudrate=spi_freq, polarity=1, phase=1, bits=8, firstbit=SPI.MSB, sck=Pin(jtagpin.tck), mosi=Pin(jtagpin.tdi), miso=Pin(jtagpin.tdo))

To read flash content, softspi is called to read 256 bytes into bytearray
swspi.readinto(data) # retrieve whole block

usually there is at least 1 wrong bit in 256 byte block. Wrong bit can be
0 instead of 1 and 1 instead of 0. I tried lowering clock speed and all
4 combinations of polarity=0/1 phase=0/1 and either nothing works
or it works with error bits

this code is used for testing and reporting errors

import ecp5
import time

def run():
  er = 0
  d1 = ecp5.flashrd(addr=0x300000, length=256)
  for i in range(100):
    d2 = ecp5.flashrd(addr=0x300000, length=256)
    if d2 != d1:
      er = 1
      break
  time.sleep(1)
  if er:
    print("read error")
    diff = [(i, d1[i], d2[i]) for i in range(len(d1)) if d1[i] != d2[i]]
    print(d1)
    print(d2)
    print(diff)
  else:
    print("ok")
    print(d1)

dpgeorge, 2024-07-12:
The software SPI driver has been essentially unchanged since v1.14 (or even earlier).  And it's a very simple protocol that doesn't really depend on timing (due to the controller driving the clock signal).

So it's hard to think how it could be unreliable.  Maybe it's a change to the IDF version??

Is this still a problem?  To make progress here would require full working code that shows the error, so others can reproduce it.

projectgus, 2024-08-27:
Closing as needs more info. @emard if you're still trying to figure this out then as well as reproducer code some logic analyser or scope traces that include an error bit might be useful. It sounds more like a signal integrity issue than a bug to me, although hard to explain why the older MicroPython doesn't also see it.

closed 2024-08-27 by projectgus

hstarmans, 2025-01-20:
I can confirm this issue, I have a FPGA hooked up to the spi, which replies every word I sent.
If i sent array of bytes with size 10 and do this 10_000 times, an error is produced.