micropython/micropython #12776: Crashes while pushing to a TaskQueue instance

gwangmu, 2023-10-23:
# Description

The UNIX port of MicroPython crashes while attempting to push an object to `asyncio.TaskQueue()` (or the `waiting` member of `asyncio.Event()`. It generally exhibited null-dereference when the passed object was a value and global-buffer-overflow when it was a module. We have attached three PoCs of each bug type.

[poc.zip](https://github.com/micropython/micropython/files/13069498/poc.zip)

# Proof of Concept

```
$ # build unix port with ASAN, at the root source code directory.
$ export CC=clang
$ export CXX=clang++
$ export CFLAGS="-fsanitize=address -fno-omit-frame-pointer"
$ export CXXFLAGS=$CFLAGS
$ export LDFLAGS=$CFLAGS
$ export DEBUG=1
$ make -C mpy-cross -j
$ make -C ports/unix -j all lib
$
$ # run a poc.
$ export ASAN_OPTIONS="detect_leaks=0"
$ ./ports/unix/build-standard/micropython <poc_file>
```

# Environment

Ubuntu 20.04
Intel(R) Xeon(R) Gold 5218 CPU @ 2.30GHz
Memory: 64 GB

# Affected Version

v1.20.0 (commit a3862e726, latest as of 2023-09-26)
v1.20.0 (commit 813d559bc, 2023-06-19)
Discovered in the UNIX port version.

peterhinch, 2023-10-24:
I'm puzzled by this. As far as I'm aware `asyncio.TaskQueue` and `Event.waiting` are not part of the public API (in CPython or MicroPython).

gwangmu, 2023-10-24:
@peterhinch I wasn't aware of that. Do you think `asyncio.TaskQueue` or `Event.waiting` must not have been accessible to general users in the first place?

peterhinch, 2023-10-25:
Under CPython 3.10.12:
```py
>>> 'TaskQueue' in dir(asyncio)
False
>>> 'waiting' in dir(asyncio.Event)
False
```
Unfortunately Python does not have a way of enforcing method privacy, but the intention in MicroPython seems clear: these are not part of the public API.

jonnor, 2024-08-24:
Considering the non-public nature of these APIs, is this a wont-fix?

dpgeorge, 2024-08-26:
Indeed, this is not part of the public Python API.

We could add type checking to `push()`, but it would add code size and slow things down a little.

Alternatively, we could rename it to `_TaskQueue` to indicate that it's a private class.

Or, not change anything.

gwangmu, 2024-09-04:
Thank you for considering this issue. My small opinion is to at least prevent global-buffer-overflow by adding a primitive `if` check (e.g., is the passed object a module?) because if someone with ill intention discovered this and tries to exploit this, it can damage both the user's system and the reputation of MicroPython overall. But I'm not well-informed about the policy of MicroPython :sweat_smile: