micropython/micropython #12817: Unable to set hostname on esp32 for LAN DHCP request

harbaum, 2023-10-27:
I am trying to set the hostname for DHCP requests on a ESP32 with µP 1.21.0. I am trying it in all kinds of ways I can imagine:

```
import network
import machine

network.hostname("name1")

# setup network
lan = network.LAN(mdc=machine.Pin(23),mdio=machine.Pin(18),power=machine.Pin(16),id=0,phy_addr=1,phy_type=network.PHY_LAN8720)

network.hostname("name1")
lan.config(hostname="name2", dhcp_hostname="name3")

lan.active(1)

network.hostname("name1")
lan.config(hostname="name2", dhcp_hostname="name3")

lan.connect()

network.hostname("name1")
lan.config(hostname="name2", dhcp_hostname="name3")
```

But nothing of this seems to have any effect and tcpdump still sees this:

```
09:11:50.884258 IP (tos 0x0, ttl 255, id 24, offset 0, flags [none], proto UDP (17), length 336)
    0.0.0.0.68 > 255.255.255.255.67: [udp sum ok] BOOTP/DHCP, Request from ec:62:60:a2:7f:0f, length 308, xid 0x11b7830c, Flags [none] (0x0000)
          Client-Ethernet-Address ec:62:60:a2:7f:0f
          Vendor-rfc1048 Extensions
            Magic Cookie 0x63825363
            DHCP-Message (53), length 1: Discover
            MSZ (57), length 2: 1500
            Hostname (12), length 9: "espressif"
            Parameter-Request (55), length 4: 
              Subnet-Mask (1), Default-Gateway (3), BR (28), Domain-Name-Server (6)
            Client-ID (61), length 7: ether ec:62:60:a2:7f:0f
            END (255), length 0
            PAD (0), length 0, occurs 34
```

How can I change the hostname for LAN DHCP requests?

robert-hh, 2023-10-27:
This is indeed an omission in the code. PR https://github.com/micropython/micropython/pull/12736 should fix it.

harbaum, 2023-10-27:
Great, thanks. I'll give that PR a try ...

dpgeorge, 2023-10-31:
Should be fixed by bbbd484f5ba27f31c142a6402662422c0634e271.  You'll need to call `network.hostname(...)` before `lan.active(1)`.

harbaum, 2023-11-09:
Works like a charm! Thank you!

closed 2023-11-09 by harbaum