micropython/micropython #12826: ESPRESSIF-ESP32-S2-WROOM

fabse-hack, 2023-10-29:
Hello,

when i want to flash my EP32 S2 WROOM with:
ESP32_GENERIC_S2 Firmware,

then i get this error:

```python
MicroPython v1.21.0 on 2023-10-06; Generic ESP32S2 module with ESP32S2
Type "help()" for more information.
>>> ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x1 (POWERON),boot:0xb (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe6108,len:0xf34
load:0x4004c000,len:0x8ec
load:0x40050000,len:0x2f18
entry 0x4004c13c
E (303) quad_psram: PSRAM ID read error: 0x00ffffff, PSRAM chip not found or not supported
E (304) esp_psram: PSRAM enabled but initialization failed. Bailing out.
```

With this firmware is everything fine: ESP32_S2_WROVER-20220618-v1.19.1.bin
but i not found the v1.21 from it ... i only found is discussion here in github, that i have to use the generic (but there is this error above)

fabse-hack, 2023-10-29:
i think is a problem with flashing ... i use the espressif flash tool: https://www.espressif.com/en/support/download/other-tools
and @ 0x1000 ... i try 0x0000 and 0x10000 too. There is the problem.

nubcore, 2023-10-30:
I don't know what to make of a ESP32-S2-WOOR, listing typo I guess. I see from the picture though the ESP32-S2-WROOM module. Interesting error though that it's trying to get a PSRAM ID as those modules don't have them.

Are you flashing with esptool, and are you doing an erase and then write? It would be interesting to see what it reports identifying for your chip. My S2 WROVERs look like this, note it's not detecting and PSRAM.

`Detecting chip type... ESP32-S2                                                                                         Chip is ESP32-S2 (revision v0.0)                                                                                        Features: WiFi, No Embedded Flash, No Embedded PSRAM, ADC and temperature sensor calibration in BLK2 of efuse V1        Crystal is 40MHz                                                                                                        MAC: 7c:df:a1:00:8e:94                                                                                                  Uploading stub...                                                                                                       Running stub...                                                                                                         Stub running...                                                                                                         Manufacturer: c8                                                                                                        Device: 4016                                                                                                            Detected flash size: 4MB                                                                                                Flash type set in eFuse: quad (4 data lines)  `

I've been able to run v1.21 fine on a S2 WROVER, I don't have a WROOM to try.

I also found what looks like a bug and not sure which S2 chips might be affected regarding WiFi on latest builds #12750

fabse-hack, 2023-10-31:
@nubcore thank you! i testing now much with ubuntu, There is a driver error. I think is something about the ch340 / ch341se. I think is a driver problem in windows too. But thank you for the 0x1000 ,,, this helps me already :)
- so is not a micropython issue - is an microcontroller issue xD
//closed...

closed 2023-10-31 by fabse-hack