micropython/micropython #12833: Not able to update from 1.19.1 to 1.20.0 in Fedora due to failing build.

stratakis, 2023-10-30:
Hello,

When trying to update the micropython package on Fedora with our [current SPEC file](https://src.fedoraproject.org/rpms/micropython/blob/c47fecf9750eb11436544a6c0c02ab65ad216ee1/f/micropython.spec)  and by removing the relevant axtls bits and adding the mbedtls as a build requirement I get  an error `Error: micropython-lib submodule is not initialized. Run 'make submodules'`

If I add that bit I get `fatal: not a git repository (or any of the parent directories): .git`

Any ideas on how to proceed here? 

Current SPEC file changes: https://src.fedoraproject.org/rpms/micropython/pull-request/21#request_diff

Full build log is also attached.
[build.log](https://github.com/micropython/micropython/files/13209713/build.log)

dpgeorge, 2023-10-30:
You'll need to explicitly fetch the `micropython-lib` repo and copy it into `lib/micropython-lib`.  See how the SPEC file currently handles axtls and do the same thing for micropython-lib.

Also, you can probably remove a lot of the .patch files applied in the SPEC, I think all those issues have been resolved here.

fermented-beaver, 2023-11-09:
The issue has been resolved and I've managed to update the package: https://src.fedoraproject.org/rpms/micropython/pull-request/21#request_diff

Thank you for the pointers!

stratakis, 2023-11-09:
The issue has been resolved and I managed to update the package: https://src.fedoraproject.org/rpms/micropython/pull-request/21#request_diff

Thank you for the pointers!

closed 2023-11-09 by stratakis