micropython/micropython #12834: REPL Buffer Overflow

moefear85, 2023-10-30:
I just wanted to point out a long outstanding issue with the repl, at least for the esp32 & esp8266. It still exists as of v1.21. I'm surprised it hasn't been brought up or resolved by anyone else yet. The repl seems to have a limit of 50 characters that one can send at a time (terminated by newline), without corruption of the data occurring. Above 50 characters (it is about 10 characters for the esp8266), sporadically, especially when sending fast (such as hexlified file transfer), the segment of data (terminated by newline), will get corrupted, and thus unhexlify at the repl will fail, detecting an illegal character, specifically a double quote.

Initially I just assumed a noisy serial line, despite this being the default baudrate of 115200. Upon closer inspection, I realized it's a string, enclosed in double quotes, that ends up near the 50 byte mark in the stream after transfer. Something is overwriting into the buffer into which the repl input goes, and its happening in the chip, not on the PC, nor on the serial line, because the reference in the quotes was human readable (I think it was some QSTR), it wasn't random. This was a few months back. I don't remember which string it is, but I know it still fails if I transfer more than 50 bytes at a time. I can double check if there is interest in the issue.

dpgeorge, 2023-10-30:
Please can you give steps to reproduce the issue.  State the hardware used, the version of MicroPython (via `sys.implementation`), the PC program you use to communicate with the board, and the steps taken.

moefear85, 2023-10-31:
ok will do 2morrow

moefear85, 2023-10-31:
OK...

So using ESP32_GENERIC-20231005-v1.21.0.bin on an esp32-wroom-32 rev. 1.0

So as a preliminary test script, see the attachment. There are 3 variants of the payload variable, each a different length (71, 72, & 149 bytes). Uncomment one and run the script.
At 71, there are no errors, no matter how long you run the script. I've tested well into the 10000's range several times. At 72, the errors begin, albeit very rare (~2 per 1000 iterations). At 149, the errors are excessive, and you probably won't get past 50 iterations. The fact that the 71/72 mark is so pivotal in the errors, in my opinion rules out serial noise. If it were random noise, it would have to happen at practically any length, since that mainly affects how interleaved the commands and responses are, not the average throughput.
The hanging might be due to changes in the payload seen by the mpy that put it in a mode that expects further input, so it doesn't return `>>>`, which the test script waits on before sending the next iteration. But most likely the mpy is crashing. I wrote this fast so I'm not sure which. As for the errors themselves, that might be attributable to filling up of the buffers

Either way the exhibited behavior is problematic no matter what the reason, because many people run mpy with automated scripts that talk to it, there is no human there to detect that the repl is expecting more input or whether it has crashed or not. It makes file uploading difficult/slow for anything equal or larger than average-sized scripts of 1000 bytes, unless they're divided into 50 byte segments (to be safe).

The errors seen in the response are in random places, and appear even if unhexlify does not detect any error, which is also problematic, but only happens at the 71/72 mark, which also is unlikely due to noise. 115200 baudrate isn't that difficult to achieve cleanly with modern hardware. Anyways, when errors do occur, its either odd-length, or illegal character. That's not surprising, since there are only these 3 possible ways that unhexlified stream can be erroneous.

You might say the uart buffer is simply overflowing. Nonetheless 71/72 doesn't seem inline with any usual buffer size. Moreover, I'm unaware if its documented anywhere, what the limit is (per board?) that one can send before having to wait for the prompt. Scripts need to know this limit, and mpy ideally would handle this scenario gracefully, sending a specific easy-to-parse response indicating that its uart buffer has overflowed.

Nonetheless, I didn't expect the exact issue to be unreproducible with this test, so I left it to the end of the day. I see it's not enough. From the past I clearly remember the errors in the stream were unexpected strings and quotes that repeat in same specific positions repeatedly. It was with a file synchronization tool I wrote. I can choose the chunk size in which the files are sent to any arbitrary value. I chose 50 in the end because its reliable. 100 or above is simply unusable, because it verifies the sent data, and retries if it detects an error, and it becomes practically impossible to upload any file larger than 700 bytes. I don't know if its in my repos... I'll try again using it and uploading some larger files (1.5Kb-10Kb), and see what happens.

I'll do it tomorrow if I can, or when I have time. Right now I'm writing some modules for micropython for this project: [micropython_with_espnow_camera](https://github.com/moefear85/micropython_with_espnow_camera). I keep running into issues that are consuming alot of time, but making progress.

jimmo, 2023-11-01:
@moefear85 The ESP32 has a 128 byte hardware fifo on the uart. This feeds into a 260 byte ring buffer for stdin. It doesn't seem unreasonable that either of these (but especially the stdin ringbuf) could overflow when doing large writes with no flow control. For example the stdin ringbuf could overflow while a garbage collection was in progress.

moefear85, 2023-11-01:
@jimmo I have nothing against that, but I leave it to the reader (who decides to try the script), as an exercise, to determine why that can't be the case here. Either way, such overflows can even happen with hand-typed commands, and can lead to unintended consequences if the reader/parser doesn't detect such overflows and refuse to execute the command and raise an InputOverflowException instead. Also, 71 != 128.

moefear85, 2023-11-01:
and I'm out

closed 2023-11-01 by moefear85

dpgeorge, 2023-11-01:
@moefear85 thanks for the test script and info on how to reproduce it.

I have used ESP32_GENERIC-20231005-v1.21.0.bin firmware on a TinyPICO (standard ESP32, no SPIRAM).

Running your script on an Arch Linux machine, with length 149 input, I see no errors at all.  I can get up to 10000 iterations with zero errors in 10 minutes.

What operating system is your PC running?  That may be the difference here.