micropython/micropython #12838: Unix port: Error when building 1.21.0 for riscv64

jefferyto, 2023-10-31:
I'm updating the micropython package in OpenWrt (https://github.com/openwrt/packages/pull/22564) and I'm getting a build error when compiling for riscv64:

```
CC main.c
ccache riscv64-openwrt-linux-musl-gcc -Os -pipe -mabi=lp64d -march=rv64imafdc -fno-caller-saves -fno-plt -fhonour-copts -ffile-prefix-map=/home/temp/downloads/openwrt/testing/sifiveu-generic/build_dir/target-riscv64_riscv64_musl/micropython-mbedtls/micropython-1.21.0=micropython-1.21.0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/home/temp/downloads/openwrt/testing/sifiveu-generic/staging_dir/toolchain-riscv64_riscv64_gcc-12.3.0_musl/usr/include -I/home/temp/downloads/openwrt/testing/sifiveu-generic/staging_dir/toolchain-riscv64_riscv64_gcc-12.3.0_musl/include -I/home/temp/downloads/openwrt/testing/sifiveu-generic/staging_dir/toolchain-riscv64_riscv64_gcc-12.3.0_musl/include/fortify  -DMICROPY_VFS_FAT=1 -DMICROPY_VFS_LFS1=1 -DMICROPY_VFS_LFS2=1 -DMICROPY_PY_SSL=1 -DMICROPY_SSL_MBEDTLS=1 -DMICROPY_PY_BTREE=1 -DFFCONF_H=\"lib/oofatfs/ffconf.h\" -DLFS1_NO_MALLOC -DLFS1_NO_DEBUG -DLFS1_NO_WARN -DLFS1_NO_ERROR -DLFS1_NO_ASSERT -DLFS2_NO_MALLOC -DLFS2_NO_DEBUG -DLFS2_NO_WARN -DLFS2_NO_ERROR -DLFS2_NO_ASSERT -I../../lib/berkeley-db-1.xx/PORT/include -I. -I../.. -Ibuild-standard -I../../shared/readline -Wall -Werror -Wextra -Wno-unused-parameter -Wpointer-arith -Wdouble-promotion -Wfloat-conversion -std=gnu99 -DUNIX -Os -DNDEBUG -fdata-sections -ffunction-sections -Ivariants/standard  -g -U _FORTIFY_SOURCE -DMICROPY_USE_READLINE=1 -DMICROPY_PY_TERMIOS=1 -DMICROPY_PY_SOCKET=1 -DMICROPY_PY_THREAD=1 -DMICROPY_PY_THREAD_GIL=0 -I/home/temp/downloads/openwrt/testing/sifiveu-generic/staging_dir/target-riscv64_riscv64_musl/usr/include -DMICROPY_PY_FFI=1 -DMPZ_DIG_SIZE=16  -DMICROPY_ROM_TEXT_COMPRESSION=1 -DMICROPY_QSTR_EXTRA_POOL=mp_qstr_frozen_const_pool -DMICROPY_MODULE_FROZEN_MPY -DMICROPY_MODULE_FROZEN_STR -c -MD -o build-standard/main.o main.c || (echo -e "See \033[1;31mhttps://github.com/micropython/micropython/wiki/Build-Troubleshooting\033[0m"; false)
main.c: In function 'main_':
main.c:564:14: error: variable 'path_remaining' might be clobbered by 'longjmp' or 'vfork' [-Werror=clobbered]
  564 |         bool path_remaining = *path;
      |              ^~~~~~~~~~~~~~
cc1: all warnings being treated as errors
-e See [1;31mhttps://github.com/micropython/micropython/wiki/Build-Troubleshooting[0m
make[3]: *** [../../py/mkrules.mk:83: build-standard/main.o] Error 1
make[3]: Leaving directory '/home/temp/downloads/openwrt/testing/sifiveu-generic/build_dir/target-riscv64_riscv64_musl/micropython-mbedtls/micropython-1.21.0/ports/unix'
make[2]: *** [Makefile:119: /home/temp/downloads/openwrt/testing/sifiveu-generic/build_dir/target-riscv64_riscv64_musl/micropython-mbedtls/micropython-1.21.0/.built] Error 2
make[2]: Leaving directory '/home/jeff/Code/personal/openwrt/packages/master/lang/python/micropython'
time: package/feeds/packages/micropython/mbedtls/compile#1.69#0.56#2.60
```

This is with a [snapshot sdk](https://downloads.openwrt.org/snapshots/targets/sifiveu/generic/openwrt-sdk-sifiveu-generic_gcc-12.3.0_musl.Linux-x86_64.tar.xz) for sifiveu-generic (the only riscv64 OpenWrt target so far) with gcc 12.3.0. My build system is Ubuntu 23.04.

It seems this error only comes up for riscv64; arm/aarch64, mips, powerpc, x86 all appear to compile with no issues.

dpgeorge, 2023-10-31:
Thanks for the report, we'll have to look in to that to try and fix it.

In the meantime I suggest building with this warning disabled, like this:
```
$ make CFLAGS_EXTRA=-Wno-error=clobbered
```

jefferyto, 2023-11-01:
> In the meantime I suggest building with this warning disabled, like this:
> ```
> $ make CFLAGS_EXTRA=-Wno-error=clobbered
> ```

Thanks, that works for now :smile:

dpgeorge, 2024-07-23:
Fixed by 44527ada5f120f5f131bcd759e86e4e5323555aa

closed 2024-07-23 by dpgeorge