micropython/micropython #12842: esp32: UDP sendto causes memory leaks/ENOMEM error for most IP addresses

tangentmonger, 2023-10-31:
I'm trying to send UDP packets using socket.sendto(). I would expect this to succeed regardless of the destination IP address, but for most addresses this fails after exactly four packets with OSError ENOMEM. I do not see this exception when sending to the board's own IP or the broadcast address.

Hardware: ESP32 on an Olimex ESP32-PoE-ISO board, connected to a switch that has nothing else connected to it (as a minimal test setup).

ESPtool identifies the chip as  ESP32D0WDQ5 (revision 3).

Firmware: OLIMEX_ESP32_POE-20231005-v1.21.0.bin

Version in REPL: MicroPython v1.21.0 on 2023-10-06; Olimex ESP32 ETH with ESP32

#4029 is a similar issue but uses WiFi and instead affects broadcast only. I'm seeing the opposite behavior.

This code example attempts to send 100 packets to IPs from 192.168.4.0 through to 192.168.4.255. This succeeds for IPs 192.168.4.250 (the board's own IP) and 192.168.4.255 (broadcast) but fails for every other address after the fourth attempt with OSError 12, ENOMEM.

```
import socket
import network
import machine
import time

lan = network.LAN(mdc=machine.Pin(23), mdio=machine.Pin(18), power=machine.Pin(12),
                  phy_type=network.PHY_LAN8720, phy_addr=0)
try:
    lan.active(True)
except OSError as ex:
    if ex.value == "ethernet enable failed":
        print("Failed to enable ethernet - waiting a few seconds and doing a hardware reset.")
        time.sleep(2)
        machine.reset()
    raise

my_ip = '192.168.4.250'
lan.ifconfig((my_ip, '255.255.255.0', '192.168.4.1', '192.168.4.1'))

sock1 = socket.socket(socket.AF_INET, socket.SOCK_DGRAM)
sock1.bind((my_ip, 10005))

buf = "hello"
port = 10000

for x in range(0, 256):
    try:
        for test in range(0, 100):
            sock1.sendto(buf, (f"192.168.4.{x}", port))
        print(f"192.168.4.{x}:{port}: PASS")
    except OSError as e:
        print(f"192.168.4.{x}:{port}: FAIL after {test+1} attempts with {e}")
```

Output:

```
192.168.4.0:10000: FAIL after 4 attempts with [Errno 12] ENOMEM
192.168.4.1:10000: FAIL after 4 attempts with [Errno 12] ENOMEM
192.168.4.2:10000: FAIL after 4 attempts with [Errno 12] ENOMEM
192.168.4.3:10000: FAIL after 4 attempts with [Errno 12] ENOMEM
192.168.4.4:10000: FAIL after 4 attempts with [Errno 12] ENOMEM
...
192.168.4.248:10000: FAIL after 4 attempts with [Errno 12] ENOMEM
192.168.4.249:10000: FAIL after 4 attempts with [Errno 12] ENOMEM
192.168.4.250:10000: PASS
192.168.4.251:10000: FAIL after 4 attempts with [Errno 12] ENOMEM
192.168.4.252:10000: FAIL after 4 attempts with [Errno 12] ENOMEM
192.168.4.253:10000: FAIL after 4 attempts with [Errno 12] ENOMEM
192.168.4.254:10000: FAIL after 4 attempts with [Errno 12] ENOMEM
192.168.4.255:10000: PASS
```

My next step is to dig into the MicroPython and IDF code, would appreciate any advice on tests to try.

robert-hh, 2023-11-01:
I repeated your test here.  Similarly, I get that error message for every IP address where there is no instance on the net accepting UDP packets. The ENOMEM error could be caused by the ESP network stack running out of internal heap space (not the MicroPython heap). But it may as well be that the error reporting is wrong and Error 12 just means something like: "destination not available".

robtinkers, 2024-02-17:
I'm facing the same issue, but over wifi.

```
import machine, network, socket, sys, time
from secrets import WIFI_SSID, WIFI_PASS # or just define them here

ADDR  = '192.168.1.1'
PORT  = 12345 # UDP
DISCO = True
DELAY = None

sta_if = network.WLAN(network.STA_IF)
if DISCO:
    try: sta_if.disconnect()
    except: pass
    try: sta_if.active(False)
    except: pass

print('starting wifi ', end='')
sta_if.active(True)
sta_if.connect(WIFI_SSID, WIFI_PASS)
while not sta_if.isconnected():
    print('.', end='')
    time.sleep(1)
print()

sock = socket.socket(socket.AF_INET, socket.SOCK_DGRAM)
sink = socket.getaddrinfo(ADDR, PORT)[0][-1]

data = bytearray(1024)

try:
    for i in range(1,101):
        if DELAY is not None:
            time.sleep_ms(DELAY)
        print(i)
        sock.sendto(data, sink)
    print('SUCCESS')
except Exception as e:
    sys.print_exception(e)
```

The first three calls to `sendto()` complete successfully, and the fourth one fails `OSError: [Errno 12] ENOMEM`. It doesn't appear to matter whether the target address is in use (in the ARP cache?) or not. If I put a UDP listener on the target address, it receives the first three packets?frames? but not the fourth. Behaviour is the same whether there the port is listening or not.

If I send to the ESP32's own IP, all calls succeed. If I send to the broadcast address, it fails on the 33rd call. Sometimes if I disable `.disconnect()` and `.active(False)` then it will also fail on the 33rd call. And sometimes it still fails on the 4th call (no obvious pattern.)

Adding a delay before each `sendto()` will succeed, but it needs to be 100ms to be reliable. Lower values get variable results, sometimes a value that works on one run fails on the next.

Tested on ESP32_GENERIC v1.22.1 from micropython.org, ESP32_GENERIC compiled from git against ESP-IDF 5.1.2, and ESP32_GENERIC_C3 v.1.22.1 also from micropython.org.

The same code succeeds on Pico W.