micropython/micropython #12847: Two-dimensional array binding type problem

onecoolx, 2023-11-01:
I'm adding matrix type to [LVGL](https://github.com/lvgl/lvgl) project. I added a matrix type and its c code declaration is：
```
typedef struct {
     float m[3][3];
} lv_matrix_t;
```
But A type error occurred when generating the binding code of micropython.  An error similar to the following:
```
build-standard/lvgl/lv_mpy.c:25340:28: error: expected identifier or ‘(’ before ‘[’ token
25340 | GENMPY_UNUSED STATIC float [3] *mp_arr_to_float___3____3__(mp_obj_t mp_arr)
      |                            ^
build-standard/lvgl/lv_mpy.c:25356:70: error: expected ‘;’, ‘,’ or ‘)’ before ‘*’ token
25356 | GENMPY_UNUSED STATIC mp_obj_t mp_arr_from_float___3____3__(float [3] *arr)
      |                                                                      ^
build-standard/lvgl/lv_mpy.c: In function ‘mp_lv_matrix_t_attr’:
build-standard/lvgl/lv_mpy.c:25397:39: error: implicit declaration of function ‘mp_arr_from_float___3____3__’; did you mean ‘mp_arr_from_float___3__’? [-Werror=implicit-function-declaration]
25397 |             case MP_QSTR_m: dest[0] = mp_arr_from_float___3____3__(data->m); break; // converting from float [3][3];
      |                                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                       mp_arr_from_float___3__
build-standard/lvgl/lv_mpy.c:25397:37: error: assignment to ‘mp_obj_t’ {aka ‘void *’} from ‘int’ makes pointer from integer without a cast [-Werror=int-conversion]
25397 |             case MP_QSTR_m: dest[0] = mp_arr_from_float___3____3__(data->m); break; // converting from float [3][3];
      |
```
Its type becomes  float[3] * , Is this a bug?

dpgeorge, 2023-11-03:
It looks like this is a problem with auto-generated code/bindings.  That generator is not part of this repository so you'll have to ask at lvgl.

onecoolx, 2023-11-03:
> It looks like this is a problem with auto-generated code/bindings.  That generator is not part of this repository so you'll have to ask at lvgl.

thank you for reply！👍

closed 2023-11-10 by onecoolx