micropython/micropython #12887: Use After Free at modselect.c:151

junwha0511, 2023-11-05:
## Environment

- **OS**: Ubuntu 22.04 64bit
- **Version**: micropython at commit db06041d59ef8d5524340c4d2c4490c8e770591f
- **Build**: unix standard build
- **Bug Type**: use-after-free
- **Bug location**: `micropython/extmod/modselect.c:151`
- **Detection Tool**: our custom tool, not disclosed yet
- **Credits**: Junwha Hong and Wonil Jang, from S2Lab in UNIST

## Problem Statement

The problem is from allocating fd for a poll object (poll_obj_t), at poll_set_add_fd because it frees previously allocated fds at m_renew. 

it allocates a fd from the pre-allocated fd array if the `max_used` is not reached to `alloc`, but at the condition of `poll_set->max_used >= poll_set->alloc` , it reallocates whole array `poll_set->pollfds` by `m_renew`. it causes all accesses after m_renew as use-after-free, and we observed some fd is overlapped with an poll object, it will lead to unintended behavior of micropython, and can be used as a vulnerability by attackers in a way they try to malform an object or fd by using each other.

```bash
STATIC struct pollfd *poll_set_add_fd(poll_set_t *poll_set, int fd) {
    struct pollfd *free_slot = NULL;

    if (poll_set->used == poll_set->max_used) {
        // No free slots below max_used, so expand max_used (and possibly allocate).
        if (poll_set->max_used >= poll_set->alloc) {
            poll_set->pollfds = m_renew(struct pollfd, poll_set->pollfds, poll_set->alloc, poll_set->alloc + 4);
            poll_set->alloc += 4;
        }
        free_slot = &poll_set->pollfds[poll_set->max_used++];
    } else {
        // There should be a free slot below max_used.
        for (unsigned int i = 0; i < poll_set->max_used; ++i) {
            struct pollfd *slot = &poll_set->pollfds[i];
            if (slot->fd == -1) {
                free_slot = slot;
                break;
            }
        }
        assert(free_slot != NULL);
    }

    free_slot->fd = fd;
    ++poll_set->used;

    return free_slot;
}
```

## Example (replication)

we found this bug by the following example, which is modified from the testcase of `select_poll_fd` in standard test provided by micropython.

`micropython -X emit=bytecode select_poll_fd.py`

```bash
# Test select.poll in combination with file descriptors.

print("----------------------------------------------------------------------------------------------------0")
try:
    import select, errno

#    select.poll  # Raises AttributeError for CPython implementations without poll()
except (ImportError, AttributeError):
    print("SKIP")
    raise SystemExit

print("----------------------------------------------------------------------------------------------------1")
# Check that poll supports registering file descriptors (integers).
#try:
#    select.poll().register(0)
#except OSError:
#    print("SKIP")
#    raise SystemExit

print("----------------------------------------------------------------------------------------------------2")
# Register invalid file descriptor.
#try:
#    select.poll().register(-1)
#except ValueError:
#    print("ValueError")

print("----------------------------------------------------------------------------------------------------3")
# Test polling stdout, it should be writable.
#poller = select.poll()
#poller.register(1)
#poller.modify(1, select.POLLOUT)
#print(poller.poll())

print("----------------------------------------------------------------------------------------------------4")
# Unregister then re-register.
#poller.unregister(1)
#poller.register(1, select.POLLIN)

print("----------------------------------------------------------------------------------------------------5")
# Poll for input, should return an empty list.
#print(poller.poll(0))

print("----------------------------------------------------------------------------------------------------6")
# Test registering a very large number of file descriptors.
poller = select.poll()
for fd in range(20):
    poller.register(fd)
try:
    poller.poll()
except OSError as er:
    print(er.errno == errno.EINVAL)
```

By running this example, 

1. a fd is allocated when `poll_set->pollfds` is 128 bytes region (which means that `poll_set->alloc` is 16)
2. `poll_set->alloc` is reallocated as 16+4 = 20 bytes in different region, thus a previously allocated fd in freed
3. and the fd is referenced again at `poll_obj_get_revents` , but the region is now allocated as a poll object.

Note: the memory map can be different from pure version because our tool can modified the state of internal memory slightly. but it is obvious that this is an use-after-free.

## Patch

the simplest patch will be use renew_maybe with fixed flag as it is implemented at `parser_alloc`(micropython/py/parse.c:255)

## Log

```bash
    #0 0x55555574a32c in poll_obj_get_revents /home/qbit/testing-2023/micropython/ports/unix/../../extmod/modselect.c:131:34
    #1 0x55555574a32c in poll_poll /home/qbit/testing-2023/micropython/ports/unix/../../extmod/modselect.c:532:13
    #2 0x5555557252e6 in mp_execute_bytecode /home/qbit/testing-2023/micropython/ports/unix/../../py/vm.c:1042:21
    #3 0x5555556fc12e in fun_bc_call /home/qbit/testing-2023/micropython/ports/unix/../../py/objfun.c:273:42
    #4 0x555555821546 in execute_from_lexer /home/qbit/testing-2023/micropython/ports/unix/main.c:160:13
    #5 0x555555820a48 in main_ /home/qbit/testing-2023/micropython/ports/unix/main.c
    #6 0x7ffff7c29d8f in __libc_start_call_main csu/../sysdeps/nptl/libc_start_call_main.h:58:16
    #7 0x7ffff7c29e3f in __libc_start_main csu/../csu/libc-start.c:392:3
    #8 0x555555591854 in _start (/home/qbit/testing-2023/micropython/ports/unix/build-standard/micropython+0x3d854)
```

jimmo, 2023-11-05:
@junwha0511 Thanks for the report.

To summarize: `poll_set_add_fd` can grow `poll_set->pollfds` via `m_renew`, but if this results in a move then existing `poll_obj_t` instances will now have their `pollfd` field pointing to the old memory.

Probably the simplest fix is to turn `poll_set->pollfds` into a linked list instead.

dpgeorge, 2023-11-06:
> Probably the simplest fix is to turn poll_set->pollfds into a linked list instead.

The `pollfds` array is needed because it's passed straight to the `poll()` system call.

To fix the bug, could just adjust all `poll_obj_t::pollfd` entries after the `m_renew` call.

closed 2023-12-21 by dpgeorge

junwha0511, 2023-12-21:
Hi gimmo and dpgeorge, Thank you for your effort for this bug:)