micropython/micropython #12892: Use a GPIO to choose whether to launch main on boot

moefear85, 2023-11-05:
Issues searched. Nothing relevant came up.

Sometimes users write code that locks up the system or floods the repl with output making it practically unusable, even if one presses ctrl+D (eg. timers, interrupts, etc). It would be useful if micropython would check the value of some gpio on boot before deciding whether to import main (only boot code gets executed). ESP-IDF already has an equivalent feature. Ideally turning the feature on/off would be persistent across power cycles and configurable from the repl, rather than at compile time.

Is not a core python feature.

No new architecture involved.

Applies to any port. Can be a built-in.

No drivers needed.

Not willing to sponsor work, expect whoever spends too much time writing non-sense as responses to take initiative.

All placeholders removed.

robert-hh, 2023-11-05:
What stops you from adding suitable code at the beginning of main.py and exit it depending on the state of the Pin? Most ports also have the habit, that main.py is not executed if boot.py raises an exeception.

sosi-deadeye, 2023-11-05:
```python
from machine import Pin

# GPIPO10, Pull-Up
boot_pin = Pin(10, mode=Pin.IN, pull=Pin.PULL_UP)
if not boot_pin():
     raise RuntimeError("Normal execution of main.py aborted")

boot_pin.deinit()
del boot_pin

# main program

```

It is not a good idea to add this because there are many boards, which then require different configurations. In the end, the users are confused because a pin they used before, now has another functionality for a short duration.

closed 2023-11-06 by dpgeorge