micropython/micropython #12912: extremely slow I2C with 1.22.0-preview after 85

robhamerling, 2023-11-08:
With several 1.22.0-preview versions somewhere after 85 I2C performance is extremely slow. 
For example the following code on a board with RP2 PICO and firmware 1.22.0-preview.132 takes 'ages' to complete:
```
from machine import I2C
bus =  I2C(1)          # scl=7, sda=6
bus.scan()
```
With firmware 1.22.0-preview.85 and same code, same hardware response is almost instantaneous.

dpgeorge, 2023-11-08:
I need to ask: do you have pull up resistors on both SCL and SDA?

dpgeorge, 2023-11-08:
I can reproduce the error and bisected the code to find the problem.  It was introduced by 8c432ea2d437255eefb4aff18764936ad11b70c7

That change interacts poorly with 0b26efe73dd3396bdc2b77651a78d9f2edeb9004, see issue #8876.

robhamerling, 2023-11-08:
No, I have no separate pull-ups on the (base) board. I assume/expect that at least some of the sub-PCBs have it: DS3231+EEPROM, PCA9548, OLED.

dpgeorge, 2023-11-08:
See #12916 for a fix.

robhamerling, 2023-11-08:
Tried the fix on a locally copy: speed of I2C.scan() is 'normal' again.