micropython/micropython #12925: rp2: Timeout regression in select/poll, cyw43, others

projectgus, 2023-11-09:
Prior to commit b41055a5 (PR #12837), this sequence returns immediately (1ms delay):

```
>>> import select
>>> select.poll().poll(1)
[]
```

Since that commit, it hangs indefinitely extmod/modselect.c calls `MICROPY_EVENT_POLL_HOOK` while polling for a ready fd or for the timeout to be reached.

More generally, select/poll timeout may not be reached in a quiet rp2 system without a lot of interrupts.

The same problem will probably be seen in network_cyw43.c:226 (connect timeout), lwip recfrom (recv timeout), NimBLE ble stack timeout... basically anywhere that calls `MICROPY_EVENT_POLL_HOOK` in a tight loop which also checks a timeout condition.

Similar bug to #12912, but harder to fix.

projectgus, 2023-11-09:
@dpgeorge What do you recommend here? (cc @jimmo )

Changing all these to `MICROPY_EVENT_POLL_HOOK_FAST` seems too excessive for power consumption.

One fix is the one you suggested in 12916, to add a `MICROPY_WAIT_EVENT_WITH_TIMEOUT(t);` or `MICROPY_WAIT_EVENT_WITH_DEADLINE(d);` type macro. This can be a tight loop on every port except rp2, but more ports could be changed over if they also become tickless.

Another fix (maybe temporary) would be to merge the changes from here, which enables SysTick on rp2: 
https://github.com/micropython/micropython/commit/bd0823c8946189cfceb905280723a001578e498b

jimmo, 2023-11-09:
So just to make sure I understand:
* We have `MICROPY_EVENT_POLL_HOOK_FAST` which is designed to be used in places where you just want to make sure the scheduler and USB keep running, but you want to do very low-latency polling.
* `MICROPY_EVENT_POLL_HOOK` is the "idle" version. It previously used wfe-with-timeout, but the overhead of the "with timeout" part was non-trivial. So now it just uses `wfe`, but this means it no longer works for waiting for things that need to be explicitly polled.
* In general, we'd expect `wfe` to work here because anything that can suddenly become "ready" must have trigged an "e", but in some cases (e.g. Nina wifi), this isn't the case.

So the options (so far) are:
* Add systick (which means the `wfe` cannot take more than N milliseconds, but no management of the hardware timer required)  (we don't really want this, we'd rather move towards tickless)
* Just use `MICROPY_EVENT_POLL_HOOK_FAST` everywhere (implemented by just having only `MICROPY_EVENT_POLL_HOOK`, and not do `wfe` in it). (this would indeed be bad for power)
* Require the polling loop (i.e. the thing that is calling `MICROPY_EVENT_POLL_HOOK`) know whether the thing it's waiting for is `wfe`-suitable. Otherwise it would fallback to the heaver-weight wfe-with-timeout.  (e.g. modselect must use `wfe-with-timeout`, but something that knows the thing it's waiting for triggers an IRQ, e.g. USB, could safely use the `wfe` version).

dpgeorge, 2023-11-09:
Yes the above is correct.  Except the "make sure the scheduler and USB keep running" bit is going to be subtly different after #12846 is merged, because then the only thing `MICROPY_EVENT_POLL_HOOK_FAST` does is call `mp_handle_pending()` (which may eventually call the low-priority USB task).

jimmo, 2023-11-09:
This is a bit of an aside, but I'd quite like to rename this macro... The `_HOOK` makes it seem like the other vm macros `MICROPY_VM_HOOK_LOOP` (i.e. you want to hook into the vm), but really this is more the port providing its "background task" for things that want to wait on an event (or...timeout), but really all ports are implementing the functionality independently.

To put that another way, for it to be called `_HOOK`, would make more sense if there was a central implementation of the "wait for event" macro which did

```
#define MICROPY_WAIT_FOR_EVENT(timeout) \
  mp_handle_pending(); \
  MICROPY_WFE(timeout) \
  MICROPY_WAIT_FOR_EVENT_HOOK \
```

and then the port just provided `MICROPY_WFE(timeout)` (i.e. unconditionally `__wfe` on systick ports, or `timeout ? <custom> : __wfe` on tickless ports), and most ports wouldn't implement the hook at all.

projectgus, 2023-11-09:
I like that structure.

It might also be worth asking if this could be a `mp_wait_for_event()` function rather than an inlined macro (except for the "fast" hook). In the places where this is used, the extra function call time overhead is unlikely to count for much but it'd save some code size.

dpgeorge, 2023-11-09:
We can use `static inline` functions for speed.  The downside is that they sometimes require including extra headers in `mpconfigport.h` which can slow down compilation.

But actually, for these event macros/functions, we should move them to `mphalport.h`, which is only included by `py/mphal.h` and only when needed.

projectgus, 2023-11-09:
> We can use static inline functions for speed. The downside is that they sometimes require including extra headers in mpconfigport.h which can slow down compilation.

IME the upside is you can put generic declarations somewhere with comments so that anyone implementing or calling the function can see what each of them is supposed to do (or not do) across all ports.

In this case I was thinking the opposite, though: these are probably not called from tight enough code paths that a single function call (or small stack frame) matters, but the macro is inlined in a dozen or more places so de-inlining it could save a hundred or more bytes of binary size.

closed 2023-12-08 by dpgeorge