micropython/micropython #12946: UART.init() not working correctly on Raspberry Pi Pico W

mh-g, 2023-11-11:
Using UART.init() on RPi Pico W breaks UART functionality for the respective port when changing pins. (Even when changing back to the original pins.)

```
>>> from machine import UART, Pin
>>> uart = UART(0, baudrate=2400, tx=Pin(0), rx=Pin(1))
>>> uart.readline()
b'\x00Data on Pin 1\r\n'
>>> uart.readline()
b'\x00Data on Pin 1\r\n'
>>> uart.init(baudrate=2400, tx=Pin(12), rx=Pin(13))
>>> uart.readline()
>>> uart.readline()
>>> uart.init(baudrate=2400, tx=Pin(0), rx=Pin(1))
>>> uart.readline()
>>> uart.readline()
>>> uart2 = UART(1, baudrate=2400, tx=Pin(4), rx=Pin(5))
>>> uart2.readline()
b'\x00Data on Pin 5\r\n'
```

I would expect that I can change the pins on the fly (as per the note in the documentation of UART init: https://docs.micropython.org/en/latest/library/machine.UART.html#machine.UART.init).

External wiring is between RPi Pico W and Arduino Nano with the following wires:
```
RPi Pico W  |  Arduino Nano
------------+----------------------------
1 (GP0)     | (19) A00 (configured as RX)
2 (GP1)     | (22) A03 (configured as TX)
16 (GP12)   | (20) A01 (configured as RX)
17 (GP13)   | (23) A04 (configured as TX)
21 (GP16)   | (21) A02 (configured as RX)
22 (GP17)   | (24) A05 (configured as TX)
11 (GP8)    | (19) A00 (configured as RX)
12 (GP9)    | (22) A03 (configured as TX)
18 (GND)    | (29) GROUND
```
(UART1, first ports, deliberately connected to the same ports on the Nano as UART0, second ports.)

firmware file: RPI_PICO_W-20231005-v1.21.0.uf2
version information: MicroPython v1.21.0 on 2023-10-06; Raspberry Pi Pico W with RP2040

robert-hh, 2023-11-12:
I cannot confirm this. I tried the same here by connecting Pin GP0 and GP1  as well as Pins GP12 and GP13  (local loopback).
I can write and read back the data that has been send after changing the pins forth and back.

mh-g, 2023-11-12:
@robert-hh : I can confirm your observation with local loopback. I need to further investigate, why local loopback works, but not with an Arduino Nano on the other side.

mh-g, 2023-11-13:
Further testing: I replaced the Arduino Nano with a second Pico W. The wiring is as follows:
Pico#0 GND <-> Pico#1 GND

connection (a):
Pico#0 pin 0 <-> Pico#1 pin 1
Pico#0 pin 1 <-> Pico#1 pin 0

connection (b):
Pico#0 pin 12 <-> Pico#1 pin 13
Pico#0 pin 13 <-> Pico#1 pin 12

connection (c):
Pico#0 pin 16 <-> Pico#1 pin 17
Pico#0 pin 17 <-> Pico#1 pin 16

I can set up UARTs on both sides with the constructor to the same connection and communicate bidirectionally as expected.

I can change UARTs on both side using UART.init(...) to a different connection (of course the same one on both sides) and communicate bidirectionally as expected.

However, if I change connection only on one Pico, e.g. init() to connection (b) and keep the other Pico configured to connection (a), or if I change both connections to different ones (e.g. from (a) to (b) on Pico#0 and from (a) to (c) on Pico#1), I can still communicate with the other Pico. See the attached picture for an exact sample scenario using only connections (a) and (b).
[image]

robert-hh, 2023-11-13:
Did you do a soft reset between each test? If you change for a uart the setting from one pin to another, the first pin is not disabled. That may lead to strange effects. You could try to re-configure the not-any-more-used pins as machine.Pin objects.

robert-hh, 2023-11-14:
I would configure all not-used Pins as IN. That's a safe state. Besides that, the implementation of uart.init() look correct.

mh-g, 2023-12-04:
I gave it some more tries, did not succeed, and then reverted to fully closing the UART and creating a new instance with the other ports. This works.

uwekoenig, 2024-11-24:
I had the same issue with Waveshare Pico-2CH-RS232. Sometimes the UART connection was established like expected (20%), sometimes not (80%).
My workaround which works 98%:
```
uart0 = UART(0, baudrate=9600, tx=Pin(0), rx=Pin(1), bits=8, parity=None, stop=1)
uart0.deinit()
uart0 = UART(0, baudrate=9600, tx=Pin(0), rx=Pin(1), bits=8, parity=None, stop=1)
```