micropython/micropython #12951: Micropython compile error when calling mp_obj_is_float(mp_obj_t)

mo-FEAR, 2023-11-12:
I'm on esp-idf commit f333925, using xtensa-esp32-elf-gcc (crosstool-NG esp-2022r1) 11.2.0. I have the following function:

```
mp_obj_t some_func(mp_obj_t self_,mp_obj_t some_val) {
        if (mp_obj_is_float(some_val)) {
               // do something useful (no nonsense)
        }
        return mp_const_none;
    }
```

It produces the following compile error:
```
In file included from /media/EXTRA_EXT4/source_mods/micropython/py/mpstate.h:33,
                 from /media/EXTRA_EXT4/source_mods/micropython/py/runtime.h:29,
                 from /media/BACKUP/Projects/Programming/Micropython/Modules_Static/libraries/libraries.hpp:7,
                 from /media/BACKUP/Projects/Programming/Micropython/Modules_Static/foo/foo.hpp:5,
                 from /media/BACKUP/Projects/Programming/Micropython/Modules_Static/foo/foo.cpp:1:
/media/BACKUP/Projects/Programming/Micropython/Modules_Static/foo/foo.cpp: In function 'void* foo_duty(mp_obj_t, mp_obj_t, mp_obj_t)':
/media/EXTRA_EXT4/source_mods/micropython/py/misc.h:54:53: error: size of array is not an integral constant-expression
   54 | #define MP_STATIC_ASSERT(cond) ((void)sizeof(char[1 - 2 * !(cond)]))
      |                                                   ~~^~~~~~~~~~~~~
/media/EXTRA_EXT4/source_mods/micropython/py/misc.h:58:40: note: in expansion of macro 'MP_STATIC_ASSERT'
   58 | #define MP_STATIC_ASSERT_NOT_MSC(cond) MP_STATIC_ASSERT(cond)
      |                                        ^~~~~~~~~~~~~~~~
/media/EXTRA_EXT4/source_mods/micropython/py/obj.h:933:5: note: in expansion of macro 'MP_STATIC_ASSERT_NOT_MSC'
  933 |     MP_STATIC_ASSERT_NOT_MSC((t) != &mp_type_bool), assert((t) != &mp_type_bool),         \
      |     ^~~~~~~~~~~~~~~~~~~~~~~~
/media/EXTRA_EXT4/source_mods/micropython/py/obj.h:939:31: note: in expansion of macro 'mp_type_assert_not_bool_int_str_nonetype'
  939 | #define mp_obj_is_type(o, t) (mp_type_assert_not_bool_int_str_nonetype(t) && mp_obj_is_exact_type(o, t))
      |                               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/media/EXTRA_EXT4/source_mods/micropython/py/obj.h:120:28: note: in expansion of macro 'mp_obj_is_type'
  120 | #define mp_obj_is_float(o) mp_obj_is_type((o), &mp_type_float)
      |                            ^~~~~~~~~~~~~~
/media/BACKUP/Projects/Programming/Micropython/Modules_Static/foo/foo.cpp:325:13: note: in expansion of macro 'mp_obj_is_float'
  325 |         if (mp_obj_is_float(duty_)) {
      |             ^~~~~~~~~~~~~~~
/media/EXTRA_EXT4/source_mods/micropython/py/misc.h:54:53: error: size of array is not an integral constant-expression
   54 | #define MP_STATIC_ASSERT(cond) ((void)sizeof(char[1 - 2 * !(cond)]))
      |                                                   ~~^~~~~~~~~~~~~
/media/EXTRA_EXT4/source_mods/micropython/py/misc.h:58:40: note: in expansion of macro 'MP_STATIC_ASSERT'
   58 | #define MP_STATIC_ASSERT_NOT_MSC(cond) MP_STATIC_ASSERT(cond)
      |                                        ^~~~~~~~~~~~~~~~
/media/EXTRA_EXT4/source_mods/micropython/py/obj.h:934:5: note: in expansion of macro 'MP_STATIC_ASSERT_NOT_MSC'
  934 |     MP_STATIC_ASSERT_NOT_MSC((t) != &mp_type_int), assert((t) != &mp_type_int),           \
      |     ^~~~~~~~~~~~~~~~~~~~~~~~
/media/EXTRA_EXT4/source_mods/micropython/py/obj.h:939:31: note: in expansion of macro 'mp_type_assert_not_bool_int_str_nonetype'
  939 | #define mp_obj_is_type(o, t) (mp_type_assert_not_bool_int_str_nonetype(t) && mp_obj_is_exact_type(o, t))
      |                               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/media/EXTRA_EXT4/source_mods/micropython/py/obj.h:120:28: note: in expansion of macro 'mp_obj_is_type'
  120 | #define mp_obj_is_float(o) mp_obj_is_type((o), &mp_type_float)
      |                            ^~~~~~~~~~~~~~
/media/BACKUP/Projects/Programming/Micropython/Modules_Static/foo/foo.cpp:325:13: note: in expansion of macro 'mp_obj_is_float'
  325 |         if (mp_obj_is_float(duty_)) {
      |             ^~~~~~~~~~~~~~~
/media/EXTRA_EXT4/source_mods/micropython/py/misc.h:54:53: error: size of array is not an integral constant-expression
   54 | #define MP_STATIC_ASSERT(cond) ((void)sizeof(char[1 - 2 * !(cond)]))
      |                                                   ~~^~~~~~~~~~~~~
/media/EXTRA_EXT4/source_mods/micropython/py/misc.h:58:40: note: in expansion of macro 'MP_STATIC_ASSERT'
   58 | #define MP_STATIC_ASSERT_NOT_MSC(cond) MP_STATIC_ASSERT(cond)
      |                                        ^~~~~~~~~~~~~~~~
/media/EXTRA_EXT4/source_mods/micropython/py/obj.h:935:5: note: in expansion of macro 'MP_STATIC_ASSERT_NOT_MSC'
  935 |     MP_STATIC_ASSERT_NOT_MSC((t) != &mp_type_str), assert((t) != &mp_type_str),           \
      |     ^~~~~~~~~~~~~~~~~~~~~~~~
/media/EXTRA_EXT4/source_mods/micropython/py/obj.h:939:31: note: in expansion of macro 'mp_type_assert_not_bool_int_str_nonetype'
  939 | #define mp_obj_is_type(o, t) (mp_type_assert_not_bool_int_str_nonetype(t) && mp_obj_is_exact_type(o, t))
      |                               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/media/EXTRA_EXT4/source_mods/micropython/py/obj.h:120:28: note: in expansion of macro 'mp_obj_is_type'
  120 | #define mp_obj_is_float(o) mp_obj_is_type((o), &mp_type_float)
      |                            ^~~~~~~~~~~~~~
/media/BACKUP/Projects/Programming/Micropython/Modules_Static/foo/foo.cpp:325:13: note: in expansion of macro 'mp_obj_is_float'
  325 |         if (mp_obj_is_float(duty_)) {
      |             ^~~~~~~~~~~~~~~
/media/EXTRA_EXT4/source_mods/micropython/py/misc.h:54:53: error: size of array is not an integral constant-expression
   54 | #define MP_STATIC_ASSERT(cond) ((void)sizeof(char[1 - 2 * !(cond)]))
      |                                                   ~~^~~~~~~~~~~~~
/media/EXTRA_EXT4/source_mods/micropython/py/misc.h:58:40: note: in expansion of macro 'MP_STATIC_ASSERT'
   58 | #define MP_STATIC_ASSERT_NOT_MSC(cond) MP_STATIC_ASSERT(cond)
      |                                        ^~~~~~~~~~~~~~~~
/media/EXTRA_EXT4/source_mods/micropython/py/obj.h:936:5: note: in expansion of macro 'MP_STATIC_ASSERT_NOT_MSC'
  936 |     MP_STATIC_ASSERT_NOT_MSC((t) != &mp_type_NoneType), assert((t) != &mp_type_NoneType), \
      |     ^~~~~~~~~~~~~~~~~~~~~~~~
/media/EXTRA_EXT4/source_mods/micropython/py/obj.h:939:31: note: in expansion of macro 'mp_type_assert_not_bool_int_str_nonetype'
  939 | #define mp_obj_is_type(o, t) (mp_type_assert_not_bool_int_str_nonetype(t) && mp_obj_is_exact_type(o, t))
      |                               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/media/EXTRA_EXT4/source_mods/micropython/py/obj.h:120:28: note: in expansion of macro 'mp_obj_is_type'
  120 | #define mp_obj_is_float(o) mp_obj_is_type((o), &mp_type_float)
      |                            ^~~~~~~~~~~~~~
/media/BACKUP/Projects/Programming/Micropython/Modules_Static/foo/foo.cpp:325:13: note: in expansion of macro 'mp_obj_is_float'
  325 |         if (mp_obj_is_float(duty_)) {
      |             ^~~~~~~~~~~~~~~
[1266/1285] Building CXX object esp-idf/main_esp32/CMakeFiles/__idf_main_esp32.dir/media/BACKUP/Projects/Programming/Micropython/Modules_Static/espnow/espnow.cpp.objninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the /tmp/build-ESP32_GENERIC-SPIRAM/log/idf_py_stderr_output_3672 and /tmp/build-ESP32_GENERIC-SPIRAM/log/idf_py_stdout_output_3672
```

nspsck, 2023-11-13:
Hi, are you trying to add external module to your firmware? If that were the case:

First thing I've noticed is that you are using cpp. Am not sure if that is supported by Microypthon since in the documentation it says: `USER_C_MODULES`.

Also, your function probably has to be `STATIC` (it is essentially `static` in C) and you also have to add something like this to your code:
```C
STATIC MP_DEFINE_CONST_FUN_OBJ_2(some_func_obj, some_func)
```
so the compiler knows what your function actually is.

Disclamer: I am a novice to this project and what I have commented might be irritating or completely false.

stinos, 2023-11-13:
Issue seems to be simply that you're compiling C++ code (that MP_STATIC_ASSERT macro is not valid C++ AFAIK hence the "size of array is not an integral constant-expression"), strange this did not come up earlier. There's no readily available workaround currently. You could change obj.h around line 937 so that it has

```
#define mp_obj_is_type(o, t) (mp_obj_is_exact_type(o, t))
```

instead of what it has now.

Not sure what the best way to go is here for a proper fix. Either wrapping the thing in `#ifndef __cplusplus` or else use modern C++'s `static_assert` builtin but older compilers won't have that.

> since in the documentation it says: USER_C_MODULES.

Not really an issue, examples/usercmodule/cppexample is there to demonstrate it works with C++. But that example doesn't contain `mp_obj_is_float(...)`; from the moment that is there this example also fails to compile.

> I believe that the variable has to be the type mp_const_obj_t i

Should also not be an issue

stinos, 2023-11-13:
> Either wrapping the thing in #ifndef __cplusplus or else use modern C++'s static_assert builtin but older compilers won't have that.

Checked this and while I didn't dig all the way into the standard, I thing static_assert is never going to work here: the mp_obj_is_type checks basically come down to

```
typedef struct _mp_obj_type_t mp_obj_type_t;

extern const mp_obj_type_t mp_type_bool;
extern const mp_obj_type_t mp_type_float;

void foo() {
    static_assert(&mp_type_bool != &mp_type_float);
}
```

but comparing pointers like that is not a constant expression in C++.

projectgus, 2023-11-13:
Well spotted, @nspsck - you are right on that the problem is related to compiling this code as C++ not C.

The compilation error is actually a static assert failing, because comparing addresses of extern variables is not compile-time constant in C++ but it is in C. (The build fails with a compilation error, but in this context a compile error in `MP_STATIC_ASSERT` is an opaque way of saying "assertion failed").

i.e. this C code is OK - https://gcc.godbolt.org/z/jsbYa64oc but the same code is not OK as C++ - https://gcc.godbolt.org/z/hxvE3bTzT

The deeper reason for this is, I assume, hidden in the details of C vs C++ standard rules about variable aliasing. It'd be interesting to know the actual detail, but it also won't help us resolve this.

Until such time as there's a fix, the best workaround is probably to split out any interface code into a C file and have it call into the C++ code.

This workaround might also work: Add the following after the `#include` lines and before your C++ code:

```cpp
#undef MP_STATIC_ASSERT_NOT_MSC
#define MP_STATIC_ASSERT_NOT_MSC(cond)
```

(This is a bit hacky and there may be other macro chains that lead to the same assertion failure, though.)

projectgus, 2023-11-13:
Sorry @stinos, I had the old version of this page open from before you replied - but your reply is 100% on point as well.

closed 2023-11-17 by dpgeorge