micropython/micropython #12955: Simple I2S streaming code only works for sample rates <22050

aguaviva, 2023-11-13:
notes:
- I know 32bit samples is an overkill, any efficient way to make it 24bits? I really need this range
- you can use ulab numpy (I actually love it)
- I can still hear some clicking at the chunk boundaries
- As for web.py I am using the great https://github.com/wybiral/micropython-aioweb
- As for queue.py I am using https://github.com/peterhinch/micropython-async/blob/master/v3/primitives/queue.py

```python
from machine import I2S
from machine import Pin
import web
import asyncio 
from queue import Queue    
  
bck_pin = Pin(33)  # Bit clock output
ws_pin = Pin(32)   # Word clock output
sdin_pin = Pin(25) # Serial data input

au_chunk_len = 8192
au_chunks = 8 # must be power of 2

audio_in = I2S(1,                                 # create I2S peripheral to read audio
               sck=bck_pin, ws=ws_pin, sd=sdin_pin,    # sample data from an INMP441
               mode=I2S.RX,
               bits=32,
               format=I2S.MONO,
               rate=8000,
               ibuf=au_chunk_len)

sreader = asyncio.StreamReader(audio_in)

queue = Queue(au_chunks)

async def produce():
    print("Running produce()")
    
    #ring buffer
    dma_buffer = memoryview(bytearray(au_chunk_len*au_chunks))

    f=0
    while True:
        b = dma_buffer[f*au_chunk_len:(f+1)*au_chunk_len]
        t = au_chunk_len
        while t>0:
            t -= await sreader.readinto(b[au_chunk_len-t:])            
        await queue.put(b)
        f= (f+1)&(au_chunks-1)
        
            
app = web.App(host='0.0.0.0', port=80)

@app.route('/')
async def scan_network_handler(r, w):
    w.write(b'HTTP/1.0 200 OK\r\n')
    w.write(b'Content-Type: application/octet-stream\r\n')
    w.write(b'\r\n')

    # empty queue
    for i in range(au_chunks):
        _ = await queue.get()

    for i in range(30):
        print(f"c {i}, {queue.qsize()}")
        samples = await queue.get()
        w.write(samples)
    await w.drain()   

# Start event loop and create server task
loop = asyncio.get_event_loop()
loop.create_task(app.serve())
loop.create_task(produce())
loop.run_forever()
```

miketeachman, 2023-11-13:
Can you describe the problem you see?   
Here is a thought on the problem that I'm imagining you are encountering.  For streaming audio your internal buffer,`ibuf`, might be too small and overflows when the scheduler is running the web app.  Try an ibuf=40000 or larger.  Please refer to this example to understand more on how the I2S protocol works in MicroPython.
https://github.com/miketeachman/micropython-i2s-examples#explaining-the-i2s-protocol-with-buckets-and-water

aguaviva, 2023-11-13:
Thanks for your reply, I'll try that.

I actually saw your repo and I found it very useful, but I was not sure how close was your implementation with Micropython's.

peterhinch, 2023-11-13:
You might like to consider [RingbufQueue](https://github.com/peterhinch/micropython-async/blob/master/v3/docs/DRIVERS.md#7-ringbuf-queue). This is a circular buffer which is `asyncio` friendly.

aguaviva, 2023-11-14:
The issue is that there are lots of clicks due to the wave being discontinuous. This should be a chirp.

[image]

and this is a close up of the discontinuity between the 3rd and the 4th block

[image]

aguaviva, 2023-11-14:
With au_chunk_len=40000 I get a discontinuity every 10_000 samples. So in the 40_000 samples below I get 4 clicks.

[image]

aguaviva, 2023-11-14:
> Can you describe the problem you see? Here is a thought on the problem that I'm imagining you are encountering. For streaming audio your internal buffer,`ibuf`, might be too small and overflows when the scheduler is running the web app. Try an ibuf=40000 or larger. Please refer to this example to understand more on how the I2S protocol works in MicroPython. The example is for writing, but the same concept applies to reading samples from an I2S codec. https://github.com/miketeachman/micropython-i2s-examples#explaining-the-i2s-protocol-with-buckets-and-water

BTW I already knew the technique you are describing in your post and if you look at the code I am actually using a FIFO .

aguaviva, 2023-12-02:
How can I help?  I'd appreciate getting some guidance on how to troubleshoot/debug this.

Are there any experiments you would like me to run?

aguaviva, 2023-12-12:
@miketeachman I was hoping you could have a quick look at this issue as you seem interested in I2S.

miketeachman, 2023-12-15:
> @miketeachman I was hoping you could have a quick look at this issue as you seem interested in I2S.

I have been planning to look deeper into this issue, but I need to finish a personal project which has a deadline.  This project should be completed today.  I will have time on weekend to take a look.  I hope I can help.

aguaviva, 2023-12-15:
Much appreciated, I'd be great to hear your opinion

miketeachman, 2023-12-18:
Can you please post the complete modified code that shows a discontinuity every 10000 samples?  I'm having to guess what you modified from your original code and that is making it harder to understand what might be happening.  Thanks.

aguaviva, 2023-12-21:
The code is the one I posted, and the variable values I changed are described in the rest of the messages. Could you repro the very first issue?

miketeachman, 2023-12-22:
I'm in the process of trying to reproduce the problem.   I have a couple of questions:
1. Can you share the networking code?  The posted code doesn't indicate how the streaming samples get off the ESP32 and into a client application.
2. What client application did you stream the audio samples into?

aguaviva, 2024-01-15:
Sorry for the delay :)

Here is the client application, it is a jupyter notebook: [audio_esp32.zip](https://github.com/micropython/micropython/files/13943358/audio_esp32.zip)

Let me know if it works for you.