micropython/micropython #12957: RTC issues on PYBD (STM32)

victorallume, 2023-11-13:
We've encountered a number of issues using the RTC on PYBD across a number of devices. These include

* RTC stopping entirely (subsequent calls to the RTC return exactly the same value)
* RTC blocking for ~1 sec when getting or setting the time
* RTC returning a plausible but incorrect value (I think a bit has become 'stuck')

These faults are often intermittent making debugging quite difficult.
As such, for our application code we are now using a 'soft' RTC based off CPU ticks, and synchronising from the network on a regular basis.

peterhinch, 2023-11-13:
This sounds like it may be a hardware problem. Do you have a means of checking whether the crystal oscillator is still running when the RTC stops entirely? Which Pyboard variant are you using?

andrewleech, 2023-11-13:
It sounds like you can roll out firmware updates at least to test / workaround the issue? 

If so you could try reconfiguring the board to use the internal osc (LSI or divided HSE) for RTC rather than the external crystal to at least check if that's working - then the rest of the RTC code wouldn't need to be touched. This would be less accurate like you current workaround though, so the regular updates would probably still be needed, but it would help to isolate the problem to check if it's indeed just the clock source causing the problems.

Regarding the points 2 and 3, I believe a failing external crystal clock input would certainly cause these issues too; if the crystal signal is weak the osc could certainly miss a few cycles which would directly translate into losing time (stalling or stuck).
Also the entire RTC peripheral module is clocked by the same source, so if it's missing clock cycles then attempts to write to its registers will also likely stall while it's waiting for enough clock cycles to come in to complete the operation.

Oh... https://micropython.org/resources/PYBD_SFxW_schematics.pdf
PYBD is using an external clock module, not an xtal, and that same clock is going to the wifi module also.

Regardless, you could still try switching to LSI to see if that helps, the settings for this are in `micropython/ports/stm32/boards/PYBD_SF2/mpconfigboard.h` (it's shared with SF6) 
```
// There is an external 32kHz oscillator
#define RTC_ASYNCH_PREDIV           (3)
#define RTC_SYNCH_PREDIV            (0x1fff)
#define MICROPY_HW_RTC_USE_BYPASS   (1)
#define MICROPY_HW_RTC_USE_US       (1)
#define MICROPY_HW_RTC_USE_CALOUT   (1)
```
I think you just need to change `MICROPY_HW_RTC_USE_BYPASS` to `0` to trigger it to use LSI. Could also add 
`#define MICROPY_HW_RTC_USE_LSE      (0)` if you wanted to be explicit.

You could also try changing some of those other 32khz clock settings to see if they have any impact, though testing stuff remotely doesn't make this easy.

peterhinch, 2023-11-14:
Some years ago I accidentally stopped the oscillator on a Pyboard 1.1. Apart from the obvious outcome of the RTC stopping, I also remember the RTC blocking as you describe. Could the boards be subject to condensing humidity or some other source of relatively low impedance?

victorallume, 2024-01-22:
@andrewleech as I understand it, the wifi module only uses the oscillator for sleep mode. I'm not sure if micropython utilises this, or the default power management mode (this doesn't seem to be documented, and I can't read the `pm` option from `WLAN.config()` - I think it's using performance power mode, but don't know if this relies on the LSE osciallator. 

We don't need any wifi power saving for our purposes, and if there is any chance that a failing oscillator could be impacting wifi stability I'd be very interested. 

Unfortunately we don't have the capacity to play around with changing to LSI, and our work-around is sufficient for the time being

@peterhinch - I don't believe there is much chance of condensing humidity - we're in a relatively warm box. But I'll look into it.

peterhinch, 2024-01-22:
I can't recall how I stopped the oscillator but it didn't restart until I power cycled and there was definitely the blocking behaviour you described.

victorallume, 2024-01-31:
I've caught a unit on my bench doing this. I attached it to a breakout board and probed the OSC32 line which is showing a nice clean waveform:
[image]

When I first started it up, `time.time()` was consistently returning 0. At some point (after attaching breakout board and probe while it was still running), `time.time() started returning `473385612` (over and over). The sudden jump is unexpected, and I don't really know the significance of that number.

Digging into the RCC registers shows that it thinks the LSE is on and ready, the RTC is enabled, and configured to use LSE.

`time.time()` continues returning the same value after a soft reboot

I'm running micropython v1.18 (locally compiled, but unchanged).

victorallume, 2024-01-31:
After running `machine.RTC().init()` the RTC starts working, albeit in LSI mode. After a hard reset, the RTC starts working again in LSE mode. 

Given that, while it was broken, there was a valid LSE waveform on the OSC32 line, and the RTC was configured to use the LSE (I checked the RCC registers), but was not ticking over, I can guess is either:

- An intermittent connection/soldering issue, or
- Some internal to the STM32

The errata for the STM32F767 (https://www.st.com/resource/en/errata_sheet/es0334-stm32f76xxx-and-stm32f77xxx-device-errata-stmicroelectronics.pdf) notes two issues, but one (2.2.3) is for the TFBGA216 package (PYBD uses BGA176), but notes instability when toggling certain pins close to the OSC32 line, and advises not using high or low LSE drive modes. The other (2.2.6) notes that toggling pin PC13 disturbs the LSE clock. But we don't use PC13 on our application.

victorallume, 2024-02-12:
I have tried hammering the GPIO pins near OSC32, but was unable to re-create the problem. 
I also thing the likelihood of a connection/soldering issue is very low, since we are not seeing any problems on any other lines, and this specific issue has shown up on multiple units in the field.

victorallume, 2024-04-10:
There are some possibly related notes at http://www.efton.sk/STM32/gotcha/index.html