micropython/micropython #12972: Keeping user c module build self contained.

kdschlosser, 2023-11-14:
I am going to do my best to explain this and I hope I am able to get it right. This could be a bug and is not working the way it should be or it could be a feature request if it wasn't done this way.

The objective here is to be able to add C code to be compiled as part of MicroPython without the need to alter any of the MicroPython build system. This would be an ideal thing to be able to do for someone wanting to extend functionality. 

In my use case I need to have access to the esp_lcd component of the ESP-IDF from the python interpreter. When building MicroPython for the ESP32 the esp32_common.cmake file gets used and inside of that file is the following

```
list(APPEND IDF_COMPONENTS
    app_update
    bootloader_support
    bt
    driver
    esp_adc
    esp_app_format
    esp_common
    esp_eth
    esp_event
    esp_hw_support
    esp_netif
    esp_partition
    esp_pm
    esp_psram
    esp_ringbuf
    esp_rom
    esp_system
    esp_timer
    esp_wifi
    freertos
    hal
    heap
    log
    lwip
    mbedtls
    newlib
    nvs_flash
    sdmmc
    soc
    spi_flash
    ulp
    vfs
)
```

Now if I wanted to have access to the esp_lcd component I would think that I should be able to do the following in the micropython.cmake file that is in my user c module

```
list(APPEND IDF_COMPONENTS esp_lcd)
```

This does not work. I am forced to edit the  esp32_common.cmake file and add the esp_lcd component.

It would be nice to not have to modify the MicroPython build system at all to be able to add the needed component to the list. Keeping the user c module completely self contained would make distributing the module a whole lot easier to do. This same idea would need to work across all of the different boards/SDK's that are used when building. 

I am aware this has been mentioned before in #8041 But I think it needs to be brought up again seeing as how that was back in 2021 and  it has not been done.

andrewleech, 2023-11-14:
That's a very interesting find @kdschlosser, I hadn't seen that previously raised issue myself! This looks like it should explain why I was having such a hard time getting my lvgl user c module work going on esp32.

To be clear, this definitly looks like a (hopefully minor) bug, the user c modules system across all ports is definitely intended to be used without needing any mods to microypthon repo files.

kdschlosser, 2023-11-15:
It would be nice if it worked this way. It would definitely make things a whole lot easier that for sure. I don't know why the component is not able to be added from a user module. The user module gets processed by cmake before any compilation of the source files even starts. It looks like the user module might compile in a kind of sandbox so any changes to cmake variables do not populate out of the user module built into the MicroPython build. Maybe a possible solution to the problem is to set an environment variable with the additional components and then check for the existence of that environment variable is a user c module is being compiled and if it is then extend the IDF components list with what is stored in that environment variable. I don't think there would be much else that would need to be populated from the user c module build over to the MicroPython build. I am going to mess about with it and see if it will work doing that.

andrewleech, 2023-11-15:
I agree this bug needs to be fixed in micropython to be able to use these user-C modules properly. This is one of the things I hate about `cmake`, it does seem to compartmentalise things a lot (too much) making it hard to bring modules together. I find regular `make` so much easier personally.

The way the user-c-module cmake fragment is brought into the main micropython / idf cmake build structure probably needs to be changed a bit to fix this, though it might take a while to figure out what needs fixing. While I came up with some of the original reference implementation for user c modules on the stm port back in the day, @jimmo did the heavy lifting here to get it working on other ports and the cmake stuff in particular.... though I believe he dislikes cmake at least as much as I do if not far more :-D

LnnrtS, 2023-12-18:
This is not mainly a cmake issue but one related to how esp-idf discovers required components (see https://github.com/micropython/micropython/issues/8041#issuecomment-1859964653)

kdschlosser, 2023-12-18:
I  just found out that all components that are packaged with the ESP-IDF are by default added to the list of files that need to be compiled. The thing that is missing are the include paths. The MicroPython build system sets the list of components as requirements when it gets registered as a component. Then later on it collects that list of requirements from the IDF build system, iterates over them and sets any include paths to the flags that get passed to the compiler. Unfortunately this all takes place before the user_c_module cmake file gets imported. BUT knowing the information above this can be done.

 ```cmake
# gets esp_lcd include paths
idf_component_get_property(esp_lcd_includes esp_lcd INCLUDE_DIRS)
idf_component_get_property(esp_lcd_dir esp_lcd COMPONENT_DIR)

# sets the include paths into INCLUDES variable
if(esp_lcd_includes)
    list(TRANSFORM esp_lcd_includes PREPEND ${esp_lcd_dir}/)
    list(APPEND INCLUDES ${esp_lcd_includes})
endif()

# Add include directories.
target_include_directories(usermod_lcd_bus INTERFACE ${INCLUDES})
```

where you would change the `esp_lcd` to whatever component dependency your module has. 

doing that solves the component issue with ESP-IDF and user c modules. so no alterations need to be made to esp32_common.cmake in order to access components in the ESP-IDF

kdschlosser, 2023-12-18:
I hope this is helpful for others. There is a member of the LVGL forum (bdbarnett) that discovered that adding the include path solved the issue. I took it a step further than using a static include path by collecting the information from the ESP-IDF to build the include paths. So long as the component name stays the same it will work even if the paths change for the includes/component. 

so a big thanks goes out to bdbarnett for discovering this.

LnnrtS, 2023-12-18:
Thanks, this also works for me :+1:

LnnrtS, 2023-12-18:
Wrapped as cmake function

```
function(add_esp_component_includes)

    cmake_parse_arguments(COMPONENT_INCL "" "TARGET" "COMPONENTS" ${ARGV})

    foreach(_this_component ${COMPONENT_INCL_COMPONENTS})

        idf_component_get_property(_this_include ${_this_component} INCLUDE_DIRS)

        if (_this_include)
            idf_component_get_property(_this_dir ${_this_component} COMPONENT_DIR)
            list(TRANSFORM _this_include PREPEND ${_this_dir}/)
            target_include_directories(${COMPONENT_INCL_TARGET} INTERFACE ${_this_include})
        endif()

    endforeach()
endfunction()
```

to be used as

```
add_esp_component_includes(TARGET my_usermod_library COMPONENTS component_a component_b)
```

kdschlosser, 2023-12-18:
I would name the function something that falls in line with the context it would be used in. Something like `user_c_module_add_component` That's if this was to get added to the build system and didn't need to get added by the user.

jonnor, 2024-08-24:
It seems that a fix has been identified, including some build code recipe that C module writers can use. Can we close this issue?

kdschlosser, 2024-08-24:
The hope was to have this code added to the build system for the ESP32. There are many cases in which a user might need to access the underlying SDK from within a user C module and this would provide them the mechanics to be able to do it.

closed 2025-03-03 by kdschlosser