micropython/micropython #12976: Can you restore the following configuration for ESP32C3-USB ?

qadzhang, 2023-11-15:
Can you restore the following configuration for ESP32C3-USB? This was available in version 1.20, but not in version 1.21. This is necessary for using ESP32C3 development boards without a serial port chip.

micropython/ports/esp32/boards/ Generic_C3_USB

Thank you.

robert-hh, 2023-11-15:
The ESP32_GENERIC_C3 build works with a XIAO ESP32C3 device, which does **not** use a serial port chip. It works as well with a NODEMCU ESP32C3 device with a serial USB chip. 
Tested with v1.22.0-preview.147.

qadzhang, 2024-02-29:
Perhaps due to my limited skills, I am unable to make this device run using the ROM for LOLIN_C3_MINI, regardless of whether DIO is used. Could you please provide guidance on how to write the firmware to make a board without   a serial USB chip   work?

Here is my flashing command:

esptool --chip esp32c3 --port COM5 --baud 460800 write_flash -fm dio LOLIN_C3_MINI-20240222-v1.22.2.bin

qadzhang, 2024-10-08:
I'm just an idiot... I was able to connect later using PyCharm with the MicroPython plugin.
The ESP32-C3 core board is using LOLIN_C3_MINI-20240222-v1.22.2.bin, and there are no issues.

I'm really sorry.

closed 2024-10-08 by qadzhang