micropython/micropython #12980: _thread module freezing whole device (RPI_PICO_W)

innatecuriosity, 2023-11-15:
### Summary
When using firmware built from the master branch calling` _thread.start_new_thread` freezes the whole machine. v1.22.0-preview and earlier works as expected.

Using Raspberry Pi Pico W.

I went back to find the commit where it breaks, and i found the breaking point:  bcbdee235719d459a4cd60d51021454fba54cd0f
- 2d363a23cb9e825200e772f973eab921a69e0646 still works
- bcbdee235719d459a4cd60d51021454fba54cd0f does not work anymore

### Scenario
Im using a **Raspberry Pi Pico W** with firmware built from the repository (make etc.) In my application, right at the start of boot.py I started a process on the second thread that constantly measured the orientation, then continued as normal, but yesterday the whole program started freezing. 

After some testing I found out that calling `_thread.start_new_thread(test_function, ()) ` freezes the machine, it cannot be stopped with ctrl-c or any other method.

I tested with this file, then importing and running test()

```
import _thread

def test1():
    for i in  range(4):
        print(f"test1-{i}")

def test2():
    for i in  range(4):
        print(f"test2-{i}")

def test():
    print("Main thread:")
    test1()

    print("In second thread:")
    _thread.start_new_thread(test2, ())
    print("Main thread again:")
    test1()
```

With  2d363a23cb9e825200e772f973eab921a69e0646 and earlier I get the expected:
```
Main thread:
test1-0
test1-1
test1-2
test1-3
In second thread:
Main thread again:
test2-0
test1-0
test2-1
test1-1
test2-2
test1-2
test2-3
test1-3

```
But after **bcbdee235719d459a4cd60d51021454fba54cd0f**  I get: 
[image]
[image]

then freezes completely. Power cycling does make the device responsive again. 

Sorry about the format, first time creating an issue.

innatecuriosity, 2023-11-15:
Im using pymakr2 extension for VSCode, to connect to the device, maybe that plays a role

sufyan-mukadam, 2023-11-15:
I've has similar problems, the `_thread.start_new_thread` freezes my picow. I have reverted back to 1.21.0

CTGControls, 2023-12-28:
I had the same problem with 1.22.0 but everything works on 1.21.0

peterhinch, 2023-12-29:
See https://github.com/micropython/micropython/issues/13288

anasfeelgood, 2023-12-30:
I am having the same issue, after i soldered a dozen picos for a project, I started injecting code with thonny on all of them one by one. The 1.22 update came in the middle and I didn't notice while injecting. When i started testing them I thought at first that half my picos were defect, after a few hours losing my mind some intuition told me maybe it's not the same micropython i am running on all of them, googled, and found this page.

dpgeorge, 2024-01-02:
@innatecuriosity thanks for reporting this issue.  It is the same as that reported in #13288, and is fixed by #13312.

I should have taken more notice of this issue and investigated it before the release v1.22.0 was published.  This bug should have been a show-stopper for the release.

PR #13312 (2024-01-03): rp2: fix atomic section
This is an attempt to fix #12980 and #13288.

~~There are two fixes to the `MICROPY_BEGIN_ATOMIC_SECTION`/`
MICROPY_END_ATOMIC_SECTION` on the rp2 port:~~
- ~~Use a recursive mutex to prevent deadlocks on the same thread, in the case an IRQ races against the thread level (on the same core) when acquiring the atomic section (mutex may be taken but IRQs not yet disabled, then the IRQ waits forever trying to acquire the mutux).~~
- ~~Unlock the mutex if it was locked, not if `core1_entry` is still non-null, to fix the case where the second core finishes (and sets `core1_en [... cut]
closed 2024-01-03 by dpgeorge