micropython/micropython #12991: null-dereference in mod_binascii_b2a_base64

gwangmu, 2023-11-16:
# Description

A null-dereference was found in the UNIX port while attempting to use a `framebuf.FrameBuffer()` instance for `binascii.b2a_base64()`. Below is the only PoC we found for this bug.

# Proof of Concept

- `poc.py`
```
import array
import builtins
import binascii
import hashlib
import framebuf
v2 = builtins.property();
v4 = hashlib.md5();
v5 = v2.deleter(v4);
v8 = "hello";
v9 = "21796";
v10 = v8.replace(v9, v9);
v11 = array.array(v10);
v13 = array != btree;
v15 = v5 != builtins;
v16 = framebuf.FrameBuffer(v11, v15, v15, v13);
binascii.b2a_base64(v16);
```

- Execution
```
$ # build unix port with ASAN, at the root source code directory.
$ export CC=clang
$ export CXX=clang++
$ export CFLAGS="-fsanitize=address -fno-omit-frame-pointer"
$ export CXXFLAGS=$CFLAGS
$ export LDFLAGS=$CFLAGS
$ export DEBUG=1
$ make -C mpy-cross -j
$ make -C ports/unix -j all lib
$
$ # run a poc.
$ export ASAN_OPTIONS="detect_leaks=0"
$ ./ports/unix/build-standard/micropython poc.py
```

- Stack trace
```
=================================================================
==244625==ERROR: AddressSanitizer: SEGV on unknown address 0x000000000000 (pc 0x5598a3f1f5e2 bp 0x7ffc06f765c0 sp 0x7ffc06f764a0 T0)
==244625==The signal is caused by a READ memory access.
==244625==Hint: address points to the zero page.
    #0 0x5598a3f1f5e2 in mod_binascii_b2a_base64 ../../extmod/modbinascii.c:137
    #1 0x5598a3ee1ab9 in fun_builtin_var_call ../../py/objfun.c:114
    #2 0x5598a3ecc4ec in mp_call_function_n_kw ../../py/runtime.c:712
    #3 0x5598a3ecc96b in mp_call_method_n_kw ../../py/runtime.c:728
    #4 0x5598a3f1275d in mp_execute_bytecode ../../py/vm.c:1042
    #5 0x5598a3ee1e46 in fun_bc_call ../../py/objfun.c:273
    #6 0x5598a3ecc4ec in mp_call_function_n_kw ../../py/runtime.c:712
    #7 0x5598a3ecc546 in mp_call_function_0 ../../py/runtime.c:686
    #8 0x5598a3fdacfd in execute_from_lexer /workspaces/Latest/micropython/ports/unix/main.c:160
    #9 0x5598a3fdadc2 in do_file /workspaces/Latest/micropython/ports/unix/main.c:309
    #10 0x5598a3fdbded in main_ /workspaces/Latest/micropython/ports/unix/main.c:721
    #11 0x5598a3fdc21a in main /workspaces/Latest/micropython/ports/unix/main.c:476
    #12 0x7f4ec1944d8f in __libc_start_call_main ../sysdeps/nptl/libc_start_call_main.h:58
    #13 0x7f4ec1944e3f in __libc_start_main_impl ../csu/libc-start.c:392
    #14 0x5598a3e8b4c4 in _start (/workspaces/Latest/micropython/ports/unix/build-standard/micropython+0x854c4)

AddressSanitizer can not provide additional info.
SUMMARY: AddressSanitizer: SEGV ../../extmod/modbinascii.c:137 in mod_binascii_b2a_base64
==244625==ABORTING
```

# Environment

Ubuntu 20.04
Intel(R) Xeon(R) Gold 5218 CPU @ 2.30GHz
Memory: 64 GB

# Affected Version

v1.20.0 (commit a3862e726, latest as of 2023-09-26)
v1.20.0 (commit 813d559bc, 2023-06-19)
Discovered in the UNIX port version.

agatti, 2024-09-29:
@gwangmu Can you please confirm the POC is correct?  It tries to access the `btree` module without importing it, which stops execution right there and execution never gets to the framebuffer crash you report.  And after adding the `btree` module import line myself after `import framebuffer` it doesn't seem to crash on my machine:

```bash
$ make VARIANT=standard -j 4
...

$ ./build-standard/micropython
MicroPython 17d823445 on 2024-09-29; linux [GCC 14.2.1] version
Use Ctrl-D to exit, Ctrl-E for paste mode
>>> 

$ ./build-standard/micropython poc.py
Traceback (most recent call last):
  File "poc.py", line 16, in <module>         # line 16 is `v16 = framebuf.FrameBuffer(v11, v15, v15, v13);`
ValueError: 

$
```

gwangmu, 2024-10-02:
@agatti I'm afraid I lost my testing environment, so I can't test the PoC either. Sorry. But I noticed that you're using the version (2024-09-29) a year after the version that we checked the bug (2023-09-26). I think it's possible that the PoC was invalidated during that while.

agatti, 2024-10-02:
That's a fair point, thanks.  I'm still puzzled on why it crashed in the place you reported back then with that PoC, since execution should have stopped at `v13 = array != btree;`.  Same thing in your PoC for #12992, in the line `v6 = select != builtins;` as the `select` module wasn't imported first.