micropython/micropython #12995: asyncio stream.py: Is async context manager code correct?

peterhinch, 2023-11-16:
[Link to code](https://github.com/micropython/micropython/blob/8b1980ad450ea2b34af4b831a8c655ba08dee800/extmod/asyncio/stream.py#L19C1-L24C1)
```py
    async def __aexit__(self, exc_type, exc, tb):
        await self.close()

    def close(self):
        pass
```
If used in an asynchronous context manager it seems that the socket is not closed.

dpgeorge, 2023-11-22:
Looking in to this, it seems MicroPython's `StreamReader` and `StreamWriter` do not work as an async context manager, ie with `async with`.  It looks like they never did work and have no tests for this functionality.

Also as you mention, CPython doesn't support this.

So I think we should just delete both `__aenter__` and `__aexit__` from `Stream`.

-----

Regarding `StreamWriter.close()`: the CPython docs also say:

    The method closes the stream and the underlying socket.

    The method should be used, though not mandatory, along with the wait_closed() method:

So apparently you don't need to call `.close()`, only `.wait_closed()`?  I find that strange, but that's what the docs say.  Therefore we need to actually close the socket in `.wait_closed()` (but I guess we could also close it in `.close()`, it won't hurt to close twice).

The CPython docs also don't mention anything about `StreamReader.close()`... I wonder what's going on there.

peterhinch, 2023-11-23:
> So I think we should just delete both __aenter__ and __aexit__ from Stream.

Agreed. Any thoughts on supporting `async for`?

dpgeorge, 2023-12-20:
See #13234 which removes these methods.

> Any thoughts on supporting `async for`?

I guess iterating on a stream uses readline...?  We could implement this but it obviously adds code and I don't think anyone missed it yet.

commit f46269a1 (2023-12-20): extmod/asyncio: Remove non-working Stream __aenter__/__aexit__ methods.

It looks like these never worked and there are no tests for this
functionality.  Furthermore, CPython doesn't support this.

Fixes #12995.
PR #13234 (2023-12-20): extmod/asyncio: Remove non-working Stream __aenter__/__aexit__ methods.
It looks like these never worked and there are no tests for this functionality.  Furthermore, CPython doesn't support this.

Fixes #12995.
closed 2023-12-20 by dpgeorge