micropython/micropython #13006: heap-buffer-overflow: incorrect length choice on memcpy at vfs unmount

junwha0511, 2023-11-17:
# Summary

- **OS**: Ubuntu 22.04
- **version**: micropython@a00c9d56db775ee5fc14c2db60eb07bab8e872dd
- **port**: unix
- **contribution**: Junwha Hong and Wonil Jang @S2Lab, UNIST
- **description**:
    - at `mp_vfs_umount` extmod/vfs.c:276, the comparison between unmount string and previous mounts are performed based on the length of the unmount string
    - thus,
        - there is always global-buffer-overflow for non “/” unmount with the mount “/”
        - there is heap-buffer-overflow if longer mount strings unmount earlier than shorter mount strings

# PoC

```cpp
# Test for VfsLittle using a RAM device, with mount/umount

try:
    import os

    os.VfsLfs1
except (ImportError, AttributeError):
    print("SKIP")
    raise SystemExit

class RAMBlockDevice:
    ERASE_BLOCK_SIZE = 1024

    def __init__(self, blocks):
        self.data = bytearray(blocks * self.ERASE_BLOCK_SIZE)

    def readblocks(self, block, buf, off=0):
        addr = block * self.ERASE_BLOCK_SIZE + off
        for i in range(len(buf)):
            buf[i] = self.data[addr + i]

    def writeblocks(self, block, buf, off=0):
        addr = block * self.ERASE_BLOCK_SIZE + off
        for i in range(len(buf)):
            self.data[addr + i] = buf[i]

    def ioctl(self, op, arg):
        if op == 4:  # block count
            return len(self.data) // self.ERASE_BLOCK_SIZE
        if op == 5:  # block size
            return self.ERASE_BLOCK_SIZE
        if op == 6:  # erase block
            return 0

def test(vfs_class):
    bdev = RAMBlockDevice(10)
    vfs_class.mkfs(bdev)  # construction
    vfs = vfs_class(bdev)

# mount
    os.mount(vfs, "/lfs")
# umount
    os.umount("/lffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffs")

import sys

sys.path.clear()
sys.path.append("/lfs")
sys.path.append("")

# run tests
test(os.VfsLfs1)
```

```cpp
micropython bof_vfs_unmount.py
```

# Problem Statement

at `mp_vfs_umount` extmod/vfs.c:276, the comparison is performed based on the string of the request of unmount, but the string of one of the previous mount string can be shorter than the string of unmount.

```c
mp_obj_t mp_vfs_umount(mp_obj_t mnt_in) {
    // remove vfs from the mount table
    mp_vfs_mount_t *vfs = NULL;
    size_t mnt_len;
    const char *mnt_str = NULL;
    if (mp_obj_is_str(mnt_in)) {
        **mnt_str = mp_obj_str_get_data(mnt_in, &mnt_len);**
    }
    for (mp_vfs_mount_t **vfsp = &MP_STATE_VM(vfs_mount_table); *vfsp != NULL; vfsp = &(*vfsp)->next) {
        **if ((mnt_str != NULL && !memcmp(mnt_str, (*vfsp)->str, mnt_len + 1)) || (*vfsp)->obj == mnt_in) {**
            vfs = *vfsp;
            *vfsp = (*vfsp)->next;
            break;
        }
    }

    if (vfs == NULL) {
        mp_raise_OSError(MP_EINVAL);
    }

    // if we unmounted the current device then set current to root
    if (MP_STATE_VM(vfs_cur) == vfs) {
        MP_STATE_VM(vfs_cur) = MP_VFS_ROOT;
    }

    // call the underlying object to do any unmounting operation
    mp_vfs_proxy_call(vfs, MP_QSTR_umount, 0, NULL);

    return mp_const_none;
}
```

In our PoC, the comparison was performed between “/lfs” and “lffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffs”, using the length of the later one. and because “/lfs” is a part of the qstr which is allocated from `qstr_from_strn`, the actual buffer overflow occurs.

Also, it lead to global buffer overflow at the next for loop, because it compare the string with “/”.

Thus, to sum up,

- there is always global-buffer-overflow for non “/” unmount with the mount “/”
- there is heap-buffer-overflow if longer mount strings unmount earlier than shorter mount strings

# Crash Log

```c
#1 0x5555557c96c1 in mp_vfs_umount /home/qbit/testing-2023/micropython/ports/unix/../../extmod/vfs.c:276:34
#2 0x555555782c1c in mp_execute_bytecode /home/qbit/testing-2023/micropython/ports/unix/../../py/vm.c:1042:21
#3 0x55555574261b in fun_bc_call /home/qbit/testing-2023/micropython/ports/unix/../../py/objfun.c:273:42
#4 0x555555781b2c in mp_execute_bytecode /home/qbit/testing-2023/micropython/ports/unix/../../py/vm.c:957:21
#5 0x55555574261b in fun_bc_call /home/qbit/testing-2023/micropython/ports/unix/../../py/objfun.c:273:42
#6 0x555555903f3d in execute_from_lexer /home/qbit/testing-2023/micropython/ports/unix/main.c:161:13
#7 0x555555902ad5 in do_file /home/qbit/testing-2023/micropython/ports/unix/main.c:310:12
#8 0x555555902ad5 in main_ /home/qbit/testing-2023/micropython/ports/unix/main.c:722:19
#9 0x7ffff7c29d8f in __libc_start_call_main csu/../sysdeps/nptl/libc_start_call_main.h:58:16
#10 0x7ffff7c29e3f in __libc_start_main csu/../csu/libc-start.c:392:3
#11 0x555555593a34 in _start (/home/qbit/testing-2023/micropython/ports/unix/build-standard/micropython+0x3fa34)
```

# Patch

we need to use “min” length between them, or compare the length first.

```c
mp_obj_t mp_vfs_umount(mp_obj_t mnt_in) {
    // remove vfs from the mount table
    mp_vfs_mount_t *vfs = NULL;
    size_t mnt_len;
    const char *mnt_str = NULL;
    if (mp_obj_is_str(mnt_in)) {
        mnt_str = mp_obj_str_get_data(mnt_in, &mnt_len);
    }
    for (mp_vfs_mount_t **vfsp = &MP_STATE_VM(vfs_mount_table); *vfsp != NULL; vfsp = &(*vfsp)->next) {
        if ((mnt_str != NULL && !memcmp(mnt_str, (*vfsp)->str, **min((*vfsp)->len, mnt_len + 1))**) || (*vfsp)->obj == mnt_in) {
            vfs = *vfsp;
            *vfsp = (*vfsp)->next;
            break;
        }
    }
...
```

***Thank you for taking the time to review our bug report! :)***

projectgus, 2023-11-21:
@junwha0511,

I feel bad for closing all the .mpy-related bug reports on you, so I will say quickly - thank you for finding this bug and reporting it, and for the clarity of all your bug reports. It will be fixed! :100:

dpgeorge, 2024-07-23:
Fixed by 29943546343c92334e8518695a11fc0e2ceea68b

closed 2024-07-23 by dpgeorge