micropython/micropython #13007: heap-buffer-overflow: mis-interpretation of float as int at slice_indices

junwha0511, 2023-11-17:
# Summary

- **OS**: Ubuntu 22.04
- **version**: micropython@a00c9d56db775ee5fc14c2db60eb07bab8e872dd
- **port**: unix
- **contribution**: Junwha Hong and Wonil Jang @S2Lab, UNIST
- **description**: `slice_indices` misinterpret float value as integer value, and leads to buffer overflow.

# PoC

```c
# A claan item key
class A:
    def __getitem__(self, idx):
        return idx

print(A()[:].indices(.0))
```

# Expected result from Python 3.10

```c
>>> # A claan item key
>>> class A:
...     def __getitem__(self, idx):
...         return idx
... 
>>> print(A()[:].indices(.0))
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: 'float' object cannot be interpreted as an integer
```

# Problem Statement

The problem occurs because the indices function does not handle float as an exception 

## Allocation

The .0 is allocated as a 16-bytes object by `mp_obj_new_float` py/objfloat.c:197 

In our debugging, the address range of this chunk is `[0x7fffef005760,0x7fffef005770)`.

## Access

At `slice_indices` py/objslice.c:57:23, it tries to interpret the length_obj as int.

```c
STATIC mp_obj_t slice_indices(mp_obj_t self_in, mp_obj_t length_obj) {
    mp_int_t length = mp_obj_int_get_checked(length_obj);
```

Now, `mp_obj_int_get_checked` parameterizes `&self→mpz` into `mpz_as_int_checked`, which is 8-bytes offset from `self_in` , `0x7fffef005768`

```c
mp_int_t mp_obj_int_get_checked(mp_const_obj_t self_in) {
    if (mp_obj_is_small_int(self_in)) {
        return MP_OBJ_SMALL_INT_VALUE(self_in);
    } else {
        const mp_obj_int_t *self = MP_OBJ_TO_PTR(self_in);
        mp_int_t value;
        if (mpz_as_int_checked(&self->mpz, &value)) {
```

At `mpz_as_int_checked` py/mpz.c:1553, the `dig` field of mpz  `i` is parsed, which is `0x7fffef005778` in our debugging, thus overflowed  `[0x7fffef005760,0x7fffef005770`.

```c
bool mpz_as_int_checked(const mpz_t *i, mp_int_t *value) {
    mp_uint_t val = 0;
    mpz_dig_t *d = i->dig + i->len;
```

To sum up, heap-buffer-overflow occurs from type-confusion between `int` and `float` object 

# Crash Log

```c
#0 0x5555556ae08b in mpz_as_int_checked /home/qbit/testing-2023/micropython/ports/unix/../../py/mpz.c:1553:23
#1 0x555555746a46 in mp_obj_int_get_checked /home/qbit/testing-2023/micropython/ports/unix/../../py/objint_mpz.c:427:13
#2 0x5555557522ff in slice_indices /home/qbit/testing-2023/micropython/ports/unix/../../py/objslice.c:57:23
#3 0x555555782c1c in mp_execute_bytecode /home/qbit/testing-2023/micropython/ports/unix/../../py/vm.c:1042:21
#4 0x55555574261b in fun_bc_call /home/qbit/testing-2023/micropython/ports/unix/../../py/objfun.c:273:42
#5 0x555555903f3d in execute_from_lexer /home/qbit/testing-2023/micropython/ports/unix/main.c:161:13
#6 0x555555902ad5 in do_file /home/qbit/testing-2023/micropython/ports/unix/main.c:310:12
#7 0x555555902ad5 in main_ /home/qbit/testing-2023/micropython/ports/unix/main.c:722:19
#8 0x7ffff7c29d8f in __libc_start_call_main csu/../sysdeps/nptl/libc_start_call_main.h:58:16
#9 0x7ffff7c29e3f in __libc_start_main csu/../csu/libc-start.c:392:3
#10 0x555555593a34 in _start (/home/qbit/testing-2023/micropython/ports/unix/build-standard/micropython+0x3fa34)
```

# Patch

we can handle only int object at `slice_indices` function, or just cast the float value. 

***Thank you for taking the time to review our bug report! :)***

dpgeorge, 2023-11-21:
Thank you for the bug report, and the detailed steps to reproduce the issue.

Should be fixed by #13039.

closed 2023-11-22 by dpgeorge

junwha0511, 2023-11-22:
Thank you for your effort on this issue!:)