micropython/micropython #13024: Feature-request: Put the GENERIC download on the ESP32 web page, and move all non-critical modules to vfs

gitcnd, 2023-11-19:
Please put the ESP32_GENERIC link into this page: https://micropython.org/download/?port=esp32

Please make the ESP32_GENERIC build only contain these modules in the firmware (with suitable adjustments if needed so auto-detect when needed, so the firmware works on boards potentially missing that hardware):

Dual-core, External Flash, External RAM, DAC, WiFi

Please move all of these (plus any others) into the VFS partition, and allow us to specify the pin(s) etc they use ourselves:

Audio Codec, BLE, Battery Charging, CAN, Camera, Display, Environment Sensor, Ethernet, Feather, IMU, JST-PH, JST-SH, LoRa, Microphone, PoE, RGB LED, SDCard, Secure Element, USB, USB-C, microSD, mikroBUS...

Please make 2 release available: the main one, which includes the VFS partition (flashing this will erase all user files), and a minimal one, which does not include the VFS partition (flashing this leaves user files alone).

The above feature-requests maximize all tradeoffs between code-size, feature-support, and hardware-support, while simultaneously reducing the vastly unnecessary number of builds and firmware releases: everyone can use the same firmware on all boards, with just different VFS for all the different boards out there.

I expect this would be a minor change to the build process (omitting the step which merges the .mpy modules into the firmware, and replacing it with one which puts them into a vsf.bin file instead).

gitcnd, 2023-11-19:
That is to say - put some text someplace on https://micropython.org/download/?port=esp32 (maybe as the first link in the "Vendor" list) that says "GENERIC" - so users of boards not-pictured know which link to use (being https://micropython.org/download/ESP32_GENERIC/ )

andrewleech, 2023-11-19:
The modules listed refer to on-board features. The whole point of having multiple board definitions is to provide users a simple install for their hardware that has most of these features ready to use! 
Many new users don't know how to find and read schematic diagrams to define their own pinouts for things, so having them already defined in a board profile is a huge advantage for them.

Also, most of those "modules" that have support included in the build are compiled into the codebase in C so can't just be copied onto a filesystem instead. 
Other modules / drivers that are written in python are currently frozen into the firmware which provides a significant ram saving over having them on VFS.

There is the generic build for original esp32 still, though I guess existing users who are used to looking for the Generic label might not realise it's now called "ESP32 / WROOM
Espressif". 
I can see making the "generic' builds in a separate section at the top of the page could make it clearer... though this page is auto generated so not a simple change I guess.

gitcnd, 2023-11-21:
The main problem with "simple" is you assume you've somehow defined a board for every bit of hardware out there - I've literally got around a dozen different ESP32 boards, and not a single one appears in your list.  I usually buy inexpensive stuff from AliExpress, so it's actually amazing how such common boards aren't listed?

The next problem with "simple", is that "users" are going to have to re-flash their entire firmware whenever they need a fix for a component (not to mention: you're going to have to build a whole new firmware for it too) - all of which would be unnecessary if they could just get the relevant newer .mpy file when needed.  reflashing instead of updating a file is the opposite of simple.

"ram" and "flash" are 2 different things - I think you meant "flash" when you said "ram" before?  Piling things that are not being used into the flash, absolutely is wasting both flash *and* ram (depending on how the underlying paging system works).  Nobody is ever going to use every single feature you offer, all at once, so you can absolutely guarantee that lumping it all into the firmware is severely restricting ram (and vfs, for users who want to erase unneeded modules).  Somehow, micro-python is using more than 75% of my 520kb ram... it's worth moving unused stuff out of flash, so it never gets paged in to ram when it's not being used (flash blocks are quite large)... plus it makes the firmware smaller, so you can have a larger vfs as well.

supplying the driver files, as files (rather than baked in to the firmware), would of course include the board configuration (pins) as well, so there's no change with any of that: from a users point of view, they still flash their chip that first time - the only difference to them, is there's a smaller firmware component and a larger vfs component that now holds some extra files.

the build system is already plenty powerful enough to make custom C .mpy binary files, and already does, so the "can't just be copied onto a filesystem instead." statement is simply not true (notwithstanding minor adjustments to build them that way)

andrewleech, 2023-11-23:
> "ram" and "flash" are 2 different things - I think you meant "flash" when you said "ram" before

No, I meant ram. Having mpy files on vfs means they need to be loaded into ram before they can be run. When they're compiled into flash they can often be exectured directly from flash (on most ports), resulting in significant ram savings at runtime. I'm not sure whether esp32xx can execute C / frozen code from flash or needs to load into ram... stm32 certainly can (both internal and external flash).

More broadly, you're making a lot of assumptions about the capabilities of the dynamic C module (natmod) framwork, might want to look a bit closer at the background and functionality available. I've used and contributed to both the user c module and natmod interfaces a lot over the years. Most C modules _cannot_ be trivially turned into mpy files. 

The dynamic C module framework is quite powerful, but also constrained and has a number of limitations. 
It cannot provide access to all public functions in micropython, nor can it provide direct access to hardware, nor can it provide access to "shared libraries" like soft-float, nor can the generated mpy files be conveniently mip installed.

For a micropython codebase C function to be accessible in a natmod, it either:
* needs to be listed in the public natmod api which takes up extra flash space, and any change to this uprevs the natmod api version, which breaks compatibility with previously built modules.
* or needs to be exposed and called via a python interface / api - not all functions are suitable to be exposed as python functions however and adding a python interface to a C function adds code size, ram use and small amounts of extra execution time.

Every separate natmod that wants to use a hardware / hal / manufacturer library needs its own copy of said library (eg. https://github.com/orgs/micropython/discussions/12811 ). This can result in a lot of duplication of libraries if you're not careful.

Any and every separate natmod that needs to use `float` on a chip without hardware float capabiliy needs its own copy of the softfloat libraries (can be ~20KB each time https://github.com/orgs/micropython/discussions/9640). 

Also things like vfs can't talk to spi can't talk to dma in natmods without published public api's for all of these - which could be built but adds to code space use, adds extra function layers (so slows execution) and cannot trivially be abstracted to a common api that works on all ports (eg. dma's on different chips are all wildly different), so instead these things get implemented directly in tightly coupled C per-port and "hidden" behind a simple python api. 

Currently dynamic c mpy files cannot be frozen into firmware (I know you're against this anyway) but they also can't be handled by the mip install system currently nor is there a way to robustly build and declare their architecture to control the way they're installed to help users install the arch mpy that matches their chip. This can certainly be built with planning, time, resources etc... hasn't happened yet though.
If you can find a way to fix some of these and other limitations I'd be very interesteed in reviewing the PR's!

A significant proportion of micropython users (professional ones in particular) are _not_ just using random cheap stuff from aliexpress, we use reliable sources of trusted hardware and/or things designed in house with reliable chipsets, and compiling and flashing boards with updates is a one click operation that is standard practice for anyone experienced in embedded.

The cheap stuff from ali usually comes and goes in single batch builds, it's not the micropython projects' responsibility to try to keep up with them - it should be the board manufacturers responsibility to submit and maintain a board definition if they want to promote micropython support. Whether said board def is in core firmware or post-loaded in a separate step is a moot point. 

There are lots of different workflows used by people developing on micropython, there's no one correct way to work. Flashing firmware files that contain everything needed to run an application is easiest for many people, others prefer to copy a python / mpy file on and run that, others prefer to use mpremote mount and run things directly from pc. All are supported and generally work well for most people. 
Slicing and dicing the C core of micropython into tens if not hundreds of modules that need to be assembled somehow and the correct ones loaded on to meet the needs of a particular board while ensuring the version / arch / compatibility of all of them is in sync would likely require a dramatic overhaul of the board, build and deployment systems for micropython. 
If you're keen to implement this I'd be quite interested in seeing how it all fits together in practice!

jonnor, 2024-07-03:
Ask discussed above, this is not really feasible with current state of MicroPython. There have been talks of enabling native modules, which would be one prerequisite for something like this. But even with that, there would need to be much more APIs exposed to enable drivers/components which do I/O and access peripherals. So this will have to wait a few years, at least.

closed 2024-07-05 by projectgus