micropython/micropython #13026: Supplying a custom partition file

kdschlosser, 2023-11-20:
I don't believe this exists as I have not been able to locate any information about it. So here goes.

What I would like to do is to be able to compile MicroPython with several user c modules. The idea is to be able to have MicroPython as a submodule in my repository and I would be able to compile it using a single command without having to alter or modify any of the MicroPython files at all. This would make it a whole lot easier for the user and also for any developer wanting to package MicroPython with any additions they may have made. 

The problem I am running into is the partition is not large enough to store MicroPython after my user c module has been added to it. I don't want to duplicate all of the different boards that are available only to have some small changes made to the partitions file.

It would be nice if I was able to provide a path to a partitions file from the command line. Same idea as how the user c modules are compiled. I was kind of surprised that this is not something that is available already. The availability of not needing to modify the build system in MicroPython when adding a user c module I believe would make a developers job so much easier to not only write code for but also make it a lot easier from a maintenance standpoint. It would allow a developer to not have to pull the latest released version of micropython into their project and then update the build system in order to get their project to compile. It would help to stream line the process.

In my specific use case the boards that are supplied with micropython are fine it's just the partition that is made is not correct. 
I have not looked to see what MCU's have the partition files. I know the ESP32 does and it might be the only one that does.

andrewleech, 2023-11-20:
It is only esp platform that uses the partition file, the file is a requirement of the idf.

I use micropython (unmodified) as a submodule alongside custom user_c_modules in all my work projects, this is certainly how it's expected to work.

I guess the difference is I'm usually building just one or two custom board profiles that are also in my repo, not trying to mass-build existing boards with different build settings.

Like the other issue you raised recently this sounds like something else that would be easy to fix with make, but wildly complicated with cmake. Sorry but I'm not sure how easy or hard it would be to inject different partition description files, nor do I know how many different ones you'd need to maintain for the different esp chip types etc.

kdschlosser, 2023-11-20:
I was looking at it and cmake is a real hinderance I will say that. It makes things really difficult that's for sure.

andrewleech, 2023-11-20:
Yeah I hate that esp (and rp2040) manufacturers have locked their sdk's into using it, it ends up being so restrictive.

kdschlosser, 2023-11-20:
I am having a hell of a time trying to figure out why cmake isn't running `add_custom_command`. I have tried adding dependencies to the interface, I have tried adding a target of `ALL`. It is simply not running. When it doesn't run and a source file gets added that is supposed to be created it errors out. I know cmake is processing the `add_custom_command` because I have placed a message before it and after it. I am having it run a python file and in the python file I have to creating a file that has the passed in arguments written to it. I know the python file works because I have tried it and there are no errors. No file ever gets created because the python script never gets run. So the source file gets added to the compilation causing it to trigger an error because it doesn't exist.

closed 2025-03-03 by kdschlosser