micropython/micropython #13042: machine.xxxsleep(): Inconsistent behavior with a negatve number as argument.

robert-hh, 2023-11-21:
When the argument to machine.lightsleep() or machine.deepsleep() is negative, the ports behave different. Even if that is an unreasonable value, the behavior should be consistent and well behaving.

stm32: return shortly after the call. The time seems not to depend on the value.
renesas-ra: treat negative numbers like 4 ms.
rp2: wait in a call to mp_hal_delay_ms() until the numbers match.
mimxrt: behave as if being called without an argument. Get woken by an interrupt at the WAKEUP pin.
samd:  behave as if being called without an argument. Wake up if a Pin irq happens.
esp32: Ignore the value and return immediately
esp8266;  Ignore the value and return immediately

So the best options are either a) raising an error or b ) silent immediate return. I can make a PR for any of them, and I prefer b).

jonnor, 2025-03-03:
Silent immediate return seems fine. Any objections @dpgeorge or @projectgus  ?

Josverl, 2025-03-06:
Or raise an `ENODELOREAN`

projectgus, 2025-03-10:
I don't feel strongly about it, but it seems to me that passing a negative value here may indicate a bug in the calling code - unless there's a common situation where a timeout value of -1 might get passed through to lightsleep?

If it's a bug in the calling code then it's probably better to fail here and draw the developer's attention to it rather than silently keep running and potentially mask it. But if it's convenient for something like timeout case then silent immediate return seems fine, too.

projectgus, 2025-03-10:
> If it's a bug in the calling code then it's probably better to fail here and draw the developer's attention to it

Muddying the waters, it seems `time.sleep(-1)` fails on both CPython and some MicroPython ports. And`time.sleep_ms(-1)` is also inconsistent.

Following CPython in all these cases seems (to me) like the best approach. But, again, I don't feel strongly about it.

robert-hh, 2025-03-15:
Sometimes looking into the code answers a question. extmod/modtime.c handles time.sleep(), time.sleep_ms() and time.sleep_us(). There argument values < 0 for time.sleep_ms() and time.sleep_us() are simply ignored. Adding a check for time.sleep() and raising an error would have to be done there. Still there would be a inconsistent behavior in the port's mp_hal_sleep_xx() implementations, but these would not show up at the Python API.
And yes, raising an error may be the better approach, since it has the chance to show a coding error.