micropython/micropython #13051: RP2 I2S  does not allow 24 bit data.  Is there a work around?

frohro, 2023-11-22:
I want to read I2S data using micropython and the RP2 from the PCM1804 ADC.  Unfortunately, it only sends data in the 24-bit format of I2S.  I looked at  the [source](https://github.com/micropython/micropython/blob/master/ports/rp2/machine_i2s.c) for i2s.c,
and at the [docs](https://docs.micropython.org/en/latest/library/machine.I2S.html) and it is evident that only 16 and 32-bit formats for I2S can be used, which leaves me "high and dry".   I had a look at the relatively recent C++ examples [here.](https://github.com/malacalypse/rp2040_i2s_example)  They make me think that by changing the system clock to 129600000 Hz, 24-bit transfers should be possible.  I was able to do that just fine with the following code:
`import machine

print(machine.freq())

machine.freq(129600000)

print(machine.freq())`

Is this something that is likely to be added soon, or is my understanding incorrect that it is possible, or are there other issues that make it hard?  Also, what are possible workarounds?  The ones I have thought of are switching MCUs to the ESP32 or another from the RP2 (maybe the ESP32 has the same 24-bit problem), ditching micropython in favor of C++ (not optimum, because my design goals include using micropython, as I have found it is much easier for those using my design than working with C++), trying to add 24-bit I2S to micropython myself (seems like a lot more difficult than switching MCUs), changing to a different ADC (not as good for my design as switching MCUs).  I really like the RP2, but this little defect is major for my design.

frohro, 2023-11-23:
It is tempting to think that changing the PIO numbers in set( x, 14) to set( x, 22) on lines 110 and 115 of [machine_i2s.c](https://github.com/micropython/micropython/blob/master/ports/rp2/machine_i2s.c) would change from 16 bits to 24.  It cannot be just this simple, or it would have been fixed long ago.

frohro, 2023-11-23:
It must be difficult to generate the BCK for 24 bits.  I haven't recognized where this is done yet, but it is what I'm checking on next.

andrewleech, 2023-11-23:
I wouldn't have thought the clock rate would need to be any different for 16 / 24 / 32 bit support - surely clock rate only needs to be matching the capabilities of the AC used?

It wouldn't surprise me at all if just changing the number of bits being read/copied in one place eg. PIO is all that's needed, it presumably hasn't been done yet because no one's needed it enough to make and submit the change!

The current implementation was contributed in https://github.com/micropython/micropython/commit/b6dbbbe82f7114cb5b56c54e916e304f416cd47a where I'm guessing the author only used / needed 16 and 32 bit support.

I've never used I2S on micropython myself though so really don't know for sure sorry!

frohro, 2023-11-23:
Yes, thanks for the reply.  I looked at other micropython machine variants now, and they ALL have this limitation, so the port to RP2 is not really the root cause of the problem.  The clocks do need to change when you are doing 3/2 as many bits; the bit timing has to happen 3/2 faster.  This is also explained in the protocol specs from Philips and perhaps more concisely on [Wikipedia](I²S. (2023, October 13). In Wikipedia. https://en.wikipedia.org/wiki/I%C2%B2S).  The thing that makes 16 and 32 easier to do is that they are just twice as fast.  Fortunately my part lets me pick the SCLK to be 384*fs instead of only 256*fs or 128*fs, so I think if I change the RP2 main system clock to 129600000 and use a divider of 7.03 somehow it might work.  I haven't figured out yet how to do that in PIO, but I suspect that because the authors of [this C++ I2S](https://github.com/malacalypse/rp2040_i2s_example) say so, and I think likely the closer I can come to an integer divider, the better accuracy I will get on my ADC speed.  This is a pretty steep learning curve for me.  It is too bad someone else, better prepared didn't come along earlier that needed 24 bit I2S data.  I guess there is a good reason docs say this I2S micropython library is a "Technical Preview."

miketeachman, 2023-11-23:
The existing I2S implementation should work:  
- read the ADC in 32 bit mode
- the receiving bytearray `buf` will contain the 24-bit audio data samples, in little-endian byte ordering, as shown below.

```
I2S SD line                                                                             MicroPython buf[]
=======                                                                                 ============
1st byte (Most significant byte, first to be transmitted from the ADC)                  buf[i+3]
2nd byte                                                                                buf[i+2]
3rd byte (Least significant byte of data transmitted from the ADC)                      buf[i+1]
4th byte (pad byte. ADC may output 0, or random data)                                   buf[i]              
```
- discard the pad byte.  The remaining 3 bytes contain the 24 bit sample data.  

For a more in-depth understanding on how this works take a look at the following references:
- Figure 37 (page 22) of the PCM1804 ADC datasheet which shows padding bits following the 24 bits of significant data.  The pad bits are shown as dashes
- Section 3.1 "Serial Data" of the I2S Bus Specification document (Feb 1996).  In particular, focus on the distinction between the `system word length` and `transmitter word length`.  That is, 24 bits of sample data are transported in a 32-bit transmission frame.  The vast majority of I2S ADCs work this way.

Hopefully this explanation above is enough to get you further in your project.  Please report back on your progress.

frohro, 2023-11-23:
The docs say this a "Technical Preview" and ask for comments on the API.  Here are some:

- A good number of sound ADC/DAC datasheets need to be read to make sure the API will work with as many as possible.

- At a minimum, it needs to work with 24-bit as well as 16 and 32.  All the good candidate ADCs I have looked at use 24-bit.
- Some manufacturers change the phase of the BCK that data is read out on and read into on.  (Usually, on one edge of the clock the data is loaded, and on the other edge it is valid and stable, and so is read.  This needs to be software-selectable.
- There needs to be an option for the SCLK signal (master clock signal) so it is selectable what the multiplier on the sample rate is, (not only 256).  For example, the PCM1808 and PCM1804 can use 64, 128, 256, 384, and 512.  
- There should be a way to select if the RP2 is the master or slave.
- There should be a note in the documentation on which i2S devices this has been tested and is working on, especially in the "Technical Preview" period.

frohro, 2023-11-23:
Thanks Mike!
I missed this.  I will try it on the PCM1808.  Your note was very helpful.

frohro, 2023-11-24:
You can see my results so far, at the tail end of this encouraging question [here.](https://github.com/orgs/micropython/discussions/11675).  A quick summary is that after building it up and testing, the PWM generated SCKI is not synchronous with the other signals, LRCK and BCK, which together are synchronous.