micropython/micropython #13063: Nucleo-H743ZI2 Micropython Hang up

MartinManzinger, 2023-11-24:
I wanted to flash micropython onto a Nucleo-H743ZI2 development board and micropython seems to reset the controller after startup.

**Problem description:**
I flashed the STM32H743 controller with the release built (v1.21.0 (2023-10-05) [.hex])
The firmware download itself seems to work, because I get the typical led sequencing when selecting safe boot modes of micropython (https://docs.micropython.org/en/latest/pyboard/tutorial/reset.html). This startup part is working so far. Also when I select factory reset to restore the file system that seems to work, indicated by the led behaviour.
After that the problems begin. The yellow LED glows up lightly. And I get no REPL interface. Neither on the STLink USB port (that is routed to the UART of the main controller) nor get I even a virtual com port on the User USB port, that is connected directly to the main controller. I also get no flash drive on my pc.
Another indication, that micropython resets the main controller after startup is, that when I press the USR button, it immediately enters those boot sequence led state. Without requiring me pressing Reset (See video).
I checked the yellow LED with an oscilloscope and it turns on for a short duration and is off most of the time. The frequency is about 300Hz. As far as I know, the LED should stay off when micropython is running and no python code activates it.

**Things I already tried, that all had the same results:**
- Using two different Nucleo-H743ZI2 boards to exclude a hardware failure
- Trying different versions of release firmwares (https://micropython.org/download/NUCLEO_H743ZI2/)
- Trying release firmware of the older NUCLEO_H743ZI board
- Cloning the actual micropython repository and compile my own firmware binaries
- Flash controller with dfu-programming with dfu-util (on User USB port)
- Flash controller with hex-file programming with STM32Cube Programmer (on STLink USB port)
- Using a benchtop power supply to exclude a problem with power supply
- Using a different PC all together

**My ideas what could be wrong:**
- Watchdog resets the controller

https://github.com/micropython/micropython/assets/35213758/452a96e5-a0f8-4972-9f36-bc58de680550

MartinManzinger, 2023-12-06:
I tried to locate what code causes the error (reset). In **ports/stm32/main.c** I put a while(1){} and looked at what point the reset gets triggered. The code that causes it are these lines:
```
    // Initialise the local flash filesystem.
    // Create it if needed, mount in on /flash, and set it as current dir.
    bool mounted_flash = false;
    #if MICROPY_HW_FLASH_MOUNT_AT_BOOT
    mounted_flash = init_flash_fs(state.reset_mode);
    #endif
```

MartinManzinger, 2023-12-07:
With the same method I found out that the error occurs, when following code is executed:
in **ports/stm32/main.c** in the function **init_flash_fs**:
when the line ret = factory_reset_create_filesystem(); is executed.

MartinManzinger, 2023-12-07:
This leads to **ports/stm32/factoryreset.c**. In function **factory_reset_create_filesystem** the function call FRESULT res = f_mkfs(&vfs.fatfs, FM_FAT, 0, working_buf, sizeof(working_buf)); results in an reset.

MartinManzinger, 2023-12-07:
This leads to **/lib/oofatfs/ff.c**. Execution of line if (disk_write(pdrv, buf, b_vol, 1) != RES_OK) LEAVE_MKFS(FR_DISK_ERR);
The function call disk_write() causes the error.

MartinManzinger, 2023-12-07:
I changed the line **if (disk_write(pdrv, buf, b_vol, 1) != RES_OK) LEAVE_MKFS(FR_DISK_ERR);** to **if (1) LEAVE_MKFS(FR_DISK_ERR);**. Now I finally get a REPL response over the STLink COM port, that says:
MicroPython v1.22.0-preview.168.gb0e5af8a3.dirty on 2023-12-07; NUCLEO_H743ZI2 with STM32H743
Type "help()" for more information.
>>> MPY: can't create flash filesystem
MPY: can't mount flash

The board still resets and this REPL response repeats indefinitely.

So I guess there is a problem with low level writing to the flash of the controller. How can I solve this problem???

dpgeorge, 2023-12-12:
It sounds like there's something wrong with the internal flash when used as a filesystem, when erasing/writing this flash.

Can you try with just the following disable:
```
#define MICROPY_HW_HAS_FLASH (0)
#define MICROPY_HW_ENABLE_INTERNAL_FLASH_STORAGE (0)
#define MICROPY_HW_USB_MSC (0)
```
?

IHOXOHI, 2023-12-12:
Hi,
I use a stm32h743zi2 without any problème...?
I have download thé firmware by thé cube application...on thé USB power...?

IHOXOHI, 2023-12-12:
On other hand, I use too thé Discovery kit H7...
I'm looking for a library for the display, if possible with the touch function... If someone has an advice...

IHOXOHI, 2023-12-12:
On the second other hand, the import framebuffer command seems to no works on the featherMO express. It works on the M4...
Thanks.

MartinManzinger, 2023-12-12:
> Hi, I use a stm32h743zi2 without any problème...? I have download thé firmware by thé cube application...on thé USB power...?

Oh, that's interesting. Maybe it is indeed a faulty hardware I'm using. But two new boards with the exact same issue seems very unprobable. Maybe ST made something different with the fuse bits. I tried to mess around with those and bricked one board of mine XD.

> It sounds like there's something wrong with the internal flash when used as a filesystem, when erasing/writing this flash.
> 
> Can you try with just the following disable:
> 
> ```
> #define MICROPY_HW_HAS_FLASH (0)
> #define MICROPY_HW_ENABLE_INTERNAL_FLASH_STORAGE (0)
> #define MICROPY_HW_USB_MSC (0)
> ```
> 
> ?
I'll have a trie.

IHOXOHI, 2023-12-13:
The stm32G0B1RE works too with cube programmer...
Check the st-link jumper...
I download the .hex file by the erasing and programming tab, I just click on the start automatic mode button... The application tell me to disconnect the first device for connect the second...? But, it works, just like that.

IHOXOHI, 2023-12-13:
The stm32F411 works too...
Normally the application cube programmer see the board via the usb port, if the st-link jumper is right...?

pschindler, 2024-03-06:
I have a similar problem with a NUCLEO-H7232ZG board. After a reset, the yellow LD2 blinks briefly but then i cannot connect. I could solve the issue by pressing the USER button (connected to PC13) during reset. If I press the button during reset, I get a REPL and the shell works as expected.

jonnor, 2024-08-24:
It seems we conclude that it is a hardware issue on the particular stm32h743zi2 board, not MicroPython? Proposing to close this issue.

closed 2024-08-26 by dpgeorge

IHOXOHI, 2024-08-26:
Hi,
It's possible that I will create my 'virtual shop' in Spain. If there is a
way or something which could facilitate the market with original pyboard...
I'm interested. I would like to sold only one or 2 boards, really it's just
for legalized my hobby.
In this case, thanks to notify me.
Best regards,
Sellegri Jérôme.