micropython/micropython #13078: dynruntime.mk:103: *** architecture 'amd64' not supported.  Stop.

premanandaembed, 2023-11-28:
i'm trying to make file for use of micropython device but it was not worked on step **make ARCH=amd64** it is getting error as **dynruntime.mk:103: *** architecture 'amd64' not supported.  Stop.**

premanandaembed, 2023-11-29:
steps to reproduce or steps i tried:
	Microcontrol device:pico w
	IDE: Thony 4.1.3
	Ubuntu 22.04.3 LTS 
	log of sys.implementation._mpy:
		mpy version: 6
		mpy flags: -march=armv6m -mno-unicode
	Step-1: Tried to compile micropython release version 1.19 code from github. 
	Step-2: using Ubuntu 22.04.3 LTS for build as command make ARCH=armv6m and It is complied and successfully generated factorial.mpy file
	Step-2: Copied factorial.mpy file to pico w
	Step-3: created fact.py file inside pico w as below code:
			import factorial
			print(factorial.factorial(10))
[image]

	After these steps I'm getting when executing fact.py in pico w as ValueError: incompatible .mpy file

dpgeorge, 2023-11-29:
What does `sys.version` report on the board you are using?

> Tried to compile micropython release version 1.19 code from github.

Please use the version corresponding to `sys.version` retrieved from the board (likely 1.20.0 or 1.21.0).

premanandaembed, 2023-11-29:
thanks for reply it is 1.21.0

dpgeorge, 2023-11-29:
Please compile the factorial program with MicroPython v1.21.0.

premanandaembed, 2023-11-29:
great thanks it worked.

closed 2023-11-29 by premanandaembed