micropython/micropython #13084: formatting character values >= 128 gives unexpected results

jepler, 2023-11-28:
```
$ ./build-coverage/micropython
MicroPython 9c7067d9ad on 2023-11-28; linux [GCC 12.2.0] version
Use Ctrl-D to exit, Ctrl-E for paste mode
>>> s = f"{160:c}"
>>> len(s)
0
>>> s
' '
>>> print(s)
�
```

The same for `s = "%c" % 160`.

I think this is because the `'c'` formatter in objstr.c doesn't handle non-ASCII characters properly. It looks like this ends up being another way to get improper UTF-8 into a str() object, too.

dpgeorge, 2023-11-29:
Can probably use a fixed (stack allocated) vstr along with vstr_add_char(), and then copy the resulting vstr into the printer used in the formatter.

jepler, 2023-11-29:
There are at least 4 code paths affected:
 * "%c" % integer
 * "%c" % string
 * "{:c}".format()
 * format's "fill character" can also be unicode

An additional complication is whether the target type is a bytes or str.

I started down the road of implementing this but things seemed to be getting out of hand and I don't actually need this so I'll just put it aside.

jepler, 2025-09-05:
micropython counts precision/field width in bytes, not code points.
```
>>> print("|{:_^5}|".format(chr(225) * 2))
|_áá__|
```
vs
```
>>> print("|{:_^5}|".format(chr(225) * 2))
|áá_|
```

As with other operations, this means you can form a string with invalid UTF8 structure.

The format padding "character" is also treated as a byte, so attempting to pad with a non-ASCII code just fails (today)
```
>>> print("|{:\xe1^5}|".format(" " * 2))
ValueError: invalid format specifier
```

vs CPython
```
|á  áá|
```