micropython/micropython #13092: ESP32-S3 compile warnings

kdschlosser, 2023-11-29:
Compiling version 1.21.0 using ESP-IDF 5.0.4 for an esp32-s3 with octal SPIRAM I am getting the following warnings. 

This is the make command I am using

```
make BOARD=ESP32_GENERIC_S3 MICROPY_BOARD_VARIANT=SPIRAM_OCTAL
```

And here are the warnings. 
```
In file included from  /micropython/ports/esp32/managed_components/espressif__tinyusb/src/osal/osal_freertos.h:33,
                 from  /micropython/ports/esp32/managed_components/espressif__tinyusb/src/osal/osal.h:57,
                 from  /micropython/ports/esp32/managed_components/espressif__tinyusb/src/tusb.h:38,
                 from  /micropython/ports/esp32/managed_components/espressif__esp_tinyusb/include/tinyusb.h:11,
                 from  /micropython/ports/esp32/usb.c:35:
 /micropython/lib/berkeley-db-1.xx/PORT/include/queue.h:85: warning: "LIST_INIT" redefined
   85 | #define LIST_INIT(head) {                                               \
      |
In file included from  /esp-idf/components/esp_wifi/include/esp_private/esp_wifi_types_private.h:20,
                 from  /esp-idf/components/esp_wifi/include/esp_wifi_types.h:11,
                 from  /micropython/ports/esp32/mpconfigport.h:13,
                 from  /micropython/py/mpconfig.h:75,
                 from  /micropython/py/mpstate.h:31,
                 from  /micropython/py/runtime.h:29,
                 from  /micropython/ports/esp32/usb.c:27:
 /.espressif/tools/xtensa-esp32s3-elf/esp-2022r1-11.2.0/xtensa-esp32s3-elf/xtensa-esp32s3-elf/sys-include/sys/queue.h:563: note: this is the location of the previous definition
  563 | #define LIST_INIT(head) do {                                            \
      |
In file included from  /micropython/ports/esp32/managed_components/espressif__tinyusb/src/osal/osal_freertos.h:33,
                 from  /micropython/ports/esp32/managed_components/espressif__tinyusb/src/osal/osal.h:57,
                 from  /micropython/ports/esp32/managed_components/espressif__tinyusb/src/tusb.h:38,
                 from  /micropython/ports/esp32/managed_components/espressif__esp_tinyusb/include/tinyusb.h:11,
                 from  /micropython/ports/esp32/usb.c:35:
 /micropython/lib/berkeley-db-1.xx/PORT/include/queue.h:89: warning: "LIST_INSERT_AFTER" redefined
   89 | #define LIST_INSERT_AFTER(listelm, elm, field) {                        \
      |
In file included from  /esp-idf/components/esp_wifi/include/esp_private/esp_wifi_types_private.h:20,
                 from  /esp-idf/components/esp_wifi/include/esp_wifi_types.h:11,
                 from  /micropython/ports/esp32/mpconfigport.h:13,
                 from  /micropython/py/mpconfig.h:75,
                 from  /micropython/py/mpstate.h:31,
                 from  /micropython/py/runtime.h:29,
                 from  /micropython/ports/esp32/usb.c:27:
 /.espressif/tools/xtensa-esp32s3-elf/esp-2022r1-11.2.0/xtensa-esp32s3-elf/xtensa-esp32s3-elf/sys-include/sys/queue.h:567: note: this is the location of the previous definition
  567 | #define LIST_INSERT_AFTER(listelm, elm, field) do {                     \
      |
In file included from  /micropython/ports/esp32/managed_components/espressif__tinyusb/src/osal/osal_freertos.h:33,
                 from  /micropython/ports/esp32/managed_components/espressif__tinyusb/src/osal/osal.h:57,
                 from  /micropython/ports/esp32/managed_components/espressif__tinyusb/src/tusb.h:38,
                 from  /micropython/ports/esp32/managed_components/espressif__esp_tinyusb/include/tinyusb.h:11,
                 from  /micropython/ports/esp32/usb.c:35:
 /micropython/lib/berkeley-db-1.xx/PORT/include/queue.h:97: warning: "LIST_INSERT_HEAD" redefined
   97 | #define LIST_INSERT_HEAD(head, elm, field) {                            \
      |
In file included from  /esp-idf/components/esp_wifi/include/esp_private/esp_wifi_types_private.h:20,
                 from  /esp-idf/components/esp_wifi/include/esp_wifi_types.h:11,
                 from  /micropython/ports/esp32/mpconfigport.h:13,
                 from  /micropython/py/mpconfig.h:75,
                 from  /micropython/py/mpstate.h:31,
                 from  /micropython/py/runtime.h:29,
                 from  /micropython/ports/esp32/usb.c:27:
 /.espressif/tools/xtensa-esp32s3-elf/esp-2022r1-11.2.0/xtensa-esp32s3-elf/xtensa-esp32s3-elf/sys-include/sys/queue.h:584: note: this is the location of the previous definition
  584 | #define LIST_INSERT_HEAD(head, elm, field) do {                         \
      |
In file included from  /micropython/ports/esp32/managed_components/espressif__tinyusb/src/osal/osal_freertos.h:33,
                 from  /micropython/ports/esp32/managed_components/espressif__tinyusb/src/osal/osal.h:57,
                 from  /micropython/ports/esp32/managed_components/espressif__tinyusb/src/tusb.h:38,
                 from  /micropython/ports/esp32/managed_components/espressif__esp_tinyusb/include/tinyusb.h:11,
                 from  /micropython/ports/esp32/usb.c:35:
 /micropython/lib/berkeley-db-1.xx/PORT/include/queue.h:104: warning: "LIST_REMOVE" redefined
[... 114 lines cut ...]
                 from  /micropython/py/mpstate.h:31,
                 from  /micropython/py/runtime.h:29,
                 from  /micropython/ports/esp32/usb.c:27:
 /.espressif/tools/xtensa-esp32s3-elf/esp-2022r1-11.2.0/xtensa-esp32s3-elf/xtensa-esp32s3-elf/sys-include/sys/queue.h:773: note: this is the location of the previous definition
  773 | #define TAILQ_INSERT_AFTER(head, listelm, elm, field) do {              \
      |
In file included from  /micropython/ports/esp32/managed_components/espressif__tinyusb/src/osal/osal_freertos.h:33,
                 from  /micropython/ports/esp32/managed_components/espressif__tinyusb/src/osal/osal.h:57,
                 from  /micropython/ports/esp32/managed_components/espressif__tinyusb/src/tusb.h:38,
                 from  /micropython/ports/esp32/managed_components/espressif__esp_tinyusb/include/tinyusb.h:11,
                 from  /micropython/ports/esp32/usb.c:35:
 /micropython/lib/berkeley-db-1.xx/PORT/include/queue.h:161: warning: "TAILQ_REMOVE" redefined
  161 | #define TAILQ_REMOVE(head, elm, field) {                                \
      |
In file included from  /esp-idf/components/esp_wifi/include/esp_private/esp_wifi_types_private.h:20,
                 from  /esp-idf/components/esp_wifi/include/esp_wifi_types.h:11,
                 from  /micropython/ports/esp32/mpconfigport.h:13,
                 from  /micropython/py/mpconfig.h:75,
                 from  /micropython/py/mpstate.h:31,
                 from  /micropython/py/runtime.h:29,
                 from  /micropython/ports/esp32/usb.c:27:
 /.espressif/tools/xtensa-esp32s3-elf/esp-2022r1-11.2.0/xtensa-esp32s3-elf/xtensa-esp32s3-elf/sys-include/sys/queue.h:839: note: this is the location of the previous definition
  839 | #define TAILQ_REMOVE(head, elm, field) do {                             \
      |
```

robert-hh, 2023-11-29:
Micropython uses esp idf 5.02. Please check if the error happens as well with that version. If yes, please re-check your build set-up.

kdschlosser, 2023-11-29:
really??.. why did i think it was 5.0.4... huh. i will try that.

kdschlosser, 2023-11-29:
nope, same warnings using ESP-IDF 5.0.2

robert-hh, 2023-11-29:
I see. Just trued it an get the same warnings here. It is in the Btree module, where warnings are accepted. If that troubles you and you do not need Btree, the you can disable it.

kdschlosser, 2023-11-29:
The warning should be fixed no? Those same macros are defined in the ESP IDF and the ones in btree are overriding them. Are they the same macros?

and in mobtree.c you have the following code that is not running?

```
#if MICROPY_PY_BTREE

#include <stdio.h>
#include <errno.h> // for declaration of global errno variable
#include <fcntl.h>

// Undefine queue macros that will be defined in berkeley-db-1.xx headers
// below, in case they clash with system ones defined in headers above.
#undef LIST_HEAD
#undef LIST_ENTRY
#undef LIST_INIT
#undef LIST_INSERT_AFTER
#undef LIST_INSERT_HEAD
#undef LIST_REMOVE
#undef TAILQ_HEAD
#undef TAILQ_ENTRY
#undef TAILQ_INIT
#undef TAILQ_INSERT_HEAD
#undef TAILQ_INSERT_TAIL
#undef TAILQ_INSERT_AFTER
#undef TAILQ_REMOVE
#undef CIRCLEQ_HEAD
#undef CIRCLEQ_ENTRY
#undef CIRCLEQ_INIT
#undef CIRCLEQ_INSERT_AFTER
#undef CIRCLEQ_INSERT_BEFORE
#undef CIRCLEQ_INSERT_HEAD
#undef CIRCLEQ_INSERT_TAIL
#undef CIRCLEQ_REMOVE
```

There is actually 3 different locations the macros are defined

\esp-idf\components\openthread\openthread\third_party\tcplp\bsdtcp\sys\queue.h
\micropython\lib\berkeley-db-1.xx\PORT\include\queue.h
\micropython\lib\mynewt-nimble\porting\nimble\include\os\queue.h

the the esp-idf and nimble have the same macros where as berkeley-db has something completely different.

kdschlosser, 2023-11-30:
Th issue there is if those macros exist in nimble and also in the esp-idf than the errors are still going to continue. The only  way top go about it would be to undefine them somehow so that when the nimble components get compiled it can define the macros without any issue.

It looks like the macros used in nimble are the same as the ones in the esp-idf. The macros used in btree however are not the same thing. This could cause issues. If btree is able to be modified then maybe changing the macro names to something different should be done to keep it from causing any issues down the road.

robert-hh, 2023-11-30:
On a second glance I see that queue.h belonging to Btree is included when compiling esp32/usb.c. That should not happen. So something in the list of include directories has to be fixed.

andrewleech, 2023-11-30:
In many cases it shouldn't matter if (isolated) libraries have overalappying defines, as long as eg. nimble files don't need to #include btree headers and vice versa. But yeah when library directories need to be added to global Include paths, and they have generic filenames like `queue.h` it gets tricky.

It's possible to have different cflags for each file, but messy to have to define to many custom include paths. 
cmake will also let you have separate includes for each library which is probably what's going to be neccesary here.

kdschlosser, 2023-11-30:
While I am not having any issues due to the warnings that doesn't mean that one may not show up or this could cause issues in the future. It is going to be really tricky to handle because of having to add global include flags. 

I do not know the propagation semantics of C code. So I do not know if say I include the header with those definitions from btree and in the same source file also include the header from nimble would the macros do a backwards propagation? meaning the defines in btree end up being overridden by the defines in nimble so the code replacement that ends up being done by the preprocessor ends up putting the nimble replacement code into btree.. 

It may all depend on the order in which the source files get compiled because btree is not going to be including anything from nimble is it? The only time an issue would come up is if btree and nimble get imported into the same source file. 

IDK how that entire thing works. I would think that something has to be including both of those libraries into the same namespace in order for the compiler to be complaining about it.

dpgeorge, 2024-03-17:
Fixed by cd8eea2ae99542257a33b29c9dfdc02b63fc1bd6

closed 2024-03-17 by dpgeorge

kdschlosser, 2024-03-18:
FANTASTIC!!!