micropython/micropython #13094: Incorrect float values obtained from strings / program code with RP2 and esp8266

rkompass, 2023-11-29:
The RP2 Pico (and ESP8266) transfer floats wrongly into the internal representation:

Demonstration code:

```py
import array
import uctypes
from machine import mem32

def printfp32(s, x):
    a = array.array('f', (0,))
    a[0] = x
    a0 = mem32[uctypes.addressof(a)]
    sign = '-' if a0 >> 31 else ''
    exp = ((a0>>23) & 0xff) - 150
    frac = (0x800000 | (a0 & 0x7fffff))
    # print(f"{a[0]:12.6f} {a0:032b}")
    print(f'{s}   {x}: {sign} {frac} * 2**{exp}')
    
y = 360.98564736629
printfp32('y', y)
```
`y   360.9857:  11828778 * 2**-15`  on stm32 (Blackpill) <--- correct, closest value
but
`y   360.9858:  11828782 * 2**-15`  on RP2 Pico       <--- incorrect
`y   360.986:    11828776 * 2**-15`  on ESP 8266       <--- incorrect

this is a deviation of 4 LSB on RP2 Pico.

This was found to be one contributor to inaccuracies in a calculation discussed [here](https://github.com/orgs/micropython/discussions/13075), where both Pico and esp8266 exhibited much poorer precision than the other platforms using 32 bit FP math.

I was trying to trace the function for string to float conversion in the RP2040 SDK but didn't find any.
On the other hand, if this function is in the MP code base, how could it be different then on these platforms?

dpgeorge, 2023-11-29:
The easiest way to see the issue, and see if it's a conversion, arithmetic, or printing issue, is:
```py
import array
ar = array.array('f', (float("360.98564736629"),))
print(bin(int.from_bytes(bytes(ar), "little")))
```
On CPython, MicroPython unix and MicroPython stm32 this gives:
```
0b1000011101101000111111000101010
```
On MicroPython rp2 it gives:
```
0b1000011101101000111111000101110
```
Which is out by the 3rd LSB.  That's not good!  The error will be in the string-to-float conversion which is implemented in MicroPython.

(For the esp8266 it uses `MICROPY_OBJ_REPR_C`, which is 30-bit precision, so known to have inaccuracies.)

dpgeorge, 2023-11-30:
It looks like an issue with the pico-sdk and its floating point routines.  It does not compute 10**-11 correctly.

dpgeorge, 2023-11-30:
Bug reported to pico-sdk: https://github.com/raspberrypi/pico-sdk/issues/1566

rkompass, 2023-11-30:
In the discussion there was is remark by @scruss 

> If it turns out to be an RP2040 FP library issue (which it's increasingly looking like it may not be), the library used is Mark Owen's [Qfplib-M0](https://www.quinapalus.com/qfplib.html). Mark is extremely approachable.

So this library may do be the culprit?! My feeling was it is, then it is not, now: it is.

rkompass, 2023-11-30:
Looked it up in the Pico SDK:

powf, after checking arguments uses powf_1:

```
static inline float fpow_1(float x,float y) {
    return (float)exp(log((double)x)*(double)y); // using double-precision intermediates for better accuracy
}
```

But in 32 bit MP the computation is more accurate:
```
import array
from math import exp, log

def printfp32(x):
    xi = int.from_bytes(bytes(array.array('f', (x,))), "little")
    sign = '-' if xi >> 31 else ''
    exp = ((xi>>23) & 0xff) - 150
    frac = (0x800000 | (xi & 0x7fffff))
    print(f"{x}:   {xi:032b}")
    print(f'{x}:   {sign}{frac} * 2**{exp}')

x = 10.0
y = -11.0

printfp32(x ** y)

printfp32(exp(log(x)*y))
```

gives 
```
1e-11:   00101101001011111110110000000011
1e-11:   11529219 * 2**-60
9.999999e-12:   00101101001011111110110000000000
9.999999e-12:   11529216 * 2**-60
```
on the Pico
and
```
1e-11:   00101101001011111110101111111111
1e-11:   11529215 * 2**-60
1e-11:   00101101001011111110110000000000
1e-11:   11529216 * 2**-60
```
on the Blackpill (sm32f411)

rkompass, 2023-12-06:
The daily build has solved this issue. I close it. Thank you for the thorough and fast solution.

closed 2023-12-06 by rkompass