micropython/micropython #13108: web server and _thread not working

awsiuk, 2023-12-01:
### Discussed in https://github.com/orgs/micropython/discussions/12843

<div type='discussions-op-text'>

<sup>Originally posted by **awsiuk** October 31, 2023</sup>
Hi Group,

I use Micropython v1.21.0 on RP2040 with simple code:

```
import network
import socket
import machine
import json
import urandom
import time
import _thread

data={
    "temp": 0.0,
    "hum": 0.0,
    "AQI": 0,
    "TVOC": 0,
    "eCO2": 0,
    "lux": 0,
    "pressure": 0.0}

#read configuration for the application
def read_the_config():
    global config
    with open("pogoda.json") as json_data_file:
        config = json.load(json_data_file)

def t_web_server(ip):
    address = (ip, 80)
    addr = socket.getaddrinfo('0.0.0.0', 80)[0][-1]
    
    connection = socket.socket()
    connection.bind(addr)
    connection.listen(1)
    print(connection)
    
    while True:
        print("Web server is waiting for connection")
        client, client_addr = connection.accept()
        print("Incoming connection from:", str(client_addr))
        try:
            print("handling the client request")
            request = client.recv(2048)
            f=open("index.html","r")
            html_main=str(f.read())
            client.send('HTTP/1.0 200 OK\r\nContent-type: text/html\r\n\r\n')
            client.send(html_main)
            time.sleep(0.1)
            
        except:
            print("ERROR: unable to handle the client")
        print("Closing connection to:", str(client_addr))
        client.close()
    print("exiting")
    return

###############
read_the_config()
global config
print(config)

#setting up WiFi
global wlan
wlan = network.WLAN(network.STA_IF)
wlan.active(True)
wlan.connect(config['SSID'], config['PASSWORD'])
while wlan.isconnected() == False:
    print('Waiting for WiFi connection...')
    time.sleep(1)
ip = wlan.ifconfig()[0]
print(f'Wifi connected on {ip}')

#initiate collecting data thread
_thread.start_new_thread(t_web_server, (ip,) )

while True:
    time.sleep(1)

```    

it is almost ok, expect that after 2nd connection to RP2040 it no longer accept connections and stops responding to ping. There is no crash on RP2040 and it seems program is still running awaiting on connections, but nothing happens. try section does not trigger as there is no connection error, it simply does not even start. 
If i run t_web_server function outside _thread it works properly. So either i do something wrong with _thread init, or there is a BUG for network support within the _thread.  Can you help me with this?</div>

daed, 2023-12-10:
I suspect this is related:

https://github.com/micropython/micropython/issues/13079

In addition to the commenter in the other thread, I also recall reading somewhere that network interrupts are only handled by the first core.

peterhinch, 2024-01-12:
Consider using `asyncio` instead of threading.

daed, 2024-01-12:
> Then how to tell which CPU should handle interrupts? How to enforce threads to be handled by CPU 0 instead of CPU 1? As per mentioned thread, I quote "Apparently _thread does not support interrupts.". This imply that you cannot run web server and socket client at the same time. Unless there is other way to do that and share data cross 2 areas. To my understanding ARM Cortex M0/M0+ has just 1 core as per [ARM documentation](https://www.st.com/content/st_com/en/arm-32-bit-microcontrollers/arm-cortex-m0-plus.html). It has 2 stage pipelines to increase performance but it does not mean that it has 2 cores or 2 CPUs. To me that is single core and there is no way to chose different core as there is just 1 core. It supports interrupts so web server + TCP client should be possible. I'm worried that either:
> 
>     1. CPU is being occupied waiting for connection and then other actions are not possible
> 
>     2. there is a BUG in firmware of micropython that is not handling interrupts properly.

Like @peterhinch suggested, I'd recommend something asyncio-based.

My approach here on my current project is to use Microdot (https://github.com/miguelgrinberg/microdot/blob/main/src/microdot/microdot.py) for the webserver.  Microdot 2 uses asyncio by default, and the syntax is Flask-like.  This works well on the Pico W and it's pretty easy to use.

jonnor, 2024-08-24:
There is an alternative solution given above. Proposing to close this issue.

awsiuk, 2024-08-24:
I have found a solution better than walkaround. I switched from raspberry pico to Arduino nano ESP32 and all works.

closed 2024-08-24 by awsiuk