micropython/micropython #13124: WebAssembly/PyScript - random not really random

WebReflection, 2023-12-04:
turns out the `random` in MicroPython is really predictable due, probably, some hard-coded initial seed.

I have suggested to do this instead, and is seems to better represent the `random.seed(None)` intent for Python:

```python
import js
from random import random

random.seed(js.Date.new())
```

I believe defaulting to the current system time is the most sensible way to provide a non predictable, always same, random module, but I trust you to know better where to place that initial seed in your core.

robert-hh, 2023-12-04:
Whether random numbers in MicroPython are true random, or Pseudo-Random depends on the port and interface used.
The random module uses a Pseudo RNG, which however is seeded by a hardware RNG for many ports, especially UNIX, STM32, RP2, ESP32, ESP8266, MIMXRT, SAMD and some of the nrf boards . That hardware RNG seed should be sufficiently unpredictable, better than using a date/time. Unless however software deliberately seeds it with a constant value.
os.random, if supported, used a hardware RNG, which is available for the ports listed above.

WebReflection, 2023-12-04:
to answer further, we could monkey-patch MicroPython in WASM by using `crypto` namespace, but the issue seems to be in the published WASM port which always returns the same sequence of seeds between different initializations. I think there is a constant somewhere in there that produces this and it has nothing to do with the RNG you mentioned.

The prefix of the issue is, by no accident, WebAssembly indeed.

WebReflection, 2023-12-04:
@robert-hh thanks for the label, if anyone would be so kind these other two issues have the exact same label: https://github.com/micropython/micropython/issues/13125 https://github.com/micropython/micropython/issues/13126

dpgeorge, 2023-12-07:
This should be easy to fix by defining `MICROPY_PY_RANDOM_SEED_INIT_FUNC` to something that can generate true random values, or just a timestamp.

WebReflection, 2023-12-07:
```js
const [seed] = crypto.getRandomValues(new Uint32Array(1));
```

seed will be a real random value but I think `Date.now()` is also "*good enough*" for the purpose. It's really odd that two MicroPython engines would bootstrap at the exact same millisecond, imho.

dpgeorge, 2023-12-13:
Now fixed, the random module is seeded by `crypto.getRandomValues` (interpreters would start in roughly the same time, so the variation on the seed when using `Date.now()` would be quite limited and might be noticeable that there are only a few different starting seeds).

Also `random.seed(None)` now works.

See branch: https://github.com/dpgeorge/micropython/commits/webassembly-add-modjs/

closed 2023-12-13 by dpgeorge