micropython/micropython #13126: WebAssembly/PyScript - Missing Proxy traps

WebReflection, 2023-12-04:
In more than one occasion the lack of `Proxy.has` handler or others caused issues to code and otherwise perfectly valid JS use cases.

```js
const proxy_handler = {
    isExtensible() {
        return true;
    },
    # here
    has(target, prop) {
        throw Error("has not implemented");
    },
    get(target, prop) {
        //console.debug("GET", target._ref, prop);
        if (prop in target || typeof prop === "symbol") {
            return Reflect.get(target, prop);
        }
        const value = Module._malloc(3 * 4);
        Module.ccall("proxy_c_to_js_lookup_attr", "number", ["number", "string", "pointer"], [target._ref, prop, value]);
        return convert_mp_to_js_obj_jsside_with_free(value);
    },
    set(target, prop, value) {
        //console.debug("SET", target, prop, value);
        if (typeof prop === "symbol") {
            return Reflect.set(target, prop, value);
        }
        # here
        throw Error("set not implemented");
        return true;
    },
    # here
    deleteProperty(target, prop) {
        throw Error("deleteProperty not implemented");
    },
    # here
    ownKeys(target) {
        throw Error("ownKeys not implemented");
    },
};
```

I can help with the details or the code neede to make these traps work but right now basically anything more complex in the JS world, including Proxies from that world, would fail.

dpgeorge, 2023-12-13:
@WebReflection Can you please give some examples (in JS) of how you would use these proxy traps (has, delete, ownKeys) and what the expected semantics are?

dpgeorge, 2023-12-13:
As mentioned on Discord, maybe it's simpler to just delete these 3 not-implemented traps?

WebReflection, 2023-12-14:
not having traps that throws helps understanding what's missing or failing .... right now any `hasOwn` or `Object.keys` might fail unpredictably so ... broken-for-broken, please remove not handled traps first and let's see what else is needed to fix use cases. The examples are all over the place specially if in the JS world the proxy is ... a proxy of something else.

WebReflection, 2024-01-05:
Hopefully useful to better debug or understand what is going on ... I have managed to create some minimal example around this simple Web page where JS is there just to showcase everything should be fine:

```html
<!DOCTYPE html>
<script type="module">
  import "https://cdn.jsdelivr.net/npm/polyscript/core.js";

  globalThis.Class = class {
    constructor(ref) {
      this.ref = ref;
    }
    get py() {
      return this.ref;
    }
    log(env) {
      console.log(env, this.ref);
    }
  }

  const js = new Class({});
  js.log('JS');
  console.log('JS', 'py' in js);        // true
  console.log('JS', js.py === js.ref);  // true
</script>
```

## MicroPython case No.1

Adding this script to that page reveals nothing surprising but that's only because the passed along constructor value is a primitive:

```html
<script type="micropython">
  use_case = 'MicroPython #1'
  print(use_case)

  import js

  # passing any primitive just works
  instance = js.Class.new('OK')
  instance.log(use_case)

  # this is OK too
  print(use_case, hasattr(instance, 'py'))

  # these are all True
  print(use_case, instance.ref == instance.py)
  print(use_case, instance.ref == instance.ref)
  print(use_case, instance.py == instance.py)
</script>
```

Even passing nothing or `None` produces the same result: it's all good, that's great ... **but** ...

## MicroPython case No.2

```html
<script type="micropython">
  use_case = 'MicroPython #2'
  print(use_case)

  import js

  # pass a dict directly
  instance = js.Class.new({})
  instance.log(use_case)

  # Error: has not implemented - Why is that?
  print(use_case, hasattr(instance, 'py'))
</script>
```

Try above code instead and see that as soon as a non primitive reference is passed to JS, most basic operations are broken. In this case, `hasattr(instance, 'py')` should not ever even reach the Python reference behind the scene but somehow it throws out of the blue on the `instance` Proxy.

It's not super clear to me why Proxies would behave differently if their constructors or methods receive some Python reference ... is there any easy way to explain this and, hopefully, an easy fix too?

Is it at least clear that something is broken due missing `has` trap in the Proxy? That should just return True like it did in the previous use case, right?

## MicroPython case No.3

Enter the `create_proxy` "*magic*":

```html
<script type="micropython">
  use_case = 'MicroPython #3'
  print(use_case)

  import js

  # enter the Proxy land
  from pyodide.ffi import create_proxy

  instance = js.Class.new(create_proxy({}))
  instance.log(use_case)

  # this now works: True
  print(use_case, hasattr(instance, 'py'))

  # but these are all False - Why is that?
  print(use_case, instance.ref == instance.py)
  print(use_case, instance.ref == instance.ref)
  print(use_case, instance.py == instance.py)
</script>
```

The instance kinda works but any more complex logic around it would fail or create thousand objects (think DOM based libraries, diffing algorithms around equality, and so on) for no concrete reason.

In JS there is a FinalizationRegistry which works wonderfully well and it can also communicate back to Python when a Python reference held in JS does not need to exist anymore.

This dance is done even cross Workers in PyScript when the worker attribute is used, and the funny thing is that because we automatically transform to JS anything that needs to travel back and forward, the APIs might behave differently accordingly if it's us handling *once* the same source reference, or MicroPython (or Pyodide) itself.

- - -

As summary, I hope this might be at least a starting point to figure out what to improve and/or how.

dpgeorge, 2024-03-14:
All proxy traps have now been implemented, see https://github.com/micropython/micropython/pull/13583

closed 2024-03-14 by dpgeorge