micropython/micropython #13164: [Pico W] Latest nightly build for 20231208 invalid/corrupt

samveen, 2023-12-09:
The [nightly build](https://micropython.org/resources/firmware/RPI_PICO_W-20231208-v1.22.0-preview.236.ge1a7aa23f.uf2) on the download page for the Pico W on the micropython website is not a valid UF2 file:

```
samveen@devbox-4b:~ $ wget 'https://micropython.org/resources/firmware/RPI_PICO_W-20231208-v1.22.0-preview.236.ge1a7aa23f.uf2'
--2023-12-09 15:36:02--  https://micropython.org/resources/firmware/RPI_PICO_W-20231208-v1.22.0-preview.236.ge1a7aa23f.uf2
Resolving micropython.org (micropython.org)... 176.58.119.26
Connecting to micropython.org (micropython.org)|176.58.119.26|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 32768 (32K) [application/octet-stream]
Saving to: ‘RPI_PICO_W-20231208-v1.22.0-preview.236.ge1a7aa23f.uf2’

RPI_PICO_W-20231208 100%[===================>]  32.00K  --.-KB/s    in 0.001s  

2023-12-09 15:36:03 (36.1 MB/s) - ‘RPI_PICO_W-20231208-v1.22.0-preview.236.ge1a7aa23f.uf2’ saved [32768/32768]

samveen@devbox-4b:~ $ file RPI_PICO_W-20231208-v1.22.0-preview.236.ge1a7aa23f.uf2
RPI_PICO_W-20231208-v1.22.0-preview.236.ge1a7aa23f.uf2: data
samveen@devbox-4b:~ $ head -c 16 RPI_PICO_W-20231208-v1.22.0-preview.236.ge1a7aa23f.uf2  |hexdump
0000000 0000 0000 0000 0000 0000 0000 0000 0000
0000010
```
The previous one is fine and working correctly.

robert-hh, 2023-12-09:
Duplicate of #13163

samveen, 2023-12-09:
@robert-hh is it possible to update the issue template to add  `and discussions` right after `issues` in  `Please search existing issues before raising a new issue`.

I wouldn't have opened the issue if I had been though to check in discussions in addition to an issue search.

closed 2023-12-09 by samveen

robert-hh, 2023-12-09:
There is a search field in both issues and discussions.

dpgeorge, 2023-12-10:
This issue is now fixed.